moveit/moveit2 #2374: MAX_COLINEAR_NORM - Circle Error

ClaBirk, 2023-09-17:
### Circle error with Pilz industrial motion planner and mm sized circles

the value of the MAX_COLINEAR_NORM  is quite high - this limits the size of the circle as shown with the following python programm:
There is some example Points i want to use for my movement (1,74 mm Radius Circle)

import numpy as np

# Define the three points: start, interim, and goal
start = np.array([-3.510000/1000, 118.689003/1000, 89.000000/1000])
interim = np.array([-5.255000/1000, 120.434003/1000, 89.00000/1000])
goal = np.array([-3.510000/1000, 122.179001/1000, 89.000000/1000])

# Compute the vectors t, u, and v as defined in the C++ code
t = interim - start
u = goal - start
v = goal - interim

# Compute the cross product of t and u
w = np.cross(t, u)

# Check if the norm of the cross product is less than MAX_COLINEAR_NORM
MAX_COLINEAR_NORM = 1e-4
error_triggered = np.linalg.norm(w) < MAX_COLINEAR_NORM
print (error_triggered)

sjahr, 2023-09-22:
Sure, do you mind giving it a review?

ClaBirk, 2023-09-25:
i have seen the value has been changed to   

static constexpr double MAX_COLINEAR_NORM{ 1e-5 };

in the path_circle_generator.h

For the circles i want to plann this value has to be 1e-6.

sjahr, 2023-09-25:
This is exactly what is done in #2380 :wink: 

```diff
-  static constexpr double MAX_COLINEAR_NORM{ 1e-5 }; // old value
+  static constexpr double MAX_COLINEAR_NORM{ 1e-6 }; // New value
```

ClaBirk, 2023-10-02:
how to upgrade this? i tried:
sudo apt update 
sudo apt upgrade 

but still 1e-5 in the path_circle_generator.h

sjahr, 2023-10-04:
It is not yet part of a release, so you have [to build moveit2 from source](https://moveit.ros.org/install-moveit2/source/) with the [updated version ](https://github.com/sjahr/moveit2/tree/pr-increase_MAX_COLINEAR_NORM) until #2380 is merged and a new version of moveit2 released. I have not fixed the unittests yet. Feel free to help me if you have time

closed 2024-03-12 by github-actions[bot]