moveit/moveit2 #2383: executable 'servo_node_main' not found

ravijo, 2023-09-24:
### Description

I am trying to execute `ur_moveit.launch.py` on ROS Iron but getting the following exception:

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'servo_node_main' not found on the libexec directory '/opt/ros/iron/lib/moveit_servo' 
```

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04.3 LTS
* ROS Source or Binary build: Binary
* MoveIt Source or Binary build: Error occurs in both (source and binary)
* Which RMW (Fast DDS or Cyclone DDS): `rmw_cyclonedds_cpp`

### Steps to reproduce
```
$ ros2 launch ur_moveit_config ur_moveit.launch.py use_fake_hardware:=true ur_type:=ur5e use_sim_time:=true
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-09-24-17-53-28-365043-dell-21428
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'servo_node_main' not found on the libexec directory '/opt/ros/iron/lib/moveit_servo' 
```

ravijo, 2023-09-24:
This error does not appear in ROS Humble + MoveIt Humble Source, i.e., `git clone https://github.com/ros-planning/moveit2_tutorials -b humble --depth 1`

ibrahiminfinite, 2023-09-24:
Hi @ravijo, thanks for reporting the bug!
 
MoveIt Servo recently had a major update, which is the version on the `main` branch of MoveIt2.
The `servo_node_main`  was an executable name used in the old version and that is why you get the error on Iron.
These changes are not on Humble branch, so you there would not be any error there.

Also, is this error from a package outside of MoveIt2 ? or from MoveIt2 examples?

ravijo, 2023-09-25:
@ibrahiminfinite 

Thanks for the confirmation.

> Also, is this error from a package outside of MoveIt2 ? or from MoveIt2 examples?

Ya. It is coming from `ros2 launch ur_moveit_config ur_moveit.launch.py` which is invoking `moveit_servo` internally.

BTW, I came accoss another issue while following `moveit2_tutorials`. May I kindly request you to look at https://github.com/ros-planning/moveit2_tutorials/issues/767

Thank you so much!

sjahr, 2023-10-05:
@ravijo I am closing this issue since it seemed to be answered. Feel free to re-open new ones for more questions!

closed 2023-10-05 by sjahr

ravijo, 2023-10-06:
@sjahr 

Thanks for the followup. 

Another workaround for this issue is to create a symbolic link as shown below:

```
$ cd /home/user/ros2_ws/install/moveit_servo/lib/moveit_servo
$ ln -s servo_node servo_node_main
```

wei1224hf, 2024-01-31:
> Hi @ravijo, thanks for reporting the bug!
> 
> MoveIt Servo recently had a major update, which is the version on the `main` branch of MoveIt2. The `servo_node_main` was an executable name used in the old version and that is why you get the error on Iron. These changes are not on Humble branch, so you would not have any error there.
> 
> Also, is this error from a package outside of MoveIt2 ? or from MoveIt2 examples?

I get this error from Moveit2 example , 

os2 launch moveit2_tutorials jupyter_notebook_prototyping.launch.py start_servo:=true
[INFO] [launch]: All log files can be found below /home/gene/.ros/log/2024-01-31-15-15-42-218572-gene-virtual-machine-7582
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'servo_node_main' not found on the libexec directory '/opt/ros/iron/lib/moveit_servo'

ravijo, 2024-01-31:
@wei1224hf Please try running the humble version of the repository. For more information, please feel free to read the comments above

shrutic12, 2024-04-10:
Hello @ravijo I am facing the same error 

`[INFO] [launch]: All log files can be found below /home/user/.ros/log/2024-04-10-16-56-58-476502-user-OptiPlex-790-8682
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'servo_node' not found on the libexec directory '/home/user/ws_moveit2/install/moveit_servo/lib/moveit_servo' 
`

I have tried the workaround you suggest 

`$ cd /home/user/ros2_ws/install/moveit_servo/lib/moveit_servo
$ ln -s servo_node servo_node_main
`
and I am able to launch the robot and moveit without any issue. But the `ros2 service call /servo_node/start_servo std_srvs/srv/Trigger {}` service is not available. Can you help me out? 

Here is the log I see on launching : 
`[INFO] [launch]: All log files can be found below /home/user/.ros/log/2024-04-10-16-59-31-237483-user-OptiPlex-790-9396
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ur_ros2_control_node-1]: process started with pid [9413]
[INFO] [robot_state_publisher-2]: process started with pid [9415]
[INFO] [spawner-3]: process started with pid [9417]
[INFO] [move_group-4]: process started with pid [9419]
[INFO] [servo_node-5]: process started with pid [9422]
[INFO] [spawner-6]: process started with pid [9436]
[INFO] [spawner-7]: process started with pid [9439]
[INFO] [spawner-8]: process started with pid [9441]
[INFO] [spawner-9]: process started with pid [9443]
[INFO] [spawner-10]: process started with pid [9445]
[INFO] [spawner-11]: process started with pid [9447]
[INFO] [spawner-12]: process started with pid [9449]
[INFO] [spawner-13]: process started with pid [9457]
[ur_ros2_control_node-1] [WARN] [1712761173.354705101] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] [INFO] [1712761173.361099233] [resource_manager]: Loading hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1712761173.370810550] [resource_manager]: Initialize hardware 'ur10e' 
[ur_ros2_control_node-1] [WARN] [1712761173.371063959] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ur_ros2_control_node-1] [WARN] [1712761173.371237703] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ur_ros2_control_node-1] <state_interface name="velocity"> 
[ur_ros2_control_node-1]   <param name="initial_value">0.0</param> 
[ur_ros2_control_node-1] </state_interface>
[ur_ros2_control_node-1] [INFO] [1712761173.372803987] [resource_manager]: Successful initialization of hardware 'ur10e'
[robot_state_publisher-2] [INFO] [1712761173.262008963] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1712761173.262156432] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1712761173.262170954] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1712761173.262179600] [robot_state_publisher]: got segment camera_bottom_screw_frame
[robot_state_publisher-2] [INFO] [1712761173.262187590] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-2] [INFO] [1712761173.262195147] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1712761173.262202677] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1712761173.262210139] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1712761173.262217535] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1712761173.262225036] [robot_state_publisher]: got segment tcp_screwdriver
[robot_state_publisher-2] [INFO] [1712761173.262232520] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1712761173.262239815] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1712761173.262247135] [robot_state_publisher]: got segment world
[robot_state_publisher-2] [INFO] [1712761173.262254460] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1712761173.262261744] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1712761173.262268907] [robot_state_publisher]: got segment wrist_3_link
[ur_ros2_control_node-1] [INFO] [1712761173.377254514] [resource_manager]: 'configure' hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1712761173.377293830] [resource_manager]: Successful 'configure' of hardware 'ur10e'
[ur_ros2_control_node-1] [INFO] [1712761173.378522076] [resource_manager]: 'activate' hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1712761173.378554197] [resource_manager]: Successful 'activate' of hardware 'ur10e'
[servo_node-5] [WARN] [1712761173.400255188] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node-5] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node-5] to the Servo composable node in the launch file
[servo_node-5] [INFO] [1712761173.435964228] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.019787 seconds
[servo_node-5] [INFO] [1712761173.436311093] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[... cut ...]
[ur_ros2_control_node-1] [INFO] [1712761176.451122613] [controller_manager]: Configuring controller 'cartesian_motion_controller'
[ur_ros2_control_node-1] [INFO] [1712761176.458533618] [cartesian_motion_controller]: Forward dynamics solver initialized
[ur_ros2_control_node-1] [INFO] [1712761176.458569048] [cartesian_motion_controller]: Forward dynamics solver has control over 6 joints
[INFO] [spawner-12]: process has finished cleanly [pid 9449]
[ur_ros2_control_node-1] [INFO] [1712761317.853781026] [forward_position_controller]: activate successful

`

ravijo, 2024-04-11:
@shrutic12 

As said in above comments, old versions are using `servo_node_main`. And from ROS Iron, MoveIt2 made these changes. Please try running the Humble version. For more information, please feel free to read the comments above.