moveit/moveit2 #2401: Cartesian Path Planning in Python API

BBlumhofer, 2023-10-02:
**Is your feature request related to a problem? Please describe.**
The current Python API of Moveit2 does not support Cartesian Path planning. To use moveit solely ia python this would be very important

**Describe the solution you'd like**
Similar to the move_group_Interface a function to compute the cartesian path

dsandber, 2024-01-04:
@BBlumhofer Did you make any headway with this?  Do I understand correctly that the Python API is available in ROS1/MoveIt1 but not in ROS2/MoveIt2?  Is it just the thin Python wrapper around the C code that is missing?

dsandber, 2024-01-09:
I haven't seen it confirmed, but I believe the Python API for the Pilz industrial stuff (now in MoveIt) is missing from ROS2, which is forcing me to stay on ROS1.  This ticket suggests some kind of workaround:

https://github.com/ros-planning/moveit2/pull/2463

Do I understand correctly that in ROS2 the Python code is mainly thin wrappers that call the underlying C libraries?

paulpacaud, 2025-07-28:
This is unfortunate, and it has been useful for our ROS1 UR5 setup. I vote for implenting this in ROS2 moveit.