moveit/moveit2 #2402: Humble: moveit_core bullet collision detection header not found

gabrielfpacheco, 2023-10-02:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Binary
* If binary, which release version? `2.5.5-1jammy.20230919.231650`

### Steps to reproduce

1. Update and (re-)install moveit-core:
```console
sudo apt update && sudo apt install ros-humble-moveit-core
```

2. Make sure the release version is 2.5.5:
```console
$ apt show ros-humble-moveit-core
Package: ros-humble-moveit-core
Version: 2.5.5-1jammy.20230919.231650
```

3. Create an example package depending on moveit_core only: 
```cosole
ros2 pkg create \
 --build-type ament_cmake \
 --dependencies moveit_core \
 --node-name test_node issue_moveit_core
```

4. Update `test_node.cpp` to include Bullet's collision detector allocator:
```cpp
#include <cstdio>

#include <moveit/collision_detection_bullet/collision_detector_allocator_bullet.h>

int main(int argc, char ** argv)
{
  (void) argc;
  (void) argv;

  printf("hello world issue_moveit_core package\n");
  return 0;
} 
```

5. Build the package
```
~/ws_moveit$ colcon build --packages-select issue_moveit_core
```

### Expected behaviour
One should able to `#include <moveit/collision_detection_bullet/collision_detector_allocator_bullet.h>` and the compilation process
should be completed successfully as it does if `#include <moveit/collision_detection_fcl/collision_detector_allocator_fcl.h>` is used instead.

### Actual behaviour
The build fails since the header is not found. Indeed, it is the only thing installed at `<ROS_INSTALL_PATH>/include/moveit_core/moveit/` and not at `<ROS_INSTALL_PATH>/include/moveit/`.

### Backtrace or Console output

```console
~/ws_moveit$ colcon build --packages-select issue_moveit_core
[0.721s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/jammy/ws_moveit/install/hello_moveit' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.721s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/jammy/ws_moveit/install/hello_moveit' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> issue_moveit_core
--- stderr: issue_moveit_core                             
/home/jammy/ws_moveit/src/issue_moveit_core/src/test_node.cpp:3:10: fatal error: moveit/collision_detection_bullet/collision_detector_allocator_bullet.h: No such file or directory
    3 | #include <moveit/collision_detection_bullet/collision_detector_allocator_bullet.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/test_node.dir/build.make:76: CMakeFiles/test_node.dir/src/test_node.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/test_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< issue_moveit_core [0.17s, exited with code 2]

Summary: 0 packages finished [0.74s]
  1 package failed: issue_moveit_core
  1 package had stderr output: issue_moveit_core
```

gabrielfpacheco, 2023-10-02:
I think the issue is related to the fact that the destination path has been changed (unwillingly?) just for [collision_detection_bullet](https://github.com/ros-planning/moveit2/pull/1889/files#diff-049302fd4031b4efb956b186b90e32e634cc629a6a7740c3feb205b805881522L53) without updating [target_include_directories](https://github.com/ros-planning/moveit2/blob/iron/moveit_core/collision_detection_bullet/CMakeLists.txt#L12) accordingly.

However, as per the description of https://github.com/ros-planning/moveit2/pull/1889, it seems to me that the intention of the PR was not to modify the install path for collision_detection_bullet but only to remove ament_export_targets from this subfolder (maybe it occurred due to a wrong rebase process?).

sjahr, 2023-10-04:
Closing this because it is fixed with #2403 @gabrielfpacheco please re-open if that's not the case and thank you for the fix!

closed 2023-10-04 by sjahr

gabrielfpacheco, 2023-10-04:
> Closing this because it is fixed with #2403 @gabrielfpacheco please re-open if that's not the case and thank you for the fix!

Thank you for the fast reply :)