moveit/moveit2 #2408: Extend Planning Pipeline to allow calling multiple planners in sequence

sjahr, 2023-10-06:
## Motiviation
_In order to_ get better solutions
_As a_ user
_I want to_ to seed an optimizing planner with a solution from RRTConnect

_In order to_ configure the pipeline easily
_As a_ user
_I want to_ determine the order in which certain pipeline components are being called

## Current implementation and limitations
- A planner plugin + a bunch of plan request adapters are configured and the call order is figured out by some logic of a class `PlanningRequestAdapterChain` + inside the plugins 
- It is currently not possible in MoveIt2 to chain planners and the CHOMP/ STOMP post processing adapters don't seem to work
- PlanRequestAdapters are not taken into account for the configured maximum planning time
- The component call order is [very un-intuitive](https://github.com/ros-planning/moveit2/blob/739afb649f5e248cb7d049bb35348a57fe5e5287/moveit_core/planning_request_adapter/src/planning_request_adapter.cpp#L103) and [apparently prone to errors](https://github.com/ros-planning/moveit2/pull/2234)

## Proposed refactoring
Depends on #2288 
- [ ] Call multiple planners in sequence
  - [ ] Enable loading multiple planner plugins
  - [ ] Call planner plugins in load-order
  - [ ] Make sure, planner plugins use reference trajectory if available
- [ ] Determine pipeline order with configuration in yaml
  - [ ] Distinguish between PlanRequestAdapters (Pre-processing) and PlanResponseAdapters (Post-processing)
  - [ ] Remove all logic from plugins that messes with the call order
  - [ ] Remove planner PlanRequestAdapters (CHOMP, STOMP)
  - [ ] Call PlanRequestAdapter vector, Planner vector and PlanResponseAdapters vector in the correct order

MarqRazz, 2023-10-24:
@sjahr what is the status here? Could you please update the milestone and link any pending PR's?

sjahr, 2023-10-25:
Both PRs need to be merged in order to close this issue. I've linked the second one, so this is considered done once that issue is merged

v4hn, 2023-11-03:
You spent a lot of work on this, and it's definitely a good idea to get rid of the planning request adapter recursive call structure.

The main shortcoming I keep encountering with the recursive PRA structure is that it does not allow support for anytime planners that yield (over time improving) solutions continuously. It's a very nice feature of the tesseract GUI, for example.
When using any optimizing planner, they could yield the current best solution every few steps, which would have to be processed by all response adapters before being send, e.g., to RViz for visualization. That way (in theory) the user could stop plan refinement of, e.g., RRT*/PRM* with high timeouts once they think the current plan is nice enough.
I know this is out of scope for you now, but it's an interesting other motivation to accept a redesign of the PRAs.

Could you please also comment on why you think MTC wrappers are inadequate for chaining planners the way you want to do it? In theory they also provide an alternative for response adapters altogether - with the main advantage that you can introspect them at any point for debugging.

sjahr, 2023-11-17:
@v4hn I don't think that MTC wrappers are inadequate at all but it should be possible with the core MoveIt planning pipeline. For a single state to state motion planning problem MTC seems to be overkill.

sjahr, 2023-12-06:
List of open relevant PRs

- https://github.com/ros-planning/moveit2/pull/2457
- https://github.com/ros-planning/moveit_resources/pull/191
- https://github.com/ros-planning/moveit2_tutorials/pull/834
- https://github.com/ros-planning/moveit2_tutorials/pull/820

closed 2023-12-06 by sjahr