moveit/moveit2 #2442: (moveit_py) Extend Trajectory Execution Manager

MatthijsBurgh, 2023-10-12:
The trajectory execution manager is added partially to `moveit_py` (https://github.com/ros-planning/moveit2/pull/2406). It is still missing the following interfaces, which are provided by the CPP interface:
- `getTrajectories`
- `execute`
- `executeAndWait`
- `waitForExecution`
- `getCurrentExpectedTrajectoryIndex`
- `getLastExecutionStatus`
- `getControllerManagerNode`

As these interfaces are missing, doesn't mean all these interfaces should be implemented. That decision is still open to the developers.

bhavuk2002, 2023-10-12:
!assign me

bhavuk2002, 2023-10-12:
exactly what you expect and what should i do?

MatthijsBurgh, 2023-11-27:
@sjahr @peterdavidfagan can we add a label to this issue? So GH actions doesn't close it.

JensVanhooydonck, 2023-12-02:
I've added part of the functions for what i needed at the moment. 
TODO:
 -  `getTrajectories`
 - `getCurrentExpectedTrajectoryIndex`
 - `getControllerManagerNode`