moveit/moveit2 #2458: Python bindings for move_group_interface

m-elwin, 2023-10-18:
**Is your feature request related to a problem? Please describe.**
Are there any plans for adding python bindings for the move_group_interface from C++ (e.g. something equivalent to the Move Group Python Interface in ROS 1)? 

The moveit_py API seems to only wrap the direct C++ API and does not seem to wrap `moveit::planning_interface::MoveGroupInterface`. 

**Describe the solution you'd like**
A python API for communicating with a move_group node similar to the Move Group Python Interface in ROS 1 (and probably wrapping `moveit::planning_interface::MoveGroupInterface`.

It would allow for a python tutorial that parallels [Your First C++ MoveIt Project](https://moveit.picknik.ai/main/doc/tutorials/your_first_project/your_first_project.html)

**Describe alternatives you've considered**
Using existing moveit_py functionality seems to cover most use cases. However, if an existing robot package provides a launchfile that sets up a move_group node with acceptable parameters, being able to use a `MoveGroupInterface` directly from python seems like it could provide an easier experience than needing to use C++ or needing to setup those parameters again.

**Additional context**
1. It seems like there was some initial work in moveit_ros/planning_interface toward making python bindings, but its currently commented out.
2. Would the work on these bindings be most appropriate in moveit_ros/planning_interface or as additions to moveit_py? I am willing to help work on this if it makes sense as a feature.

JaisonJose241, 2023-10-25:
I see some scripts showing examples in moveit tutorial repo, but can't find where the bindings or required libraries are>

m-elwin, 2023-12-11:
I have begun to address this issue with PR #2574