moveit/moveit2 #2483: MoveItConfigsBuilder with UR robot is not working

ravijo, 2023-10-25:
### Description
I am trying to adapt [MoveItConfigsBuilder](https://github.com/ros-planning/moveit2/blob/7b621d7bcce8ba332cec84e48ddc0ba566cdfbd2/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py) to UR3e robot while using Pilz Industrial Motion Planner. The `MoveItConfigsBuilder` expects certain files inside the [config folder](https://github.com/ros-planning/moveit2/blob/7b621d7bcce8ba332cec84e48ddc0ba566cdfbd2/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py#L13) of robot_moveit_config package. Unfortunately, the package structure is different in the case of UR robot. Below are the two important packages inside `/opt/ros/humble/share` folder:
```
├── ur_description
│   ├── cmake
│   ├── config
│   │   ├── initial_positions.yaml
│   │   └── ur3e
│   │       ├── default_kinematics.yaml
│   │       ├── joint_limits.yaml
│   │       ├── physical_parameters.yaml
│   │       └── visual_parameters.yaml
│   ├── environment
│   ├── launch
│   ├── meshes
│   ├── rviz
│   └── urdf
│       ├── inc
│       │   ├── ur_common.xacro
│       │   └── ur_transmissions.xacro
│       ├── ur_macro.xacro
│       ├── ur.ros2_control.xacro
│       └── ur.urdf.xacro
├── ur_moveit_config
│   ├── cmake
│   ├── config
│   │   ├── controllers.yaml
│   │   ├── kinematics.yaml
│   │   ├── ompl_planning.yaml
│   │   └── ur_servo.yaml
│   ├── environment
│   ├── launch
│   │   └── ur_moveit.launch.py
│   ├── local_setup.bash
│   ├── package.xml
│   ├── rviz
│   │   └── view_robot.rviz
│   └── srdf
│       ├── ur_macro.srdf.xacro
│       └── ur.srdf.xacro
```
Below is a snippet from my launch file (ur_moveit.launch.py):
```
    robot_name = "ur3e"
    moveit_config = (
        MoveItConfigsBuilder(robot_name="ur3e")
        .robot_description(file_path=get_package_share_directory("ur_description") + "/urdf/ur.urdf.xacro", mappings={"name": robot_name, "ur_type": robot_name})
        .robot_description_semantic(file_path=get_package_share_directory("ur_moveit_config") + "/srdf/ur.srdf.xacro", mappings={"name": robot_name})
        .trajectory_execution(file_path=get_package_share_directory("ur_moveit_config") + "/config/controllers.yaml")
        .planning_pipelines(pipelines=["pilz_industrial_motion_planner"])
        .to_moveit_configs()
    )
```

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build: Source (humble branch)

### Steps to reproduce
```
$ ros2 launch ur_robot_driver ur3e.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false initial_joint_controller:=scaled_joint_trajectory_controller
$ ros2 launch hello_moveit ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true
```

### Actual behaviour
Upon running my `ur_moveit.launch.py`, it can not find the package as shown below:
```
$ ros2 launch hello_moveit ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-10-25-12-29-06-950816-dell-127409
[... 168 lines cut ...]
  [move_group-1] [ERROR] [1698204681.219007457] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Workspace Bounds': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219032644] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Start State Path Constraints': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219043489] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Start State In Collision': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219060317] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Start State Path Constraints': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219071948] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Start State Bounds': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219091423] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Start State Path Constraints': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219101670] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Start State In Collision': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219117791] [moveit.planning_request_adapter]: Exception caught executing adapter 'Fix Start State Path Constraints': acceleration limit not set for group ur_manipulator
  [move_group-1] Skipping adapter instead.
  [move_group-1] [ERROR] [1698204681.219136202] [moveit.ros_planning.planning_pipeline]: Exception caught: 'acceleration limit not set for group ur_manipulator'
  [move_group-1] [INFO] [1698204681.219164403] [moveit_move_group_default_capabilities.move_action_capability]: Unknown event
  [rviz2-2] [INFO] [1698204681.219526289] [move_group_interface]: Planning request aborted
  [rviz2-2] [ERROR] [1698204681.220494998] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
  ```
</details>

### Question
I want to know how to incorporate `MoveItConfigsBuilder` to UR3e robot while using Pilz Industrial Motion Planner.

ravijo, 2023-10-25:
@peterdavidfagan

Sorry for tagging you here. But it seems related to one of your posts. I am following your suggestions from [this post](https://github.com/ros-planning/moveit2/issues/2014#issuecomment-1470836957) but got stuck in between.

ravijo, 2023-10-25:
I should have mentioned that together with `pilz_industrial_motion_planner_planning.yaml`, I also copied `joint_limits.yaml` file. For every joint, I have set the following attributes:

    has_acceleration_limits: true
    max_acceleration: 0.1

But somehow, these values are not read as indicated by the command below:
```
$ ros2 param get move_group robot_description_planning.joint_limits.elbow_joint.max_acceleration
Double value is: nan
```

Anyways, how to use `MoveItConfigsBuilder` with UR3e robot while using Pilz Industrial Motion Planner?

TZECHIN6, 2023-10-25:
Sorry, but why you choose to use this? UR provides a moveit config launch in their offical ros2 repo. I can run moveit application with that.

ravijo, 2023-10-25:
@TZECHIN6 

Thank you very much for the quick response. 

> Sorry, but why you choose to use this? UR provides a moveit config launch in their offical ros2 repo. I can run moveit application with that.

I can run MoveIt but I want to use Pilz Industrial Motion Planner. This is why, I made a copy of `ur_moveit.launch.py` with the following change:
```
    pilz_planning_pipeline_config = {
        "move_group": {
            "planning_plugin": "pilz_industrial_motion_planner/CommandPlanner",
            "request_adapters": """default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/FixWorkspaceBounds default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/SetMaxCartesianEndEffectorSpeed""",
            "capabilities" : "pilz_industrial_motion_planner/MoveGroupSequenceAction pilz_industrial_motion_planner/MoveGroupSequenceService",
        }
    }
    pilz_planning_yaml = load_yaml("ur_moveit_config", "config/pilz_industrial_motion_planner_planning.yaml")
    pilz_planning_pipeline_config["move_group"].update(pilz_planning_yaml)
```

With this change, I get the error `acceleration limit not set for group ur_manipulator`.

ravijo, 2023-10-25:
@peterdavidfagan 

Thank you for sharing a reference to `panda_moveit_config`. Actually, I have seen it already. Based on this reference, I copied `joint_limits.yaml`, `pilz_industrial_motion_planner_planning.yaml`, and `pilz_cartesian_limits.yaml` to `ur_moveit_config`.  Lastly, I used `MoveItConfigsBuilder` as said before. But there was ` "package 'ur3e_moveit_config' not found` error.

Please note that the package structure is different in the case of UR robot.

Mennosytsma, 2023-10-25:
Hi @ravijo, 

I am running into a similar problem, my goal is to control my UR5 using the moveit python bindings as also tried in #2014, but even with all the valuable hints I did not succeed yet as I don't succeed to create my own `my_ur_moveit_config` package which has my own `my_ur.urdf.xacro` in which I added some links like a table on which the robot stands and then links to `ur_description` and is based on `ur_moveit_config`. 

As far as I understand from the comment by @peterdavidfagan we should do the same, create a different package `my_ur_moveit_config` that has the specific format that [MoveItConfigsBuilder](https://github.com/ros-planning/moveit2/blob/main/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py) requires. In your case you are giving `ur3e` as robot name argument, which is why `MoveItConfigsBuilder` will look for  the`ur3e_moveit_config` package.

This is also the point where it gets confusing to me, because after creating the `my_ur_moveit_config` pacakge in the way described above it does not have a `robot_name.urdf` and `robot_name.srdf` but only the `xacro` files. which makes that  `MoveItConfigsBuilder` does not accept it.

gavanderhoorn, 2023-10-25:
It might not necessarily be any *easier* if this is your first attempt at using MoveIt, but perhaps the approach described by @tylerjw in [ROSCON 2023 - Retro ROS 2 Launch](https://tylerjw.dev/posts/xml-launch/) would be worth checking out.

Package sources can be found here: [xml_launch/easy_launch_demo](https://github.com/tylerjw/tylerjw.dev/tree/main/content/posts/xml_launch/easy_launch_demo).

We've discussed it a bit in https://github.com/ros2/launch/issues/729 as well.

ravijo, 2023-10-26:
@Mennosytsma 
> create a different package `my_ur_moveit_config` that has the specific format that `MoveItConfigsBuilder` requires

Thank you very much. I appreciate your suggestion but it seems a lot of work just to start the MoveIt!

@gavanderhoorn 
> Package sources can be found here: [xml_launch/easy_launch_demo](https://github.com/tylerjw/tylerjw.dev/tree/main/content/posts/xml_launch/easy_launch_demo).

Thank you for supporting us from a decade! Thanks for sharing this package. After looking carefully into it, I realized that the joint limits must be available inside `robot_description_planning` namespace. I added it to my launch file and it worked like a charm. Below is a summery of my work:

1. In [joint_limits.yaml](https://github.com/ravijo/ur_pilz_demo/blob/ea51c9cc0efd34406fb57419f13bdaef0185929a/config/joint_limits.yaml) file, enabled `has_acceleration_limits` and set `max_acceleration`. This file is copied from [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/60cbb20c319f41ceae470eecac97fd1000912bd9/config/ur3e/joint_limits.yaml) and kept inside the `config` subdirectory of my package.
2. Kept [pilz_industrial_motion_planner_planning.yaml](https://github.com/ravijo/ur_pilz_demo/blob/ea51c9cc0efd34406fb57419f13bdaef0185929a/config/pilz_industrial_motion_planner_planning.yaml) and [pilz_cartesian_limits.yaml](https://github.com/ravijo/ur_pilz_demo/blob/ea51c9cc0efd34406fb57419f13bdaef0185929a/config/pilz_cartesian_limits.yaml) inside the `config` subdirectory of my package.
3. Modified [ur_moveit.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/f499df8c94452ff223cc4e396374f9235f166f29/ur_moveit_config/launch/ur_moveit.launch.py) to read the edited `joint_limits.yaml` file, as show [here](https://github.com/ravijo/ur_pilz_demo/blob/ea51c9cc0efd34406fb57419f13bdaef0185929a/launch/ur_pilz.launch.py#L38-L40)
4. Read the edited `joint_limits.yaml` file, as shown [here](https://github.com/ravijo/ur_pilz_demo/blob/ea51c9cc0efd34406fb57419f13bdaef0185929a/launch/ur_pilz.launch.py#L124-L126)
5. Read pilz planning configuration files from the  the `config` subdirectory of my package, as shown [here](https://github.com/ravijo/ur_pilz_demo/blob/ea51c9cc0efd34406fb57419f13bdaef0185929a/launch/ur_pilz.launch.py#L128-L137)

Finally, I uploaded this package to GitHub. Maybe @Mennosytsma can give it a try!

Mennosytsma, 2023-10-27:
@ravijo Thank you for providing your working example. I realized that you managed to find a workaround for your issue that I cannot directly apply for my issue now as far as I can see, but it will definitely help in the later steps to customize my setup with the UR.

ravijo, 2023-10-27:
@Mennosytsma 

>  ... I cannot directly apply for my issue ...

Would you mind elaborating on it a bit more? I plan to improve the package to cover as many cases as possible.

Thanks

Mennosytsma, 2023-10-27:
@ravijo 
I have opened a new issue #2493, to follow up on the discussions in #2014 on trying to get the [python api tutorial](https://github.com/ros-planning/moveit2_tutorials/tree/main/doc/examples/motion_planning_python_api) to work for a UR robot.
One of the proposed solutions in #2014 was to create a single dictionary of the parameters using `MoveItConfigsBuilder`, where I faced exactly the same struggles as you had.

ravijo, 2023-10-27:
@Mennosytsma 

Thank you very much.

As you said, we should create a new package named `my_ur_moveit_config` with the specific format that [MoveItConfigsBuilder](https://github.com/ros-planning/moveit2/blob/main/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py) requires.

However, in the case of UR robots, there are many variations, such as UR3, UR3e, UR5, UR5e, etc. I don't think creating `my_ur3_moveit_config`, `my_ur3e_moveit_config`, `my_ur5_moveit_config`, `my_ur5e_moveit_config` and so on is a good idea. This is why I preferred directly reading the files from the original `ur_moveit_config` package.

peterdavidfagan, 2023-10-27:
> However, in the case of UR robots, there are many variations, such as UR3, UR3e, UR5, UR5e, etc. I don't think creating my_ur3_moveit_config, my_ur3e_moveit_config, my_ur5_moveit_config, [...]

It would be nice to have a shared xacro where you launch args determine which robot urdf/srdf gets loaded, this will cut down on config duplication. I may have mentioned this before but there are plans for the lab I work in to update the current workspaces for UR robots in the New Year. I'd be happy to contribute to some of the configs if they are open source. Thanks @ravijo for leading the way on this.

Mennosytsma, 2023-10-27:
@ravijo I completely agree on your point that it is not really desired to create all these separate packages for the different variants of the UR robots. 
The way that the original [ur_moveit.launch](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_moveit_config/launch/ur_moveit.launch.py) is setup also gives us the option to launch any of the above mentioned robots given the argument `ur_type`.
In practice however I almost always have the real robot on top of some table, which I prefer not to collide with and therefore I would like to include in the robot description, which is why I do create a `my_robot_config` package, where I place a slightly changed version of `ur.urdf.xacro`. This includes that `table_link` and I update `ur_macro.srdf.xacro` to also include them with collision checking.
@peterdavidfagan As far as I understand arguments like `description_package` and  `moveit_config_package` in this launch file can then be used to launch the robot in the specific setup, while you can still choose any of the URs (as we have both UR3e, UR5 and UR5e available here). I have no clue if this is a proper/clean way to do this, if you agree on this I would not mind to write this out in some usage guideline/example that can be added to any documentation. 

Getting the robot launched that way works quite well for me, but making the next step and also launching a node where I use `moveit_py`  from the launch file in this format and passing it all the needed parameters is a problem for me. As I think that my issue is more related to that I don't understand what type of config `moveit_py` exactly requires and that drifts quite a bit from this topic I started the new issue #2493

ravijo, 2023-10-27:
@Mennosytsma 

I understand your requirement. In the past, I remember creating a separate package to add my table link and CAD models. Internally, my URDF was pointing to the one given by Universal Robots. On the other hand, I looked up the source code of [MoveItConfigsBuilder](https://github.com/ros-planning/moveit2/blob/main/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py) to understand how it works.

BTW, it seems that you are mixing `moveit_py` with this issue. I suggest first making sure you can control your robot from RViz using the MoveIt Motion Planning window. Because it confirms that MoveIt configurations are done right. After that, you can start coding in `moveit_py`.

Mennosytsma, 2023-10-27:
@ravijo 

> BTW, it seems that you are mixing moveit_py with this issue. I suggest first making sure you can control your robot from RViz using the MoveIt Motion Planning window. Because it confirms that MoveIt configurations are done right. After that, you can start coding in moveit_py.

As I tried to point out in the other issue #2493 I do have part of the MoveIt configurations configured correctly to control the robot in RViz trough the motion planning window, it it is just that the `moveit_py` node does not work. As far as I concluded the only thing that could be wrong here is passing the parameters from the launch file to the `moveit_py` node is going wrong, and it is to me unclear what is exactly required by `moveit_py`. I have tried using `MoveItConfigsBuilder` for that, but as we discussed above that function is not easy to use with the UR config packages.

ravijo, 2023-11-02:
@peterdavidfagan

Looks like there is more work to be done in order to use `MoveGroupInterface`. I updated planning configuration as shown below (Note that I am able to use MoveIt Motion Planning Plugin in RViz without these updates, but C++ `MoveGroupInterface` is unhappy):

```
    # Planning Configuration
    pilz_planning_pipeline_config = {
        "planning_pipelines": ["pilz_industrial_motion_planner"],  # <- Newly added
        "default_planning_pipeline": "pilz_industrial_motion_planner",  # <- Newly added
        "pilz_industrial_motion_planner": {},  # <- Newly added
        "move_group": {},
        "robot_description_planning":{},
    }
    pilz_planning_yaml = load_yaml("ur_pilz_demo", "config/pilz_industrial_motion_planner_planning.yaml")
    pilz_planning_pipeline_config["move_group"].update(pilz_planning_yaml)
    pilz_planning_pipeline_config["pilz_industrial_motion_planner"].update(pilz_planning_yaml)  # <- Newly added

    pilz_cartesian_limits_yaml = load_yaml("ur_pilz_demo", "config/pilz_cartesian_limits.yaml")
    pilz_planning_pipeline_config["robot_description_planning"].update(pilz_cartesian_limits_yaml)
```

Now, I can use C++ and it works.

#### Quick Question

The above configuration is duplicating `pilz_industrial_motion_planner`. Is there any way to make it cleaner? (The complete launch file can be seen [here](https://github.com/ravijo/ur_pilz_demo/blob/6495d88a63519a65660a4de86ea9557523aebe56/launch/ur_pilz.launch.py))

This issue has been mentioned https://github.com/ros-planning/moveit2_tutorials/issues/808