moveit/moveit2 #2485: Cannot display trajactory line using with UR5e [rviz_visual_tool]

TZECHIN6, 2023-10-25:
### Description

I created my own move_group_interface.cpp and tried to repeat those examples demonstrated in [here](https://moveit.picknik.ai/main/doc/examples/move_group_interface/move_group_interface_tutorial.html). The motion controls are fine, but the visualization is a bit off. It cannot show the trajectory as expected (green line). Please see the videos for actual behaviors.

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Binary
* If binary, which release version? not sure, i could provide it if someone guide me
* If source, which branch? N/A
* Cyclone DDS

### Steps to reproduce
1. binary install moveit2 and ur driver
2. launch ur driver, moveit and ursim
3. launch the Cartesian_square.launch.py in this [repo](https://github.com/HKU-ur5e/hello_moveit_ur5e)

### Expected behaviour
1. The motion should follow: home position -> goal pose -> cartesian motion (down and up)
2. The trajactory for each plan should shown in a green line likes the example

### Actual behaviour
1. Motion control is fine
2. Only goal pose marker is shown, no green lines/dots for trajactory.

### Backtrace or Console output

N/A

[Screencast from Wednesday, October 25, 2023 03:47:51 PM HKT.webm](https://github.com/ros-planning/moveit2/assets/66129247/6b909e1b-777d-43d1-a8ee-b2f0a53e82f6)

ravijo, 2023-10-26:
I faced it as well. It may be due to no gripper attached.

TZECHIN6, 2023-10-27:
Have you solved this issue @ravijo? Can the trajactory line present the movement of ur_tcp?

sjahr, 2023-10-27:
The RVIZ plugin and moveit2 communicate the trajectory line over a topic interface. Have you verified that publisher and subscriber topic name match?

TZECHIN6, 2023-10-27:
In my cpp program [see here](https://github.com/HKU-ur5e/hello_moveit_ur5e/blob/feat/visualization/src/cartesian_ur5e.cpp#L51C3-L52C93), the topic should be `move_group_tutorial`,

And I add a new marker array by this using Rviz `Add` button at bottom left.

Sorry I am not so familiar with this... it would be nice if you could help me to provide the information to debug this issue

closed 2024-01-26 by github-actions[bot]