moveit/moveit2 #2498: [moveit_py] `rclpy.Time` not converted to `rclcpp::Time`

MatthijsBurgh, 2023-10-30:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source or Binary build? binary
* If binary, which release version? `2.8.0-2jammy.20231009.234856`
* Which RMW (Fast DDS or Cyclone DDS)? Humble or Rolling should be using Cyclone.: Cyclone

### Steps to reproduce
```py
moveit.get_planning_scene_monitor().wait_for_current_robot_state(self.node.get_clock().now(), 10.0)
```

### Expected behaviour
Should accept `rclpy.Time`

### Actual behaviour
`TypeError` as it ony accepts `rclcpp::Time``

### Backtrace or Console output

```py
[robot_app-1]     self._moveit_planning_scene_monitor.wait_for_current_robot_state(self.node.get_clock().now(), 10.0)
[robot_app-1] TypeError: wait_for_current_robot_state(): incompatible function arguments. The following argument types are supported:
[robot_app-1]     1. (self: moveit.planning.PlanningSceneMonitor, arg0: rclcpp::Time, arg1: float) -> bool
[robot_app-1] 
[robot_app-1] Invoked with: <moveit.planning.PlanningSceneMonitor object at 0x7f3bd6323d70>, Time(nanoseconds=1698656457063896314, clock_type=ROS_TIME), 10.0
```

peterdavidfagan, 2023-10-30:
Thanks for flagging this @MatthijsBurgh and yes relevant, I'll look to post a fix when I get the time. I plan to finalize the [policy](https://github.com/ros-planning/moveit2/pull/2494) module and an example, once complete will look to address this.

MatthijsBurgh, 2023-12-14:
We should label it persistent

JensVanhooydonck, 2024-02-06:
Still applicable

MatthijsBurgh, 2024-02-07:
@sjahr can you label it persistent?

Shobinvinod, 2025-04-22:
I've opened a PR to address this issue: [#3453](https://github.com/ros-planning/moveit2/pull/3453). It introduces a Pybind11 typecaster to convert rclpy.Time to rclcpp::Time, enabling support for wait_for_current_robot_state() in Python as expected.

@MatthijsBurgh @peterdavidfagan would appreciate it if you could take a look when you have a moment. Also looping in @JafarAbdi and @tylerjw 

Thanks!

MatthijsBurgh, 2025-04-25:
Fixed by #3453, #3455

closed 2025-04-25 by MatthijsBurgh