moveit/moveit2 #2508: moveit_kinematics won't build with the trac_ik plugin.

marioprats, 2023-11-02:
### Description

`moveit_kinematics` won't build with the `trac_ik` plugin, with the following error message:

```
--- stderr: moveit_kinematics                                
In file included from /opt/underlay_ws/src/moveit2/moveit_kinematics/cached_ik_kinematics_plugin/src/cached_ik_kinematics_plugin.cpp:57:
/opt/underlay_ws/install/trac_ik_kinematics_plugin/include/trac_ik/trac_ik_kinematics_plugin.hpp:35:10: fatal error: trac_ik_kinematics_parameters.hpp: No such file or directory
   35 | #include <trac_ik_kinematics_parameters.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

The `trac_ik` parameters header can't be found.
This may be related to how the `cached_ik_kinematics_plugin` builds against `trac_ik` here: 
https://github.com/ros-planning/moveit2/blob/main/moveit_kinematics/cached_ik_kinematics_plugin/CMakeLists.txt#L13

It doesn't declare the dependency via the `package.xml` and `ament_target_dependencies`, since `trac_ik` is a plugin that may exist or not. Therefore it somehow doesn't get the correct include directories for the parameters library that `trac_ik` defines.

A workaround (maybe fix?) is to manually export the parameters include directory in the `trac_ik` [CMakeLists.txt](https://bitbucket.org/traclabs/trac_ik/src/rolling-devel/trac_ik_kinematics_plugin/CMakeLists.txt#lines-64):

```
ament_export_include_directories(include) include/${TRAC_IK_PARAM_LIB})
```

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build of MoveIt2 with `trac_ik`

### Steps to reproduce
1. Build the workspace for the first time. It should builld, because `trac_ik_kinematics_plugin_FOUND` evaluates to false in `cached_ik_kinematics_plugin`.
2. Touch a file that requires re-building `moveit_kinematics`, or simply remove `build/moveit_kinematics`.
3. `colcon build` the workspace again. This time `trac_ik_kinematics_plugin_FOUND` evaluates to true (had been built before). `moveit_kinematics` won't build, with the error posted above.

marioprats, 2023-11-08:
Exporting the parameters library include directory in the `trac_ik` package should fix this: https://bitbucket.org/traclabs/trac_ik/pull-requests/35

mlanighan, 2023-11-14:
Please check out this alternative fix https://bitbucket.org/traclabs/trac_ik/pull-requests/36

marioprats, 2023-11-14:
Verified that hiding the header fixes the problem too. Thanks Mike!

mlanighan, 2023-11-14:
Great, the fix has been merged in.

closed 2023-11-15 by marioprats