moveit/moveit2 #2509: rolling-tutorial docker: demo complains about "multiple (2) starting interfaces for joint" on startup

ANogin, 2023-11-03:
### Description

Demo gets a bunch of error messages on startup:

```
[ros2_control_node-5] [ERROR] [1698999053.336922238] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_1' - this is not supported!
[ros2_control_node-5] [ERROR] [1698999053.336955377] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_2' - this is not supported!
[ros2_control_node-5] [ERROR] [1698999053.336970867] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_3' - this is not supported!
[ros2_control_node-5] [ERROR] [1698999053.336984181] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_4' - this is not supported!
[ros2_control_node-5] [ERROR] [1698999053.336996897] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_5' - this is not supported!
[ros2_control_node-5] [ERROR] [1698999053.337009747] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_6' - this is not supported!
[ros2_control_node-5] [ERROR] [1698999053.337027436] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_7' - this is not supported!
[ros2_control_node-5] [ERROR] [1698999053.337057736] [resource_manager]: Component 'KortexMultiInterfaceHardware' did not accept new command resource combination: 
[ros2_control_node-5] [ERROR] [1698999053.337082146] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[spawner-7] [ERROR] [1698999053.338856148] [spawner_joint_trajectory_controller]: Failed to activate controller
[move_group-4] [ERROR] [1698999053.446898338] [move_group.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[ERROR] [spawner-7]: process has died [pid 61, exit code 1, cmd '/opt/ros/rolling/lib/controller_manager/spawner joint_trajectory_controller -c /controller_manager --ros-args'].
```

### Your environment
https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/main/.docker/docker-compose.yml with DOCKER_IMAGE=rolling-tutorial

### Steps to reproduce
`wget https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/main/.docker/docker-compose.yml && DOCKER_IMAGE=rolling-tutorial docker compose -f docker-compose.yml run --rm --name moveit2_container cpu ros2 launch moveit2_tutorials demo.launch.py rviz_config:=panda_moveit_config_demo_empty.rviz`

### Expected behaviour
No error messages

### Actual behaviour
Error messages

### Backtrace or Console output
https://gist.github.com/ANogin/0a3b864a904e2fd1b2993ab5d55caeeb

MRQ, 2023-11-10:
I see very similar console output at Ubuntu 22.04 LTS.

I think a central part is this:
> [ros2_control_node-5] [ERROR] [1698999053.337082146] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
> [spawner-7] [ERROR] [1698999053.338856148] [spawner_joint_trajectory_controller]: Failed to activate controller
> …
> [ERROR] [spawner-7]: process has died [pid 61, exit code 1, cmd '/opt/ros/rolling/lib/controller_manager/spawner joint_trajectory_controller -c /controller_manager --ros-args'].

And then I cannot execute a path in rvis.

MRQ, 2023-11-11:
duplicate of https://github.com/ros-controls/ros2_control/issues/1144

closed 2024-04-29 by github-actions[bot]