moveit/moveit2 #2516: Unable to build the ROS2 rolling

bgouda1982, 2023-11-08:
### Description

After following the instructions from https://moveit.ros.org/install-moveit2/source/, I attempted to build the moveit source but encountered build errors.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source or Binary build? Source
* If source, which branch? main
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone DDS

### Steps to reproduce

1. Build ROS rolling from source following the instructions - https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html).
2. Build MoveIt2 from source. git clone and add the repositories it shows is missing when the build fails. The ones added are angles, cpp_polyutils,eigen_stl_containers, generated_parameterlibrary, geometric_shapes, object_recognition_msgs,octomap_msgs,random_numbers, RSL, srdfdom
3. Build fails when it gets to building the moveit_core package looking for random.cpp in the rsl file. This appears to be related to the recent changes for issue - Use node logging in moveit_ros #2482.
4. Tried adding *find_package(rsl REQUIRED)*  and to the *ament_export_dependencies* in the moveit_core CMakeLists.txt  file but had no luck. The moveit_core is unable to find the RSL include files.

### Expected behaviour
Build should pass

### Actual behaviour
Build fails with errors

### Backtrace or Console output
https://gist.github.com/bgouda1982/724b0ea63a3e8332ed82f51ad46b5d86

Shobuj-Paul, 2023-11-09:
Did you try running vcstool and rosdep before building?

bgouda1982, 2023-11-09:
> Did you try running vcstool and rosdep before building?

I assume you are referring to the two commands that were mentioned in the MoveIt2 source installation.

for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y

The $ROS_DISTRO is confirmed to be rolling in my case. I have run them. The vcs import only brings in moveit_msgs and moveit_resources as defined in the moveit2.repos file.

I reran them. Here is the output:
bgouda@ROS-I:~/ws_moveit2/src$ echo $ROS_DISTRO
rolling
bgouda@ROS-I:~/ws_moveit2/src$ for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done
..
=== ./moveit_msgs (git) ===

Already on 'ros2'
Your branch is up to date with 'origin/ros2'.
=== ./moveit_resources (git) ===

Already on 'ros2'
Your branch is up to date with 'origin/ros2'.
bgouda@ROS-I:~/ws_moveit2/src$ rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
#All required rosdeps installed successfully
bgouda@ROS-I:~/ws_moveit2/src$

Shobuj-Paul, 2023-11-09:
> Build MoveIt2 from source. git clone and add the repositories it shows is missing when the build fails. The ones added are angles, cpp_polyutils,eigen_stl_containers, generated_parameterlibrary, geometric_shapes, object_recognition_msgs,octomap_msgs,random_numbers, RSL, srdfdom

You shouldn't have to clone all these packages, rosdep takes care of them.

> Tried adding find_package(rsl REQUIRED) and to the ament_export_dependencies in the moveit_core CMakeLists.txt file but had no luck. The moveit_core is unable to find the RSL include files.

One possible reason can be sourcing issues. Could you check if both your underlay and overlay workspace is properly sourced?

moriarty, 2023-11-15:
@bgouda1982 I believe you're right.  although I didn't fully setup my environment to test it.

PR #2532 should fix this

> 4\. Tried adding _find_package(rsl REQUIRED)_  and to the _ament_export_dependencies_ in the moveit_core CMakeLists.txt  file but had no luck. The moveit_core is unable to find the RSL include files.

colcon uses package.xml information to create the build dependency tree and decide the order it builds things... so when you added `find_package(rsl REQUIRED)`  it would still fail to find rsl because when rsl is not installed by rosdep and in the same workspace but not declared as a build dependency it *might* not yet be built by colcon.

And from your issue description the missing step was to add the depend in the package.xml

if @Shobuj-Paul tries to reproduce but has rsl installed in a underlaying colcon workspace then the `find_package(rsl REQUIRED)` will succeed, and the build will pass, also why the CI job passes because when rosdep is called other components of moveit2 depended on rsl already so rsl would be installed

PR #2532 (2023-11-15): add rsl depend to moveit_core
### Description

- This should fix #2516
- Several moveit2 packages already depend on rsl
- PR #2482 added a depend in moveit_core

This is only broken when building all of moveit2 deps in one colcon workspace And not using rosdep because colcon uses the package.xml and rsl might not have been built

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API chan [... cut]
commit f79a070e (2023-11-15): add rsl depend to moveit_core (#2532)

- This should fix #2516
- Several moveit2 packages already depend on rsl
- PR #2482 added a depend in moveit_core

This is only broken when building all of moveit2 deps in one colcon workspace
And not using rosdep because colcon uses the package.xml and rsl might not have been built
closed 2023-11-15 by sjahr