moveit/moveit2 #2518: Planning regressions when moving from Humble to Iron due to outdated version of `ompl`

Yadunund, 2023-11-09:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Iron]
* OS Version: Ubuntu 22.04
* Binary `2.8.0`
* CycloneDDS.

My application was relying on sampling based planners from `ompl`, specifically `LazyPRM` and `LazyPRMstar` and I noticed a lot of issues when switching to the Iron binaries for `moveit2` from the Humble ones. I think I've narrowed the problem down to the version of `ompl` binary distributed with Iron vs Humble. See result of `dpkg -l | grep ompl` below but `moveit_planners_ompl` in Iron is a version higher than that in Humble while `ompl` in Iron is older than that in Humble.

```bash
ros-humble-moveit-planners-ompl                    2.5.5-1jammy.20230920.002710            amd64        MoveIt interface to OMPL
ros-humble-ompl                                    1.6.0-1jammy.20230623.020419            amd64        OMPL is a free sampling-based motion planning library.
ros-iron-moveit-planners-ompl                      2.8.0-1jammy.20230912.121431            amd64        MoveIt interface to OMPL
ros-iron-ompl                                      1.5.2-4jammy.20230622.185404            amd64        OMPL is a free sampling-based motion planning library.

```

Specifically I'm seeing
* I can no longer save/load roadmap for persistent LazyPRM variants
* Planning not terminating within specified time bounds

Looking at the commit history in `ompl` it looks like some of these issue may have been fixed between `1.5.2` and `1.6.0`. I've [requested](https://github.com/ompl/ompl/issues/1109) for a new version to be bloomed for Iron upstream.

But I'm wondering if we can flag such issues early on by pinning a minimum version of `ompl` in this [find_package() call](https://github.com/ros-planning/moveit2/blob/e880259f5e863c127e793fc94c44ea3f4af56e26/moveit_planners/ompl/CMakeLists.txt#L16)?

koonpeng, 2023-11-10:
This is a common problem I see with most ros packages, e.g. `rclcpp` also does not check the version of `rcl` https://github.com/ros2/rclcpp/blob/9c098e544ecf191b7c61f63f7f4fac6f6449cedd/rclcpp/CMakeLists.txt#L11. I have seen cases where an older version of `rcl` causing problems in `rclcpp`.

The problem also goes down to `ament_cmake` as it does not perform proper semver checking according to https://docs.ros.org/en/iron/The-ROS2-Project/Contributing/Developer-Guide.html#semver. This is the contents of `rclConfig-version.cmake`
```cmake
# generated from ament/cmake/core/templates/nameConfig-version.cmake.in
set(PACKAGE_VERSION "6.0.3")

set(PACKAGE_VERSION_EXACT False)
set(PACKAGE_VERSION_COMPATIBLE False)

if("${PACKAGE_FIND_VERSION}" VERSION_EQUAL "${PACKAGE_VERSION}")
  set(PACKAGE_VERSION_EXACT True)
  set(PACKAGE_VERSION_COMPATIBLE True)
endif()

if("${PACKAGE_FIND_VERSION}" VERSION_LESS "${PACKAGE_VERSION}")
  set(PACKAGE_VERSION_COMPATIBLE True)
endif()
```
It only checks if requested version is less than package version, ignoring major version incompatibilities. There should be no issues for core packages as there should never be a major version bump within the distro, but that can't be said for third party packages.

This also doesn't fully solve the issue as `ompl` also is not checking the version of it's dependencies, even if we checked that `ompl` is the correct version, if one of `ompl`'s dep has a wrong version then we may still see issues.

Yadunund, 2024-01-15:
@sjahr thanks for the update. I've been trying to ping the maintainers of ompl with no avail.