moveit/moveit2 #2523: Not seeing Kinova robot in Rviz when following tutorial

dsandber, 2023-11-10:
### Description

I'm following the tutorial here:

https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html

When doing: ros2 launch moveit2_tutorials demo.launch.py

rviz came up, and the "MotionPlanning" extension was already added even though the tutorial says I need to add it.  However, the robot from the tutorial didn't show up.  I tried removing MotionPlanning and adding it again but the robot still didn't show up.  I've included the program output.

Overview of your issue here.

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04.3
* The ROS was a binary build, for MoveIt I followed the instructions here: https://moveit.picknik.ai/main/doc/tutorials/getting_started

### Backtrace or Console output

[log.txt](https://github.com/ros-planning/moveit2/files/13322678/log.txt)

MRQ, 2023-11-10:
Do you use a locale, that has comma as a decimal point? Then try this:

```
LC_NUMERIC=en_US.UTF-8 ros2 launch moveit2_tutorials demo.launch.py
```

Sadly this is know, but not mentioned in the tutorial.

dsandber, 2023-11-14:
Yup, thanks, that fixed it!  Should I submit a patch to the setup page to include this?

MRQ, 2023-11-14:
I would be happy if you work to get this in the tutorial.

---
Either this is a recurring bug, or nothing improved within he last year: https://github.com/ros-planning/moveit2_tutorials/issues/508

moriarty, 2023-11-21:
I just assigned @tylerjw because the issue linked above had tagged him before https://github.com/ros-planning/moveit2_tutorials/issues/508#issuecomment-1239440054 and he's familiar with the params

henningkayser, 2023-11-21:
We have tracked and supposedly fixed this issue before: https://github.com/ros-planning/moveit2/issues/1882

dsandber, 2023-11-28:
I looked for the part of the documentation that listed a bunch of environment variables to set, but I couldn't find it even though I remember following it.  Should I suggest a change to the ros.bash that you're supposed to source?

henningkayser, 2024-01-16:
I'm closing this since ROS documentation recommends setting locale support to en_US.UTF-8. We are still tracking the very same problem in #1882, so feel free to follow up or contribute from there.

closed 2024-01-16 by henningkayser

moriarty, 2024-01-16:
Linking to ROS documentation https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html#set-locale