moveit/moveit2 #2524: Moveit Python bindings for Humble?

sl148, 2023-11-11:
I am running ROS2 Humble with Moveit2 to control the UR5e manipulator.
I wonder if there an official Python bindings that are compatible with ROS2 Humble.

I tried source installation in the main branch instead of the humble branch, but it seems like Moveit2 fails to plan the path when I do the motion plan in Rviz. I suspect that the error comes from some difference between the main and the humble branches since motion planning was working fine when I followed the Humble binary installation.
So I wonder if there is a stable version of moveit_py for Humble.

Thanks!

JensVanhooydonck, 2023-11-29:
You should be able to use the source of the main branch to use the python bindings.
Could you check the versions of ompl on the ROS2 Humble instance vs the source instance? I've seen that an older OMPL-version was installed in some cases.

gagapaga, 2023-11-29:
Thanks for your reply. I have use the binary build from main branch. Is
source build the only possibility to use time parameterization function?

The version of ompl is:
ros-humble-moveit-planners-ompl/now 2.7.4-42-g92ca89dca-2023.08.10.03.10
amd64 [installed,local] ros-humble-ompl/now 1.6.0-1jammy.20230623.020419
amd64 [installed,local]

JensVanhooydonck, 2023-11-29:
I don't think the binary build has the latest changes. Can you try the source build?
If you want to use the Python interface, the time parameterisation function was added only a few weeks ago. So i think currently only in source build.

gagapaga, 2023-11-29:
Because we are doing a project which run the script in a docker container.
We don’t plan to change the way of build. But thanks a lot for the reply
and help. Can I ask when will be the next binary version released which
includes these changes?

Thanks again

tylerjw, 2023-11-29:
Those improvements are unlikely to ever be backported to humble. However the main branch will continue to build on Humble if you'd like the latest features.

closed 2024-03-01 by github-actions[bot]

KouroshJolaei, 2025-06-14:
I want to use the moveit_py in my Python pipeline to get the IK, FK, Jacobian matrix, planning the UR5e robot in Gazebo and a real robot, but I do not know where to clone the moveit_py from?
I'd appreciate any help in this matter. 
Thanks