moveit/moveit2 #2525: Uncaught 'std::out_of_range' upon instantiating MoveItPy node

madgrizzle, 2023-11-11:
### Description

I've been previously using rolling on my system but have been forced to try to switch back to humble for compatibility with a package that is only available on humble.  I have built moveit from the main branch using the instructions but when I attempt to use the code that I previously used with rolling to instantiate a MoveItPy node, it fails with:

> [floyd_cognitive-1]  terminate called after throwing an instance of 'std::out_of_range'
[floyd_cognitive-1]   what():  map::at

I don't know how to isolate where this is occurring or what might be causing it.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build
* main branch
* Fast DDS as I'm unfortunately forced to use it for compatibility with micro-ros.

### Steps to reproduce

This is the code I used to start the MoveItPy node in my application:
```
        self.moveit_config = (
                MoveItConfigsBuilder(robot_name="floyd", package_name="floyd_moveit_config")
                .robot_description(file_path="config/robot_model.urdf.xacro")
                .trajectory_execution(file_path="config/moveit_controllers.yaml")
                .robot_description_kinematics(file_path="config/kinematics.yaml")
                .joint_limits(file_path="config/joint_limits.yaml")
                .planning_scene_monitor(
                    publish_robot_description=True, publish_robot_description_semantic=True)
                .moveit_cpp(file_path="config/floyd_right_arm_planning.yaml")
                .to_moveit_configs()
            ).to_dict()
        self.floyd = MoveItPy(node_name="moveit_py", config_dict=self.moveit_config)
```
### Expected behaviour
Not crash.

### Actual behaviour
Crash.

### Backtrace or Console output

Working on figuring out if it's possible to get gdb output of this.  Never used it before and this node is loaded via a python node so not sure.

madgrizzle, 2023-11-12:
When I load the demo.launch.py from my floyd_moveit_config package then I get this error message with all the debug info.  It suggests maybe its related to the meshes I use in my robot description package?  Interestingly, if I source my rolling binary install (/opt/ros/rolling/setup.bash) after sourcing the humble-built moveit workspace I do not get this error when I try to run demo.launch.py

> [move_group-2] terminate called after throwing an instance of 'std::out_of_range'
> [move_group-2]   what():  map::at
> [move_group-2] Stack trace (most recent call last) in thread 12841:
> [move_group-2] #30   Object "", at 0xffffffffffffffff, in 
> [move_group-2] #29   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f5b8a0afa3f]
> [move_group-2] #28   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f5b8a01dac2]
> [move_group-2] #27   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f5b8a2af252, in 
> [move_group-2] #26   Source "/usr/include/c++/11/bits/std_thread.h", line 211, in _M_run [0x7f5b7cd3cea9]
> [move_group-2]         208: 	  { }
> [move_group-2]         209: 
> [move_group-2]         210: 	void
> [move_group-2]       > 211: 	_M_run() { _M_func(); }
> [move_group-2]         212:       };
> [move_group-2]         213: 
> [move_group-2]         214:     void
> ...
> [move_group-2] #9    Source "/home/john/moveit_ws/src/moveit2/moveit_ros/perception/mesh_filter/src/gl_renderer.cpp", line 390, in createGLContext [0x7f5b7cd4ac90]
> [move_group-2]         388:   if (context_it == s_context.end())
> [move_group-2]         389:   {
> [move_group-2]       > 390:     s_context.at(thread_id) = std::pair<unsigned, GLuint>(1, 0);
> [move_group-2]         391: 
> [move_group-2]         392:     glutInitWindowPosition(glutGet(GLUT_SCREEN_WIDTH) + 30000, 0);
> [move_group-2]         393:     glutInitWindowSize(1, 1);
> [move_group-2] #8    Source "/usr/include/c++/11/bits/stl_map.h", line 541, in at [0x7f5b7cd4b629]
> [move_group-2]         538:       {
> [move_group-2]         539: 	iterator __i = lower_bound(__k);
> [move_group-2]         540: 	if (__i == end() || key_comp()(__k, (*__i).first))
> [move_group-2]       > 541: 	  __throw_out_of_range(__N("map::at"));
> [move_group-2]         542: 	return (*__i).second;
> [move_group-2]         543:       }
> [move_group-2] #7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f5b8a27849f, in std::__throw_out_of_range(char const*)
> [move_group-2] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f5b8a2814d7, in __cxa_throw
> [move_group-2] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f5b8a281276, in std::terminate()
> [move_group-2] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f5b8a28120b, in 
> [move_group-2] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f5b8a275b9d, in 
> [move_group-2] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f5b89fb17f2]
> [move_group-2] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f5b89fcb475]
> [move_group-2] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
> [move_group-2]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
> [move_group-2]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f5b8a01f9fc]
> [move_group-2] Aborted (Signal sent by tkill() 12698 1000)
> [ERROR] [move_group-2]: process has died [pid 12698, exit code -6, cmd '/home/john/moveit_ws/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_ripcyew4 --params-file /tmp/launch_params_ra2qk_cu'].
>

madgrizzle, 2023-11-14:
I switched the urdf to just use cylinders, no meshes, and it still fails.  Turns out the version I had been using with rolling actually does work with humble, but whatever changes that has been made to the main branch since I had last updated it causes my issue.  I'm trying to revert back to the last release (2.8.0 on 9/10) to see if it works there and then try to progress through commits and see which one causes it to fail.

henningkayser, 2024-01-16:
@madgrizzle any new learnings here? There have been several additions to the main branch which didn't make it into humble yet. If you could help tracking down the source of this, that would be really appreciated.

Ep3896, 2024-04-19:
Any updates? I'm facing the same error

matthew-william-lock, 2024-06-07:
Same issue

closed 2025-01-08 by github-actions[bot]