moveit/moveit2 #2533: Python binding of apply_totg_time_paramaterization() in moveit.core.robot_trajectory.RobotTrajectory

gagapaga, 2023-11-15:
### Description

I want to implement time parameterization for waypoints using apply_totg_time_paramaterization() function in moveit.core.robot_trajectory.RobotTrajectory class. But always an error "moveit.core.robot_trajectory.RobotTrajectory' object has no attribute 'apply_totg_time_parameterization'. I just saw there is a new commit about it: https://github.com/ros-planning/moveit2/pull/2527/commits/23ff3751d8bf2941c6c4b6766b9b0989f2af12db   So I changed the local file of robot_trajectory.pyi manually. But the error still appears. Does anyone know how can I solve this problem

### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04
*Binary build
* If binary, which release version:  https://raw.githubusercontent.com/moveit/moveit2_packages/jammy-humble/ 
     https://raw.githubusercontent.com/moveit/moveit2_packages/jammy-humble/local.yaml    (maybe the latest version of humble)
* Cyclone DDS

### Expected behaviour
Is there any possibility that I can use apply_totg_time_paramaterization() in python?

### Actual behaviour
ERROR: 'moveit.core.robot_trajectory.RobotTrajectory' object has no attribute 'apply_totg_time_parameterization'

gagapaga, 2023-11-15:
And when will be the next version for the binary build in humble which may include this issue

JensVanhooydonck, 2023-11-21:
Can you provide a snippet of your code calling the function? 
This works for me. 

```python
def apply_totg_parametrization(self, path_msg):
    robot_traject_cpp = RobotTrajectory(self.robot.get_robot_model())
    robot_state = get_current_robot_state(self.planning_scene_monitor)
    robot_traject_cpp.set_robot_trajectory_msg(robot_state, path_msg)
    robot_traject_cpp.joint_model_group_name = self.prefix + self.robot_name
        
    robot_traject_cpp.apply_totg_time_parameterization(velocity, acceleration)
    return robot_traject_cpp
```

closed 2024-01-16 by henningkayser