moveit/moveit2 #2535: Mobile Base and an Arm tutorial – stretch ros: Integration with real controller of the mobile base

lmendyk, 2023-11-16:
I’m struggling in adapting the demo pointed out by [mobile_base_arm_tutorial](https://moveit.picknik.ai/humble/doc/examples/mobile_base_arm/mobile_base_arm_tutorial.html)  for  mobile base and arm control  provided by [streach-ros](https://github.com/PickNikRobotics/stretch_ros/tree/ros2/stretch_moveit_config/config) for 
 [turtlebot3_manipulation](https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation).

The demo  (the config files in stretch_moveit_config/config) assume fake controller for mobile base which defines “position” as a joint name rather than “position” to be translated to control commands to wheels of the mobile base.  IN the “moveit_simple_controllers.yaml” the “position” is listed as a joint name together with joints like “joint_arm_l3” etc. which as I understand is … fake/oversimplification.

Could you advice how it should be done to integrate with the real diff_drive_controller. 
From moveIt documentation the moveit supports as MoveIt Controller Interfaces just: FollowJointTrajectory and GripperCommand controller Interfaces (which I currently have configured for Turtlebot3 manipulation arm) but there is nothing related to diff_drive_controller.

I even wonder if diff_drive_controller can be directly used here as it accepts Twist (via topic not action)  not position (x, y) so may be the higher level controller provided by Nav2 is needed to have the postion (x,y) drive the controller.

Could anybody enlighten me and show the direction I should go.

FrGe2016, 2024-02-19:
Did you achevied any result, i also struggle mo make themo work with a real modified linorobot2 with an homemade arm.

henningkayser, 2024-02-20:
I'm afraid that the configs from the tutorial are out of date. Especially, fake controller (or ros2_control mock component) support is not provided with the currently released versions. The best bet is to reproduce the stretch mobile manipulation demo from 2021 ROS world https://moveit.ros.org/events/rosworld-2021-workshop/.

FrGe2016, 2024-02-22:
very interesting. I am setting up the docker to look at it in details

lmendyk, 2024-04-17:
> Did you achevied any result ...

No and Yes .  I mean I have given up trying using the approach presented by [mobile_base_arm_tutorial](https://moveit.picknik.ai/humble/doc/examples/mobile_base_arm/mobile_base_arm_tutorial.html). Instead I used an reverse concept.  I mean I used nav2 and the behavior tree concept to invoke the action of robot arm planning by MoveIt stack once the robot has navigated to the working area for the robot arm.  In other words first the robot mobile base is moved to the working area and than just robot arm performs its action.  This does not provide exactly the same solution as in this "workaround" it is assumed that robot mobile base must go to the place where the object to be picked up must be in reach of the robot arm, but ... works.

FrGe2016, 2024-04-17:
i also had some result making few important changes. First of all the mobile part was a stand alone robot already accepting twist messages then i did not have to care about the diff drive controller. I replaced the the odom frame for the map frame
For the planar joint in the srdf  and in the stretch_ignition_control_action_server. With this setup, i can have simultaneous motions of the arm and the base. I have a remaining unsolved issue, the plan generated by moveit2 avoid obstacles in 3D, taking in account all the constraint of my complex environment, but when the plan is passed to the action server for execution, only the detailled waypoints are not passed to nav2 , only the final destinations ( not respecting the 3d restriction on the way to that destination)

floridatechnologies, 2024-08-20:
> No and Yes . I mean I have given up trying using the approach presented by [mobile_base_arm_tutorial](https://moveit.picknik.ai/humble/doc/examples/mobile_base_arm/mobile_base_arm_tutorial.html). [...]

Can you share your integration with NAV2

EzraBrooks, 2024-11-01:
(just removing the bug label as part of some project management housekeeping after having changed the Issue Type to "bug", don't worry)