moveit/moveit2 #2539: Add field to PR template to ask if contribution contains software created by generative AI

SteveMacenski, 2023-11-20:
This is to ask the question to prompt questions should software generated by ChatGPT or similar is contained within the pull request

tylerjw, 2023-11-21:
Do people ever read those, I find most people either leave them alone un-modified or erase them to write something more helpful? I've been considering removing the pull request template entirely or replacing it with something much shorter people might read. We could put something in our contribution guide as a CYA measure (not that anyone would read that document).

SteveMacenski, 2023-11-21:
In my experience, people read and fill them out. On the occasion someone doesn’t I mention them to do it and withhold a review until its there. I won’t say my detector on it is perfect, but I try 🙃