moveit/moveit2 #2541: [PSM] When using an ACM, Diff planning scene messages are not possible and always trigger a full scene update

JensVanhooydonck, 2023-11-21:
### Description

When using the Planning Scene Monitor with a planning scene with an Allowed Collision Matrix a full planning scene is published everytime. This results in an inifite loop when trying to sync multiple PSM. 

### Your environment
* ROS Distro: Rolling
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? Master
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone & Fast DDS.

### Steps to reproduce
Use two PSM which subscribe to each other and use an ACM.

### Expected behaviour
When using an ACM, only when something changes this should be send in the PlanningScene Diff. This could also be send when using the UPDATE_GEOMETRY instead of triggering a full UPDATE_SCENE.

### Actual behaviour
When using an ACM a full planning scene is published everytime. When trying to sync multiple PSM, this results in infinite loops because a full planning scene keeps triggering an full UPDATE_SCENE instead of eventualy going to UPDATE_NONE because there are no changes anymore.

JensVanhooydonck, 2024-01-08:
This issue is still relevant

henningkayser, 2024-01-16:
@JensVanhooydonck do you have any idea where this could be coming from? I'm seeing this for the first time. Also, what is your current PSM syncing setup?

JensVanhooydonck, 2024-01-29:
Currently I have one Master, which subscribes to the other (currently 2 other PSM), and this one republishes the changes again where the other 2 PSM are subscribing to.

However, if the ACM is updated, this triggers a full update, which in it's turn retriggers full updates in the other PSM.
If the ACM does not trigger a full update, this will not trigger a infitiny loop of updates.

Currently this happens here:
Here is the complete ACM assigned to the diff-Msg: https://github.com/ros-planning/moveit2/blob/a8cc2794370d53de579bea1212d341fe6e5541af/moveit_core/planning_scene/src/planning_scene.cpp#L621 Also when there are no changes between the current Scene ACM and the incoming scene ACM.

Here this ACM is seen as a complete Scene message when there is at least one entry in the ACM:
https://github.com/ros-planning/moveit2/blob/a8cc2794370d53de579bea1212d341fe6e5541af/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L760