moveit/moveit2 #2545: Cartesian Path

mhubii, 2023-11-21:
### Description
Hi, we are experiencing MoveIt crashes upon requesting a cartesian path. Any change this error is MoveIt related? 

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Binaries
* Latest
* Default

### Steps to reproduce
Launch `lbr`, see https://github.com/lbr-stack/lbr_fri_ros2_stack, and request cartesian path in RViZ.

### Expected behaviour
Executes motion.

### Actual behaviour
RViZ crashes.

### Backtrace or Console output
```shell
[INFO] [launch]: All log files can be found below /home/martin/.ros/log/2023-11-21-17-40-27-063900-mars-87332
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [87335]
[INFO] [gzclient-2]: process started with pid [87337]
[INFO] [spawn_entity.py-3]: process started with pid [87339]
[INFO] [spawner-4]: process started with pid [87341]
[INFO] [robot_state_publisher-5]: process started with pid [87343]
[INFO] [spawner-6]: process started with pid [87345]
[INFO] [static_transform_publisher-7]: process started with pid [87347]
[INFO] [static_transform_publisher-8]: process started with pid [87349]
[INFO] [move_group-9]: process started with pid [87351]
[static_transform_publisher-7] [ERROR] [1700588428.675699153] [static_transform_publisher_aHZLfbfmYiEolZEw]: cannot publish static transform from 'world' to 'world', exiting
[static_transform_publisher-7] terminate called after throwing an instance of 'std::runtime_error'
[static_transform_publisher-7]   what():  child_frame_id cannot equal frame_id
[static_transform_publisher-8] [INFO] [1700588428.681614523] [static_transform_publisher_Glz3qMIPsMGie9XE]: Spinning until stopped - publishing transform
[static_transform_publisher-8] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-8] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-8] from 'world' to 'lbr/world'
[robot_state_publisher-5] [INFO] [1700588428.714649331] [lbr.robot_state_publisher]: got segment link_0
[robot_state_publisher-5] [INFO] [1700588428.714755287] [lbr.robot_state_publisher]: got segment link_1
[robot_state_publisher-5] [INFO] [1700588428.714767128] [lbr.robot_state_publisher]: got segment link_2
[robot_state_publisher-5] [INFO] [1700588428.714773202] [lbr.robot_state_publisher]: got segment link_3
[robot_state_publisher-5] [INFO] [1700588428.714778497] [lbr.robot_state_publisher]: got segment link_4
[robot_state_publisher-5] [INFO] [1700588428.714783804] [lbr.robot_state_publisher]: got segment link_5
[robot_state_publisher-5] [INFO] [1700588428.714788894] [lbr.robot_state_publisher]: got segment link_6
[robot_state_publisher-5] [INFO] [1700588428.714794267] [lbr.robot_state_publisher]: got segment link_7
[robot_state_publisher-5] [INFO] [1700588428.714799380] [lbr.robot_state_publisher]: got segment link_ee
[robot_state_publisher-5] [INFO] [1700588428.714804524] [lbr.robot_state_publisher]: got segment world
[move_group-9] [INFO] [1700588428.794781133] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-9] [INFO] [1700588428.795636603] [moveit_robot_model.robot_model]: Loading robot model 'med7'...
[move_group-9] [INFO] [1700588428.796320079] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[spawn_entity.py-3] [INFO] [1700588429.376594657] [lbr.spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1700588429.376940866] [lbr.spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[spawn_entity.py-3] [INFO] [1700588429.378749438] [lbr.spawn_entity]: Waiting for entity xml on robot_description
[move_group-9] [INFO] [1700588429.386620998] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-9] [INFO] [1700588429.386728202] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-9] [INFO] [1700588429.387195029] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-9] [INFO] [1700588429.387485667] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/attached_collision_object' for attached collision objects
[move_group-9] [INFO] [1700588429.387501688] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-9] [INFO] [1700588429.387771413] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/planning_scene'
[move_group-9] [INFO] [1700588429.387784724] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-9] [INFO] [1700588429.388052467] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-9] [INFO] [1700588429.388332932] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-9] [WARN] [1700588429.389114919] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-9] [ERROR] [1700588429.389129867] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[spawn_entity.py-3] [INFO] [1700588429.390353827] [lbr.spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1700588429.390815377] [lbr.spawn_entity]: Waiting for service /spawn_entity
[... 189 lines cut ...]
[move_group-9] [INFO] [1700588440.364938121] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-9] [INFO] [1700588440.365026036] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to position_trajectory_controller
[gzserver-1] [INFO] [1700588440.365671671] [lbr.position_trajectory_controller]: Received new action goal
[gzserver-1] [INFO] [1700588440.365772107] [lbr.position_trajectory_controller]: Accepted new action goal
[move_group-9] [INFO] [1700588440.365901980] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: position_trajectory_controller started execution
[move_group-9] [INFO] [1700588440.365919938] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[gzserver-1] [INFO] [1700588451.905913307] [lbr.position_trajectory_controller]: Goal reached, success!
[move_group-9] [INFO] [1700588451.916655681] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'position_trajectory_controller' successfully finished
[move_group-9] [INFO] [1700588451.916717733] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-9] [INFO] [1700588451.925042424] [moveit_move_group_default_capabilities.move_action_capability]: Solution was found and executed.
[rviz2-10] [INFO] [1700588451.925550923] [move_group_interface]: Plan and Execute request complete!
[rviz2-10] [WARN] [1700588452.603011491] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Maybe failed to update robot state, time diff: 1700588429.873s
[rviz2-10] [WARN] [1700588458.965344635] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Maybe failed to update robot state, time diff: 1700588429.756s
[move_group-9] [INFO] [1700588464.467679341] [moveit_move_group_default_capabilities.cartersian_path_service_capability]: Received request to compute Cartesian path
[move_group-9] [INFO] [1700588464.467740964] [moveit_move_group_default_capabilities.cartersian_path_service_capability]: Attempting to follow 1 waypoints for link 'link_ee' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[move_group-9] [INFO] [1700588464.469081227] [moveit_move_group_default_capabilities.cartersian_path_service_capability]: Computed Cartesian path with 15 points (followed 100.000000% of requested trajectory)
[... cut ...] rgs -r __node:=rviz2 --params-file /tmp/launch_params_xq5582eh --params-file /tmp/launch_params_485c98u1 -r robot_description:=lbr/robot_description -r robot_description_semantic:=lbr/robot_description_semantic -r display_planned_path:=lbr/display_planned_path -r monitored_planning_scene:=lbr/monitored_planning_scene'].
```

Refers to https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/133#issuecomment-1820485008 and [#lbr-stack#133](https://github.com/lbr-stack/lbr_fri_ros2_stack/issues/133)

Abishalini, 2023-11-21:
I think the relevant error is 
```
[static_transform_publisher-7] [ERROR] [1700588428.675699153] [static_transform_publisher_aHZLfbfmYiEolZEw]: cannot publish static transform from 'world' to 'world', exiting
[static_transform_publisher-7] terminate called after throwing an instance of 'std::runtime_error'
[static_transform_publisher-7]   what():  child_frame_id cannot equal frame_id
[static_transform_publisher-8] [INFO] [1700588428.681614523] [static_transform_publisher_Glz3qMIPsMGie9XE]: Spinning until stopped - publishing transform
[static_transform_publisher-8] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-8] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-8] from 'world' to 'lbr/world'
```
It could be coming from [here](https://github.com/lbr-stack/lbr_fri_ros2_stack/blob/688c8bac9cdea496f55db394a32054f99a04504b/lbr_bringup/launch/sim.launch.py#L49-L52)

mhubii, 2023-11-21:
Hi @Abishalini , wow that was quick ! Good catch, thank you for the hint. Will give that a go

mhubii, 2023-11-22:
I know these static tf publishers in the launch are a little confusing. They were necessary ? for namespacing the Move Group. But the issue persists.

Please find below an updated terminal log:

```shell
ros2 launch lbr_bringup bringup.launch.py model:=med7 sim:=true moveit:=true
[INFO] [launch]: All log files can be found below /home/martin/.ros/log/2023-11-22-10-50-55-573807-mars-36707
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [36709]
[INFO] [gzclient-2]: process started with pid [36711]
[INFO] [spawn_entity.py-3]: process started with pid [36713]
[INFO] [spawner-4]: process started with pid [36715]
[INFO] [robot_state_publisher-5]: process started with pid [36717]
[INFO] [spawner-6]: process started with pid [36719]
[INFO] [static_transform_publisher-7]: process started with pid [36721]
[INFO] [static_transform_publisher-8]: process started with pid [36723]
[INFO] [move_group-9]: process started with pid [36725]
[static_transform_publisher-7] [INFO] [1700650257.690233996] [static_transform_publisher_XzZvHgYPxYjZZpbw]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-7] from 'world' to 'base_frame'
[static_transform_publisher-8] [INFO] [1700650257.690239125] [static_transform_publisher_AA4RIYbKtJhi15Hs]: Spinning until stopped - publishing transform
[static_transform_publisher-8] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-8] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-8] from 'base_frame' to 'lbr/base_frame'
[robot_state_publisher-5] [INFO] [1700650257.765112590] [lbr.robot_state_publisher]: got segment base_frame
[robot_state_publisher-5] [INFO] [1700650257.765250636] [lbr.robot_state_publisher]: got segment link_0
[robot_state_publisher-5] [INFO] [1700650257.765267467] [lbr.robot_state_publisher]: got segment link_1
[robot_state_publisher-5] [INFO] [1700650257.765278523] [lbr.robot_state_publisher]: got segment link_2
[robot_state_publisher-5] [INFO] [1700650257.765288889] [lbr.robot_state_publisher]: got segment link_3
[robot_state_publisher-5] [INFO] [1700650257.765299204] [lbr.robot_state_publisher]: got segment link_4
[robot_state_publisher-5] [INFO] [1700650257.765309248] [lbr.robot_state_publisher]: got segment link_5
[robot_state_publisher-5] [INFO] [1700650257.765319448] [lbr.robot_state_publisher]: got segment link_6
[robot_state_publisher-5] [INFO] [1700650257.765329515] [lbr.robot_state_publisher]: got segment link_7
[robot_state_publisher-5] [INFO] [1700650257.765339611] [lbr.robot_state_publisher]: got segment link_ee
[move_group-9] [INFO] [1700650257.829221414] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-9] [INFO] [1700650257.829289268] [moveit_robot_model.robot_model]: Loading robot model 'med7'...
[move_group-9] [INFO] [1700650257.829301457] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[spawn_entity.py-3] [INFO] [1700650258.408681458] [lbr.spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1700650258.409021350] [lbr.spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[spawn_entity.py-3] [INFO] [1700650258.413879021] [lbr.spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-3] [INFO] [1700650258.425611678] [lbr.spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1700650258.425923825] [lbr.spawn_entity]: Waiting for service /spawn_entity
[move_group-9] [INFO] [1700650258.434846625] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-9] [INFO] [1700650258.434963128] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-9] [INFO] [1700650258.435409055] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-9] [INFO] [1700650258.435706088] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/attached_collision_object' for attached collision objects
[move_group-9] [INFO] [1700650258.435721645] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-9] [INFO] [1700650258.435962960] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/lbr/planning_scene'
[move_group-9] [INFO] [1700650258.435976459] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-9] [INFO] [1700650258.436189281] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-9] [INFO] [1700650258.436410892] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[... cut ...]
[ERROR] [rviz2-10]: process has died [pid 36988, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/martin/Dev/lbr-stack/install/med7_moveit_config/share/med7_moveit_config/config/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_uommc4o1 --params-file /tmp/launch_params_k06pcy8c -r robot_description:=lbr/robot_description -r robot_description_semantic:=lbr/robot_description_semantic -r display_planned_path:=lbr/display_planned_path -r monitored_planning_scene:=lbr/monitored_planning_scene'].
```

josefinemonnet, 2023-12-24:
I think I solved the problem by adding these two lines:

1. ("joint_states", robot_name + "/joint_states"), into sim.launch.py:

# RViz and MoveIt
rviz_moveit = RVizMixin.node_rviz(
rviz_config_pkg=f"{model}_moveit_config",
rviz_config="config/moveit.rviz",
parameters=LBRMoveGroupMixin.params_rviz(
moveit_configs=moveit_configs_builder.to_moveit_configs()
),
condition=IfCondition(
AndSubstitution(LaunchConfiguration("moveit"), LaunchConfiguration("rviz"))
),
remappings=[
("robot_description", robot_name + "/robot_description"),
("robot_description_semantic", robot_name + "/robot_description_semantic"),
("display_planned_path", robot_name + "/display_planned_path"),
("monitored_planning_scene", robot_name + "/monitored_planning_scene"),
("joint_states", robot_name + "/joint_states"),
],
)

and into  {"use_sim_time": True} into launch_mixin.py

@staticmethod
def params_rviz(
  moveit_configs: MoveItConfigs,
) -> List[Dict[str, Any]]:
  return [
      moveit_configs.planning_pipelines,
      moveit_configs.robot_description_kinematics,
      {"use_sim_time": True},
  ]

now the cartesian path is computed and executed in rviz.

mhubii, 2023-12-24:
thanks for sharing this @josefinemonnet.

So is it that there is no namespace support for Cartesian path in moveit ?

Abishalini, 2024-02-01:
Thanks @josefinemonnet for the fix. 
I'm closing this issue. Please open another issue if you feel there should be a fix to MoveIt.

closed 2024-02-01 by Abishalini