moveit/moveit2 #2550: Using STOMP in humble

leander2189, 2023-11-24:
I see that STOMP planner is added in Iron, but not in humble.

If I wanted to use in humble, would it be as easy as coping the code from moveit_planners into my local source (humble)? are there any other nuances that I should be aware of?

closed 2024-02-23 by github-actions[bot]

angeelalg, 2024-04-16:
I am in a similar situation, I would like to be able to use STOMP in Foxy. By modifying moveit_planners could it work directly or does it require modifications in other parts of the repository?

antonio1matos, 2024-06-04:
@leander2189  were you able to setup the Stomp planner in moveit humble?

leander2189, 2024-06-11:
> @leander2189 were you able to setup the Stomp planner in moveit humble?

I downloaded moveit2 main and built from source. This branch is compatible with humble, although you have to change some of the motion planning configuration files

antonio1matos, 2024-06-11:
Hello @leander2189. What i did was: In my ws_moveit2 workspace i went to the package: moveit2  (path: /home/antonio/ws_moveit2/src) and i replaced this package by the moveit2 package provided by: https://github.com/Jakubivan/moveit2/tree/ivan/humble-w-stomp                                                 I made colcon build of the workspace ws_moveit2 and i added the file: stomp_planning.yaml to the moveit_config of the robot that i wanted to do the stomp planning. I tried to execute the robotic arm in rviz and even though the robot executes the trajectory with planner stomp i don't think it is actually using it, once in the terminal appears this: [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'stomp'
[move_group-10] Warning: Interrupting STOMP
[move_group-10]          at line 408 in ./src/stomp.cpp                                                          I think that the process that i made to install stomp was not the best way. Also if i include a colliison object in the planning scene the planner stomp isn't able to find a solution to the trajectory. Do you also have that problem or your stomp planner is working correctly with moveit humble?
Regards