moveit/moveit2 #2557: Moveit2 plans trajectories that violate joint position limits

Tejas-Shastha, 2023-11-28:
### Description

It seems that Moveit2 plans trajectories that ignore the joint position limits that were declared, leading the FER1 robot to go into a reflex stop due to joint limit violation. 

I set the joint limits in a YAML file and load it in the launch. I can verify these values are actually being loaded in, as I can even print them to console through the C++ API.

Complete param dump of the `move_group` node: https://gist.github.com/Tejas-Shastha/6612879f539f64947abc8d9ab1a738bb

How I verify that these limits are actually loaded (i.e. it prints what I load as params): 
```C++
  const auto models = move_group_interface_->getRobotModel()->getJointModelGroups();
  for(auto const &model : models)
  {
    RCLCPP_INFO(logger_, "------------------------------------------------");
    RCLCPP_INFO_STREAM(logger_, "Model group: " << model->getName());
    const auto joint_models = model->getActiveJointModels();
    for (auto const &joint_model : joint_models)
    {
      auto bounds = joint_model->getVariableBounds();
      RCLCPP_INFO_STREAM(logger_, "--------------------- joint model: " << joint_model->getName());
      for(auto const & bound : bounds)
      {
        RCLCPP_INFO_STREAM(logger_, "position bounds: [" << bound.min_position_ << ", " << bound.max_position_<< "]");
      }
    }
  }
  ```

With these params loaded, when I **sometimes** (i.e. sporadic issue) execute joint space plans that were created using the functions `MoveGroupInterface::setPoseTarget()`,  `MoveGroupInterface::plan()` and `MoveGroupInterface::execute()`, it leads joint limit violations. One example was `panda_link6` reaching a value of `-0.03548325803408056` whereas limits were `(2.61, -0.0175)`.

Related issue with more details in [franka_ros2](https://github.com/frankaemika/franka_ros2/issues/43).

I post this as a bug report here since I believe I have configured my setup correctly and is indeed a moveit bug, but please correct me if I am wrong.

### Your environment
* ROS Distro: [Humble]
* OS Version: Ubuntu 22.04
* Using Binary build
* Binary versions: https://gist.github.com/Tejas-Shastha/a6859cb69c8c7e0b73494f06f7d2fe6a
* Which RMW (Fast DDS or Cyclone DDS)? Default Humble Cyclone

### Steps to reproduce

I extend the robot description and moveit config a bit to fit our lab setup: [tami_franka_resources.zip](https://github.com/ros-planning/moveit2/files/13486178/tami_franka_resources.zip) (I run the `tami_panda4` config)

Docker devcontainer runtime environment: 
[franka_lab_rt_devcontainer.zip](https://github.com/ros-planning/moveit2/files/13486211/franka_lab_rt_devcontainer.zip)

Bringup: `ros2 launch tami_franka_moveit_config tami_panda_moveit.launch.py robot_name:=tami_panda4`

The reflex mode joint_limit_violation can be triggered randomly, but on pose I frequently run into issue with is `['panda_link0', 0.6, 0.3, 0.5, 0.707, 0, 0.707, 0]` (the link6 example quoted above is for this target)

### Expected behaviour
Moveit2 should respect joint position limits and plan trajectories within these contraints.

### Actual behaviour
Moveit2 plans trajectories that violate position limits leading to the robot stopping in reflex mode for joint limit violation while executing,

### Backtrace or Console output

```
libfranka: Move command aborted: motion aborted by reflex! ["joint_position_limits_violation"]
```

Tejas-Shastha, 2023-11-29:
I tested this in "demo" mode (i.e. fake controllers without the robot) and wrote a test to detect joint limit violations, and it seems to be happening there as well.

Tejas-Shastha, 2023-11-30:
After some more testing with the "demo" as well as the robot, I have found a simple enough workaround - reducing the joint position limits by ~7% seems to do the trick for me.

@AndyZe You commented [in a post] (https://github.com/ros-planning/moveit2/issues/2557) last year that ros2_control is unaware of joint limits. Is this still true? Could it be that the planner in Moveit2 (in my case the default OMPL/RRT) solves IK for the trajectories with joint limits, but when the controller (in my case the default ros2_control/JTC) follows it, it sometimes solves IK outside of joint limits since it is unaware of it?

gavanderhoorn, 2023-11-30:
> the controller (in my case the default ros2_control/JTC) follows it, it sometimes solves IK

the `JointTrajectoryController` doesn't use IK. It operates purely in joint space.

christophfroehlich, 2023-12-06:
> @AndyZe You commented [in a post] (#2557) last year that ros2_control is unaware of joint limits. Is this still true?

until now, yes. Joint limiter class in progress, but not released (and not included in JTC) yet.

henningkayser, 2024-01-16:
@Tejas-Shastha can you confirm that it's actually a joint limit violation during execution, and not a violation of the commanded trajectory? If MoveIt does not produce waypoints outside of the limits (which is my understanding), this issue should be closed or transferred to the ros2_control repo.

Tejas-Shastha, 2024-01-29:
> @Tejas-Shastha can you confirm that it's actually a joint limit violation during execution, and not a violation of the commanded trajectory? If MoveIt does not produce waypoints outside of the limits (which is my understanding), this issue should be closed or transferred to the ros2_control repo.

@henningkayser I tested this a little more again today and it does indeed seem to be a problem with the execution layer/ros2_control, and not the planning layer/OMPL. i.e. Moveit seems not to be generating trajectories which violate joint limits. Considering what @christophfroehlich  says [above](https://github.com/ros-planning/moveit2/issues/2557#issuecomment-1842478956), this is known and probably does not need an issue in the ros2_control repo. I can close this bug report with the understanding that once the `Joint Limiter` class is shipped with the JTC, this problem should be solved.

closed 2024-01-29 by Tejas-Shastha