moveit/moveit2 #2568: Adding Moveit Servo to Moveit Setup Assistant

Nils-ChristianIseke, 2023-12-02:
Setting up moveit servo for my own robotic arm was a bit annoying. By adding the setup of moveit servo to the moveit setup assistant, one could make moveit servo more easily available.
Feel free to discuss the advantages / disadvantages of it. Maybe i can find some time in 2024 to implement that.
Setting up moveit servo for my own robotic arm was a bit of a pain. Adding the moveit servo setup to the moveit setup assistant would make it easier to use.
Feel free to discuss the pros and cons of this. Maybe I can find some time in 2024 to implement this.

ibrahiminfinite, 2023-12-08:
Hey @Nils-ChristianIseke 
Which part of setting up the servo did you have trouble with ?
Would be great if you could explain it a bit

Nils-ChristianIseke, 2023-12-08:
@ibrahiminfinite of course! Created #2591 related to the problem that occured.
Apart from the fact that this is easy to solve, it would be even nicer if the Moveit Setup Assistant supported this by default, wouldn't it?

sea-bass, 2023-12-08:
I think it would be a great enhancement to auto-generate MoveIt Servo configurations using the Setup Assistant. If you do have time to work on that, that would be very much appreciated!