moveit/moveit2 #2586: Error while building Colcon WS in moveit

Omdiwan, 2023-12-07:
The terminal crashed when trying to build Moveit using the source. further details about the crash and logs are provided below.
Your environment

    ROS Distro: Humble (ros2)
    OS Version: Ubuntu 22.04
    Source build
  

Steps to reproduce

Follow the source build instruction provided [here.](https://moveit.ros.org/install-moveit2/source/) The error occurred after even after successfully installing all the dependencies given in the installation (exception used a different version of clang-format, because clang-format-10 is not available anymore). It occurs when running the build command.

cd $COLCON_WS
colcon build --event-handlers desktop_notification- status- --executor sequential

Expected behaviour

The expected behaviour is to successfully complete the build with minimum warning about deprecation.
Actual behaviour

Finished <<< moveit [0.30s]
Starting >>> moveit_task_constructor_capabilities
--- stderr: moveit_task_constructor_capabilities
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp: In member function ‘bool move_group::ExecuteTaskSolutionCapability::constructMotionPlan(const Solution&, plan_execution::ExecutableMotionPlan&)’:
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:155:14: error: ‘struct plan_execution::ExecutableMotionPlan’ has no member named ‘plan_components’; did you mean ‘plan_components_’?
  155 |         plan.plan_components.reserve(solution.sub_trajectory.size());
      |              ^~~~~~~~~~~~~~~
      |              plan_components_
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:159:22: error: ‘struct plan_execution::ExecutableMotionPlan’ has no member named ‘plan_components’; did you mean ‘plan_components_’?
  159 |                 plan.plan_components.emplace_back();
      |                      ^~~~~~~~~~~~~~~
      |                      plan_components_
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:160:72: error: ‘struct plan_execution::ExecutableMotionPlan’ has no member named ‘plan_components’; did you mean ‘plan_components_’?
  160 | n_execution::ExecutableTrajectory& exec_traj = plan.plan_components.back();
      |                                                     ^~~~~~~~~~~~~~~
      |                                                     plan_components_
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:164:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘description’; did you mean ‘description_’?
  164 |                 exec_traj.description = description;
      |                           ^~~~~~~~~~~
      |                           description_
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:181:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘trajectory’; did you mean ‘trajectory_’?
  181 |                 exec_traj.trajectory = std::make_shared<robot_trajectory::RobotTrajectory>(model, group);
      |                           ^~~~~~~~~~
      |                           trajectory_
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:182:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘trajectory’; did you mean ‘trajectory_’?
  182 |                 exec_traj.trajectory->setRobotTrajectoryMsg(state, sub_traj.trajectory);
      |                           ^~~~~~~~~~
      |                           trajectory_
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:183:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘controller_name’; did you mean ‘controller_names_’?
  183 |                 exec_traj.controller_name = sub_traj.execution_info.controller_names;
      |                           ^~~~~~~~~~~~~~~
      |                           controller_names_
/home/croma/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:186:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘effect_on_success’; did you mean ‘effect_on_success_’?
  186 |                 exec_traj.effect_on_success = [this, sub_traj,
      |                           ^~~~~~~~~~~~~~~~~
      |                           effect_on_success_
gmake[2]: *** [CMakeFiles/moveit_task_constructor_capabilities.dir/build.make:76: CMakeFiles/moveit_task_constructor_capabilities.dir/src/execute_task_solution_capability.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/moveit_task_constructor_capabilities.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_capabilities [7.12s, exited with code 2]

Summary: 64 packages finished [1min 57s]
  1 package failed: moveit_task_constructor_capabilities
  3 packages had stderr output: moveit_planners_chomp moveit_ros_control_interface moveit_task_constructor_capabilities
  3 packages not processed

Have seen similar threads where terminal or the OS used to crash but overcame it by using sequential execution but I still get this error at the end. alteast the thing isnt crashing but still doesnt complete

Omdiwan, 2023-12-08:
Also It always crashed at moveit_task_constructor

sjahr, 2023-12-08:
I guess you're not using the humble brunch of mtc. Please use that one https://github.com/ros-planning/moveit_task_constructor/tree/humble. If you're still running into issues feel free to re-open this issue or open a new one.

Omdiwan, 2023-12-09:
```
croma@croma-Predator-PH315-55:~$ cd ~/ws_moveit
croma@croma-Predator-PH315-55:~/ws_moveit$ cd src/
croma@croma-Predator-PH315-55:~/ws_moveit/src$ cd moveit2/
croma@croma-Predator-PH315-55:~/ws_moveit/src/moveit2$ git switch humble
Already on 'humble'
Your branch is up to date with 'origin/humble'.
croma@croma-Predator-PH315-55:~/ws_moveit/src/moveit2$ cd ..
croma@croma-Predator-PH315-55:~/ws_moveit/src$ colcon build --mixin release  --parallel-workers 1
Starting >>> moveit_common
```
I already switched to humble before starting my build yet it failed at the same point.

Omdiwan, 2023-12-12:
Pls dont close the issue before it has even been solved

sjahr, 2023-12-12:
As I noted above, you need to switch to the humble branch in the **moveit_task_constructor** repository. In your comment you showed only that moveit2 is on humble which is not sufficient.
Please try that and share in this thread if it worked.

Omdiwan, 2023-12-13:
```
Starting >>> moveit_servo
--- stderr: moveit_servo
In file included from /home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/test/test_servo_interface.cpp:39:
/home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/test/servo_launch_test_common.hpp:54:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   54 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/test/test_servo_collision.cpp:41:
/home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/test/servo_launch_test_common.hpp:54:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   54 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/test_servo_integration.dir/build.make:76: CMakeFiles/test_servo_integration.dir/test/test_servo_interface.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:393: CMakeFiles/test_servo_integration.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/test_servo_collision.dir/build.make:76: CMakeFiles/test_servo_collision.dir/test/test_servo_collision.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:421: CMakeFiles/test_servo_collision.dir/all] Error 2
In file included from /home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo.h:47,
                 from /home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/src/servo.cpp:41:
/home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo_calcs.h:52:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   52 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/src/servo_calcs.cpp:50:
/home/croma/ws_moveit/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo_calcs.h:52:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   52 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/moveit_servo_lib.dir/build.make:104: CMakeFiles/moveit_servo_lib.dir/src/servo.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/moveit_servo_lib.dir/build.make:118: CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/moveit_servo_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_servo [0.81s, exited with code 2]

Summary: 50 packages finished [18.6s]
  1 package failed: moveit_servo
  1 package had stderr output: moveit_servo
  16 packages not processed
```
alright thx that worked but now it's stuck on the next file. i had used 
```
<pre>cd ~/ws_moveit/src/moveit_task_constructor
git checkout humble
Branch &apos;humble&apos; set up to track remote branch &apos;humble&apos; from &apos;origin&apos;.
Switched to a new branch &apos;humble&apos;
</pre>
```
to fix the last issue but this one i think cant be solved with it as it is giving```
 croma@croma-Predator-PH315-55:~/ws_moveit/src/moveit2/moveit_ros/moveit_servo$ git checkout humble
Already on 'humble'
Your branch is up to date with 'origin/humble'.
```

sjahr, 2023-12-13:
`moveit_servo` is not and individual git repository. You need to switch to humble on moveit_msgs. Please carefully read the instructions in https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html#download-source-code-of-moveit-and-the-tutorials and follow them.

Rodrigo-Ramon, 2024-02-01:
> `moveit_servo` is not and individual git repository. You need to switch to humble in moveit_msgs. Please carefully read the instructions in https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html#download-source-code-of-moveit-and-the-tutorials and follow them.

Could you expand on how one would fix this? I'm literally copying and pasting the commands line by line and I cannot fix this issue. I have tried re-installing 4 times already

Abishalini, 2024-02-01:
@Omdiwan Do you have a source install of `moveit_msgs`? If so, try 
```
cd ~/ws_moveit/src/moveit_msgs
git checkout humble
```

Rodrigo-Ramon, 2024-02-02:
This is the error I get

```--- stderr: moveit_servo                                                     

In file included from /home/irisdevrr/ws_moveit2/src/moveit2/moveit_ros/moveit_servo/test/test_servo_interface.cpp:39:
/home/irisdevrr/ws_moveit2/src/moveit2/moveit_ros/moveit_servo/test/servo_launch_test_common.hpp:54:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   54 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.
gmake[2]: *** [CMakeFiles/test_servo_integration.dir/build.make:76: CMakeFiles/test_servo_integration.dir/test/test_servo_interface.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:393: CMakeFiles/test_servo_integration.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
In file included from /home/irisdevrr/ws_moveit2/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo.h:47,
                 from /home/irisdevrr/ws_moveit2/src/moveit2/moveit_ros/moveit_servo/src/servo.cpp:41:
/home/irisdevrr/ws_moveit2/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo_calcs.h:52:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   52 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/moveit_servo_lib.dir/build.make:104: CMakeFiles/moveit_servo_lib.dir/src/servo.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/moveit_servo_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_servo [36.8s, exited with code 2]
Aborted  <<< moveit_setup_framework [30.5s]            
Aborted  <<< moveit_task_constructor_core [4min 30s]

Summary: 42 packages finished [5min 19s]
  1 package failed: moveit_servo
  2 packages aborted: moveit_setup_framework moveit_task_constructor_core
  2 packages had stderr output: moveit_servo moveit_task_constructor_core
  10 packages not processed
```

I have followed each step in detail, and this is like the 4th time I try to start from zero.

Any help is highly appreciated!

Monger369, 2024-02-07:
i tried using colcon build --mixin release, the whole system froze at (14/39). then i tried using colcon build --executor sequential it again froze at (29/39). i am not able to understand what i am doing wrong. i did each and every step the tutorial said. can anyone guide me here. i'm using ubuntu 22.04 and humble-ros2.

henningkayser, 2024-02-27:
@Rodrigo-Ramon the last issue looks like you are either installing the wrong version of moveit_msgs in your workspace, or you have another ROS distro installed. The missing header is from a conflicting moveit_msgs version on your system. If you follow the source instructions correctly, all MoveIt branches should point to humble correctly and moveit_msgs should be installed as debian package.

@Monger369 please open a new issue since your description is a performance problem that is unrelated to this issue.

VladimirFokow, 2024-04-23:
I also have this error when building:

```
/robotic_ws/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo_calcs.h:52:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
```

<br />

Output of `colcon build --mixin release`:
<details>
<summary>(click to expand)</summary>
```
Starting >>> moveit_common
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> launch_param_builder
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_common [0.43s]                                                                                                                        
Starting >>> rviz_marker_tools
Finished <<< moveit_resources_fanuc_description [0.42s]
Finished <<< moveit_resources_prbt_support [0.42s]
Finished <<< moveit_resources_panda_description [0.44s]
Finished <<< moveit_resources_pr2_description [0.44s]
Starting >>> moveit_resources_panda_moveit_config
Starting >>> moveit_resources_fanuc_moveit_config
Starting >>> rosparam_shortcuts
Finished <<< srdfdom [0.59s]                                                                                                                                  
Finished <<< moveit_resources_fanuc_moveit_config [0.22s]
Finished <<< moveit_resources_panda_moveit_config [0.24s]                                                                         
Starting >>> moveit_resources
Finished <<< rviz_marker_tools [0.34s]
Finished <<< rosparam_shortcuts [0.37s]                                                                                                                        
Finished <<< moveit_resources [0.16s]
Finished <<< launch_param_builder [0.88s]
Starting >>> moveit_configs_utils
Finished <<< moveit_configs_utils [0.69s]                                                   
[Processing: moveit_msgs]                               
[Processing: moveit_msgs]                                       
Finished <<< moveit_msgs [1min 23s]                                
Starting >>> moveit_core
Starting >>> moveit_task_constructor_msgs
Finished <<< moveit_task_constructor_msgs [1.05s]                                                                        
Finished <<< moveit_core [1.57s]                            
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< moveit_simple_controller_manager [0.19s]                                                                                                 
Starting >>> moveit_plugins
Starting >>> moveit_ros_control_interface
Finished <<< chomp_motion_planner [0.21s]
Starting >>> moveit_chomp_optimizer_adapter
Starting >>> moveit_planners_chomp                                                                                                                                 
Finished <<< pilz_industrial_motion_planner_testutils [0.22s]
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [0.25s]
Finished <<< moveit_ros_occupancy_map_monitor [0.31s]
Starting >>> moveit_ros_planning
Finished <<< moveit_plugins [0.18s]                                                                                                                                             
Finished <<< moveit_chomp_optimizer_adapter [0.19s]
Finished <<< moveit_planners_chomp [0.24s]                                                                                                                          
Finished <<< moveit_ros_control_interface [0.28s]
Finished <<< moveit_ros_planning [0.78s]                             
Starting >>> moveit_kinematics
Starting >>> moveit_ros_warehouse
Starting >>> moveit_ros_robot_interaction
Starting >>> moveit_planners_ompl
[... 33 lines cut ...]
In file included from /robotic_ws/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo.h:47,
                 from /robotic_ws/src/moveit2/moveit_ros/moveit_servo/src/servo.cpp:41:
/robotic_ws/src/moveit2/moveit_ros/moveit_servo/include/moveit_servo/servo_calcs.h:52:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   52 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/moveit_servo_lib.dir/build.make:104: CMakeFiles/moveit_servo_lib.dir/src/servo.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/moveit_servo_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_servo [2.88s, exited with code 2]
Aborted  <<< moveit_runtime [0.59s]                                                                                                                                 
Aborted  <<< moveit_setup_framework [5.93s]                                                                                           
Aborted  <<< moveit_task_constructor_core [4.78s]                                   

Summary: 42 packages finished [1min 34s]
  1 package failed: moveit_servo
  3 packages aborted: moveit_runtime moveit_setup_framework moveit_task_constructor_core
  1 package had stderr output: moveit_servo
  10 packages not processed
```
</details>

My question is: are the installation instructions for humble even working at all?
Why can't I even install it?

katviltrake, 2024-04-26:
Had the same issue, built the moveit_msgs package separately without the --mixin release tag (just colcon build), and it wasn't building the .hpp files. Built with the tag and it build the needed .hpp files in moveit_msgs package install folder, then sourced the moveit_msgs package so moveit_servo could see the files and successfully built that. Hope that helps.

fzoric8, 2024-04-30:
Hi, 

I'm trying to build `moveit2` for the `ros 2` humble distro. 

Env check:  
```
echo $ROS_DISTRO 
```
output: 
`humble`

I'm also sourcing `/opt/ros/humble/setup.bash`, so there's no doubt I'm using humble. 

After that, I've installed `moveit` with: 
```
sudo apt-get install ros-humble-moveit
```

After that I've created colcon workspace following `moveit` [instructions](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html). 

And after that, I've cloned `moveit2_tutorials` and `moveit_task_constructor`, and output of the following command `cd /moveit2_tutorials && git log` is: 
```
commit 20be39797b17e460fda48975a0a8ee25f8ad87ac (HEAD -> humble, origin/humble)
...
```

Output of the `cd /ws_moveit/src/moveit_task_constructor/ && git log` is: 
```
commit 5ae22da8db152fb449b806211f9fc67f8e4f0757 (HEAD -> humble, origin/ros2, ros2)
``` 
and output of the `colcon build --mixin release` is: 

```
In file included from /home/developer/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:39:
/home/developer/ws_moveit/src/moveit_task_constructor/core/include/moveit/task_constructor/solvers/pipeline_planner.h:42:10: fatal error: moveit/planning_pipeline_interfaces/planning_pipeline_interfaces.hpp: No such file or directory
   42 | #include <moveit/planning_pipeline_interfaces/planning_pipeline_interfaces.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

VladimirFokow, 2024-04-30:
hi @fzoric8 , I also don't know if building standalone moveit packages (like `moveit_task_constructor`) is possible without building the whole moveit (so not installing with `apt`).

As of now, the [workaround with the apt packages built in CI stopped working months ago and is not supported](https://github.com/moveit/moveit2/discussions/2810#discussioncomment-9252394)

fzoric8, 2024-04-30:
Hi @VladimirFokow, thank you for your quick answer. 

In the end, I've managed to build moveit2 for `humble` from source but in clean docker environment. 
Yet `moveit_setup_srdf_plugins` failed. 

However, `apt` install doesn't work. 

If anyone needs `Dockerfile I've made, you can see it [here](https://github.com/larics/docker_files/tree/master/ros2-humble/moveit2).

closed 2024-06-28 by Omdiwan