moveit/moveit2 #2600: TOTG not respecting acceleration limits

marioprats, 2023-12-12:
### Description

TOTG seems to not care about acceleration limits other than at the beginning and end of a path.
I created an example test here to verify that a trajectory timed with TOTG satisfies the acceleration limits:
https://github.com/marioprats/moveit2/commit/0993a9a44b1c809cfbd332fe90103129837e97c9

The test is failing with accelerations up to 200 rad/s2, when the limit should be 0.2 rad/s2.

The input path consists of 10 waypoints (one single variable), in a u-shape:

[image]

Acceleration spikes to 200 rad/s2 at the inflection point:

[image]

But it behaves as expected at the beginning of the path for instance (maximum acceleration until maximum velocity is reached).

[image]

### Your environment
* ROS Distro: Rolling
* OS Version: e.g. Ubuntu 22.04
* Source, main

### Steps to reproduce
(https://github.com/marioprats/moveit2/commit/0993a9a44b1c809cfbd332fe90103129837e97c9)

### Expected behaviour
I would expect the output trajectory to satisfy velocity and acceleration limits at all times.

marioprats, 2023-12-13:
I found out of couple of things: my simple test is a non-supported edge case in TOTG, since it involves a full 180 degree turn on the path (a joint moving to zero, and back). The second finding is that the default `path_tolerance` of 0.0 is also not a supported case, since that doesn't include rounding at the corners. The `RobotTrajectory` timing wrapper sets a default different than zero, so that works, but the actual TOTG class has a default of 0.0, so if anyone is using that directly without changing the default, the velocities / accelerations won't always make sense.
I'm going to try documenting all this better and setting a default different than zero. I'll also see if we can easily detect the 180 turn case and return an error in that case.

closed 2024-01-03 by sjahr

buckleytoby, 2025-05-07:
Can you explain more what you mean by 180 degree turn? Do you mean any time a joint goes from positive to negative velocity?