moveit/moveit2 #2615: Error while planning: [move_group.moveit.planning_pipeline]: Exception caught: 'vector::_M_range_check: __n (which is 3) >= this->size() (which is 3)'

jdbravo, 2023-12-16:
### Description
While planning on Rviz2 I receive the following error:

[move_group.moveit.planning_pipeline]: Exception caught: 'vector::_M_range_check: __n (which is 3) >= this->size() (which is 3)'

And the plan just fail

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build
* Main branch
* Cyclone DDS

### Steps to reproduce
Run my launcher that opens my rviz2, in Motion Planing I choose OMPL as Planning Library and Check Approx IK Solutions, then I click plan.

### Expected behaviour
Plan the motion

### Actual behaviour
I receive a Fail status and an error in my outpu

### Backtrace or Console output

https://gist.github.com/jdbravo/665e5a815902a5a5cc2855ab4c4ffcf8

sea-bass, 2023-12-17:
Hmm... according to the output, seems like the segfault happened when running the `CheckStartStateBounds` plugin.

Would you be able to share your URDF/joint limits config to know what kind of joints there are, and their limits?

sea-bass, 2023-12-17:
Also, I've found that reverting https://github.com/ros-planning/moveit2/pull/2546 makes this segfault go away. Could you try on your end, either by reverting or going to a commit before this one?

rhaschke, 2023-12-18:
If reverting #2546 fixes this issue, we had an out-of-bounds issue for years in the old code, which simply didn't show up because there was no bounds checking. Could you please try to pinpoint the actual RobotState access that is failing here?

marioprats, 2023-12-18:
We indeed have had out-of-bounds memory access in `RobotState` for years. https://github.com/ros-planning/moveit2/pull/2546 will uncover where those happen in code that is not already covered by tests.
I couldn't reproduce this particular issue with the tutorial robot. So I'd also like to make sure you don't have any `moveit` deb packages installed (especially `moveit_core`) that may interfere with your build:

```
 dpkg -l ros-humble-moveit-core
```

That should return `dpkg-query: no packages found matching ros-humble-moveit-core`.

rhaschke, 2023-12-18:
As pointed out elsewhere, I couldn't find _actual_ out-of-bounds memory accesses in the old code.

jdbravo, 2023-12-20:
> Hmm... according to the output, seems like the segfault happened when running the `CheckStartStateBounds` plugin.
> 
> Would you be able to share your URDF/joint limits config to know what kind of joints there are, and their limits?

Sure!

My joint limits are:
default_velocity_scaling_factor: 0.1
default_acceleration_scaling_factor: 0.1

joint_limits:
  shoulder_2_female_joint:
    has_velocity_limits: true
    max_velocity: 10.0
    has_accelearation_limits: false
    max_acceleration: 0.0

  shoulder_2_male_joint:
    has_velocity_limits: true
    max_velocity: 10.0
    has_accelearation_limits: false
    max_acceleration: 0.0

  elbow_2_male_joint:
    has_velocity_limits: true
    max_velocity: 10.0
    has_accelearation_limits: false
    max_acceleration: 0.0

And the URDF https://gist.github.com/jdbravo/2d58a2dd1970cf73c2198eca7a76faed

marioprats, 2023-12-20:
That URDF has a fixed joint at the end of the chain, which could have been affected by the corner case we fixed in https://github.com/ros-planning/moveit2/pull/2617.
If you pull from `main` and try again, this is hopefully no longer an issue.

henningkayser, 2024-02-06:
looks like this has been fixed a while ago, please re-open if this is still an issue

closed 2024-02-06 by henningkayser