moveit/moveit2 #2626: logger initialized prior to `rclcpp::init()` in `MoveItPy`

TomCC7, 2023-12-21:
### Description

I tried to run the [Python script](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/motion_planning_python_api/scripts/motion_planning_python_api_tutorial.py) provided in [Python api tutorial](https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html) and get the following error:
```sh
[WARN] [1703131618.467477834] [moveit_2843130760]: exception thrown while creating node for logging: failed to create guard condition: context argument is null, at ./src/rcl/guard_condition.c:65
[WARN] [1703131618.467498124] [moveit_2843130760]: if rclcpp::init was not called, messages from this logger may be missing from /rosout
Traceback (most recent call last):
  File "/moveit_ws/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py", line 176, in <module>
    main()
  File "/moveit_ws/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py", line 62, in main
    panda = MoveItPy(node_name="moveit_py")
RuntimeError: failed to call rcl_logging_rosout_add_sublogger: The entry of logger 'moveit_2843130760' not exist., at ./src/rcl/logging_rosout.c:450
[ros2run]: Process exited with failure 1
```

Digging deeper into the code I found that the problem might be related to the same issue explained [here](https://robotics.stackexchange.com/questions/105110/exception-thrown-in-rcl-logging-rosout-remove-sublogger-on-destruction-with-a-ch) (ros node went outside scope and the entry in `__logger_map` was removed). Looking at [codes that create the global logger](https://github.com/ros-planning/moveit2/blob/main/moveit_core/utils/src/logger.cpp#L51-L70), it seems that the fallback option that initializes the logger through `auto logger = rclcpp::get_logger(name);` will cause the issue mentioned above. And because in the [moveitpy initializer](https://github.com/ros-planning/moveit2/blob/main/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L88-L102), `getLogger()` is called before `rclcpp::init()`, the fallback is triggered.

So to summarize, the problem seems to be
1. `getLogger()` called prior to `rclcpp::init()`, `__logger_map` not initialized
2. `get_child()` fails to find parent logger in `__logger_map` and triggers the error.

As a quick fix, moving all the logging after `rclcpp::init()` will solve the issue. However, it might also be worth adding another warning [here](https://github.com/ros-planning/moveit2/blob/main/moveit_core/utils/src/logger.cpp#L66) saying that the `get_child` will also fail if logger is initialized before `rclcpp::init()`.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* main branch #2599 
* Cyclone DDS

### Steps to reproduce
Setup the environment following the [getting started](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) tutorial and run
```sh
ros2 run moveit2_tutorials motion_planning_python_api_tutorial.py
```

### Expected behaviour
program runs successfully

### Actual behaviour
program crashes

### Backtrace or Console output

```sh
[WARN] [1703131618.467477834] [moveit_2843130760]: exception thrown while creating node for logging: failed to create guard condition: context argument is null, at ./src/rcl/guard_condition.c:65
[WARN] [1703131618.467498124] [moveit_2843130760]: if rclcpp::init was not called, messages from this logger may be missing from /rosout
Traceback (most recent call last):
  File "/moveit_ws/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py", line 176, in <module>
    main()
  File "/moveit_ws/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py", line 62, in main
    panda = MoveItPy(node_name="moveit_py")
RuntimeError: failed to call rcl_logging_rosout_add_sublogger: The entry of logger 'moveit_2843130760' not exist., at ./src/rcl/logging_rosout.c:450
[ros2run]: Process exited with failure 1
```

dan9thsense, 2023-12-21:
Exact same error encountered while trying to run the demo jupyter notebook in the tutorials.

TomCC7, 2023-12-21:
Hey @dan9thsense, you can try the fix I mentioned, it's working for me. Just remove or move [this line](https://github.com/ros-planning/moveit2/blob/8d2eb900d222f6609ee2588f6f5ebb97986635bd/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L90) below the `rclcpp::init` and rebuild `moveit_py`

dan9thsense, 2023-12-21:
I din't install from source, so I'll have to go back and do that.  Might take a few days to get to it but I'll let you know.
Thanks for tracking down the error; I encountered it about a week ago and it stopped me cold in doing the jupyter notebook in the tutorials.  However, I was able to get this to work from the tutorials (using ROS2 Iron):

ros2 launch moveit2_tutorials demo.launch.py rviz_config:=test.rviz

dan9thsense, 2023-12-21:
I had a meeting canceled, so I uninstalled the binary and installed with source.  Oddly enough, the code ran without that error, both in my code and in the demo jupyter notebook.  However, the demo jupyter notebook generated a different error from the same line.

Just to test, I moved
RCLCPP_INFO(LOGGER, "Initialize rclcpp");
to below
rclcpp::init(launch_arguments.size(), chars.data());
in
moveit_cpp.cpp
to see if that had any impact.  No difference, so it doesn't seem to hurt.  If the bug re-appears, I'll post about it here.

In case you are interested, here is the new bug in the demo jupyter notebook.  It does not show up in my code, so I'm not planning to follow it up in detail.
   4 # instantiate moveit_py instance and a planning component for the panda_arm
----> 5 panda = MoveItPy(node_name="moveit_py", config_dict=moveit_config)
      6 panda_arm = panda.get_planning_component("panda_arm")

RuntimeError: parameter 'ompl.request_adapters' has invalid type: expected [string] got [string_array]

closed 2024-02-11 by TomCC7