moveit/moveit2 #2630: Migrate to modern Gazebo

azeey, 2023-12-21:
As you may know, Gazebo-classic (aka Gazebo11, see [Terminology](https://gazebosim.org/about#:~:text=Terminology)) is no longer being actively developed  and will not be available on ROS Jazzy ([REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029)). I noticed that this project still uses Gazebo-classic, mainly in the setup assistant. Are there any plans to migrate to modern Gazebo ([previously known as Ignition](https://community.gazebosim.org/t/a-new-era-for-gazebo/1356))?

sea-bass, 2024-01-18:
Thanks for sharing this -- I was not aware that Gazebo classic was approaching EOL, and we should probably address this -- even if, short term, it's removing the Gazebo classic stuff in Setup Assistant for Jazzy.

We do have some examples of new Gazebo hanging around, most notably the Panda arm setup we did for World MoveIt Day last year: https://github.com/henningkayser/moveit2_tutorials/blob/panda_gz/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.rst#running-the-tutorial-in-a-gazebo-simulation -- so that could be a starting point.

henningkayser, 2024-02-27:
I'll take a look at this fairly soon. Especially in preparation for and possibly during GSoC, we should have a clear migration strategy.

orfvl, 2024-05-20:
Any updates of removing gazebo classic? I installed moveit and it is still a dependency.