moveit/moveit2 #2632: moveit_servo: Warning points user to non-working intra-process communication

SammyRamone, 2023-12-22:
### Description

If moveit_servo is started it will give the following warning:
`Intra-process communication is disabled, consider enabling it by adding:  extra_arguments=[{'use_intra_process_comms' : True}] to the Servo composable node in the launch file moveit_servo.servo_node
`
Naturally, one is inclined to enable this after reading this warning. However, this will lead to the following error:
`Component constructor threw an exception: intraprocess communication is not allowed with a zero qos history depth value`

Therefore, I think that this warning is currently very misleading.

This issue has already been discussed in https://github.com/ros-planning/moveit2/pull/722 where the intra-process communication was removed by @tylerjw to resolve this bug. @henningkayser already proposed to also remove this warning. However, this was not done.

I propose to remove/disable the warning for now and to open another issue about allowing intra-process communication in MoveIt2, (which is currently blocked by some QoS settings).
I also did a quick search if I can find the where this use of a zero history depth qos happens, but I did not find anything in the moveit code.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source build 2.8.0
* If source, which branch?
* Cyclone

### Steps to reproduce
Launch moveit_servo as a ComposableNode.

### Expected behaviour
Either no warning that the intra-process communication is not used or that the node works when intra-process communication is enabled.

### Actual behaviour
Following advice of the warning leads to an error.

ibrahiminfinite, 2024-01-02:
@sea-bass 
Did you guys come across this error ?
I don't remember testing this during the refactor, so this is probably something that was missed.

henningkayser, 2024-02-06:
@SammyRamone would you mind filing a PR to fix this? ;)

SammyRamone, 2024-02-16:
> @SammyRamone would you mind filing a PR to fix this? ;)

I can do that, but it will take some time since I will be busy till end of february

SammyRamone, 2024-03-19:
@henningkayser  I opened a PR in https://github.com/ros-planning/moveit2/pull/2752

closed 2024-04-30 by sjahr

lukestroh, 2024-09-19:
I am still getting this error in Humble, will the fix be back ported any time soon? Should I avoid the composable node in Humble?

SammyRamone, 2024-09-20:
I can not comment on back porting of this fix. However, you can just ignore the warning. You don't need to avoid the composable node. The only thing you need to avoid is following the warnings advice, since this will lead to an error.