moveit/moveit2 #2642: Error Loading DepthImageOctomapUpdater Plugin in MoveIt2 Occupancy Map Monitor

Albertdalmen, 2024-01-04:
### Description
Encountered an issue with MoveIt's Occupancy Map Monitor when trying to load the DepthImageOctomapUpdater plugin. The system reports that it cannot find the specified class derived from the occupancy_map_monitor::OccupancyMapUpdater base class, even though it is expected to be present. The sensor configuration file looks like: 

```yaml
sensors:
  - depth_camera

depth_camera:
  sensor_plugin: occupancy_map_monitor/DepthImageOctomapUpdater # name of the plugin that we are using : [PointCloudOctomapUpdater or DepthImageOctomapUpdater]
  image_topic: /camera/depth/image_rect_raw # topic to listen on for a depth image
  queue_size: 5                             # number of images to queue up
  near_clipping_plane_distance: 0.003       # minimum distance before lack of visibility
  far_clipping_plane_distance: 5.0          # maximum distance before lack of visibility
  shadow_threshold: 0.2                     # minimum brightness of the shadow map below an entity for its dynamic shadow to be visible
  padding_offset: 0.03                      # size of the padding (in cm).
  padding_scale: 4.0                        # scale of the padding
  max_update_rate: 1.0                      # octomap representation will be updated at rate less than or equal to this value
  filtered_cloud_topic: filtered_cloud      # topic on which the filtered cloud will be published (mainly for debugging)
  ns: realsense                             # namespace
```
*** I also tried with:  "occupancy_map_monitor/DepthImageOccupancyMapUpdater", and without the namespace "occupancy_map_monitor".   

### Backtrace or Console output
```
[move_group-1] [ERROR] [1704369569.576478830] [moveit.ros.occupancy_map_monitor.middleware_handle]: Exception while loading octomap updater 'occupancy_map_monitor/DepthImageOctomapUpdater': According to the loaded plugin descriptions the class occupancy_map_monitor/DepthImageOctomapUpdater with base class type occupancy_map_monitor::OccupancyMapUpdater does not exist. Declared types are 
[move_group-1] 
[move_group-1] [ERROR] [1704369569.576493275] [moveit.ros.occupancy_map_monitor]: Failed to load sensor: `depth_camera` of type: `occupancy_map_monitor/DepthImageOctomapUpdater`
```
### environment
* ROS Distro: Humble Binary build. 
* OS Version: Ubuntu 22.04 (Jammy)
* RMW:  (Fast DDS or Cyclone DDS)? rmw_fastrtps_cpp

### Expected behaviour
The DepthImageOctomapUpdater plugin should load successfully, allowing the occupancy map monitor to utilize depth images for updating the octomap.

### Actual behaviour
The plugin fails to load, with the system unable to recognize the DepthImageOctomapUpdater class as a derived class of occupancy_map_monitor::OccupancyMapUpdater. The list of declared types appears empty, suggesting an issue with plugin discovery or registration.

closed 2024-01-09 by Albertdalmen

AGREATAPE, 2024-02-21:
@Albertdalmen 
I have the same problem. Have you solved it?

pshkowda, 2024-03-01:
@Albertdalmen 
I have the same problem, could You please share how did you solve it?

johnny555, 2024-03-14:
I had this error, I was missing the moveit-ros-perception package
`sudo apt install ros-humble-moveit-ros-perception`

shrutichakraborty, 2025-04-17:
Hi @Albertdalmen, I have the same issue how did you solve it? @johnny555  the `sudo apt install ros-humble-moveit-ros-perception` package does not exist :/ How did you solve this?

Albertdalmen, 2025-04-19:
Hi @shrutichakraborty, the `sudo apt install ros-humble-moveit-ros-perception` command did work for me.  It seems there is something wrong with the package build. Have you tried the suggested `sudo apt update` first before trying to install or `sudo apt install --fix-missing ros-humble-moveit-ros-perception`?