moveit/moveit2 #2645: Failed to fetch current robot state

ZERO-EAST, 2024-01-05:
# Description

[INFO] [1704449878.278637370] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1704449877.278544, but latest received state has time 250.655000.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1704449878.278697751] [move_group_interface]: Failed to fetch current robot state

# Your environment
1. ROS Distro: humble
2. OS Version: e.g. Ubuntu 22.04
3. Moveit2

# Other Information
I have set both use_sim_time of Gazebo move_group to True，But when I get the current status through code, it always fails. The error message is as follows:

[INFO] [1704449877.278522295] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[INFO] [1704449878.278637370] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1704449877.278544, but latest received state has time 250.655000.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1704449878.278697751] [move_group_interface]: Failed to fetch current robot state

# Code

```
int main(int argc, char **argv)
{
    rclcpp::init(argc, argv);
    rclcpp::NodeOptions node_options;
    node_options.automatically_declare_parameters_from_overrides(true);
    auto move_group_node = rclcpp::Node::make_shared("moveit_move_test", node_options);

    rclcpp::executors::SingleThreadedExecutor executor;
    executor.add_node(move_group_node);
    std::thread([&executor]()
                { executor.spin(); })
        .detach();
    std::thread executor_thread([&executor]() { executor.spin(); });
    executor_thread.detach();

    static const std::string PLANNING_GROUP = "left_arm_group";

    moveit::planning_interface::MoveGroupInterface move_group(move_group_node, PLANNING_GROUP);

    moveit::planning_interface::PlanningSceneInterface planning_scene_interface;

    const moveit::core::JointModelGroup *joint_model_group =
        move_group.getCurrentState()->getJointModelGroup(PLANNING_GROUP);
    
    rclcpp::shutdown();
    return 0;
}
```

[image]

sjahr, 2024-01-08:
Please don't use the issue section to ask for help with debugging your personal code. Maybe reading through the [tutorials](https://moveit.picknik.ai/main/doc/tutorials/tutorials.html) or posting the question on https://robotics.stackexchange.com/ might help you. If you're planning to post the question there, I think you should attach the full log + your launch files too. Good luck!

closed 2024-01-08 by sjahr

fzoric8, 2024-05-13:
Hi  @ZERO-EAST this problem is probably caused by the lack of `use_sim_time` parameter for one of the nodes launched.