moveit/moveit2 #2657: MoveItPy overrides planning scene

christian-rauch, 2024-01-23:
### Description

Instantiating `MoveItPy` overrides a previously published planning scene.

### Your environment
* ROS Distro: Humble with `moveit` 2.8 from source
* OS Version: Ubuntu 22.04
* Source or Binary build? both
* If binary, which release version?
* If source, which branch? 2.8
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone.

### Steps to reproduce
1. `ros2 launch franka_moveit_config moveit.launch.py robot_ip:=dont-care use_fake_hardware:=true`
2. add a collision object in RViz under `Scene Objects` and `Publish`
4. construct a `MoveItPy` with the `motion_planning_python_api_tutorial.yaml` from the tutorials and `monitored_planning_scene_topic: "/monitored_planning_scene"`:
    ```Python
    #!/usr/bin/env python3
    from moveit.planning import MoveItPy
    from moveit_configs_utils import MoveItConfigsBuilder
    import os
    if __name__ == '__main__':
        # https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/main/doc/examples/motion_planning_python_api/config/motion_planning_python_api_tutorial.yaml
        # monitored_planning_scene_topic: "/monitored_planning_scene"
        moveit_cfg_dict = MoveItConfigsBuilder(
            robot_name="panda", package_name="moveit_resources_panda_moveit_config")\
            .moveit_cpp(file_path=os.path.join(os.getcwd(), "motion_planning_python_api_tutorial.yaml"))\
            .to_moveit_configs().to_dict()
        moveit = MoveItPy(node_name="moveit_python", config_dict=moveit_cfg_dict)
    ```

### Expected behaviour
The Python bindings should behave like the C++ API and not change the already published scene.

### Actual behaviour
The Python bindings publish an empty planning scene on `/monitored_planning_scene`, overriding the previously defined collision objects. This does not happen with the C++ API.

wei1224hf, 2024-01-26:
How you install the MoveItPy with ROS2 humble ?

christian-rauch, 2024-01-26:
> How you install the MoveItPy with ROS2 humble ?

Compile from source.

peterdavidfagan, 2024-02-07:
Hi @christian-rauch, 

Thanks for posting this issue, I'll try to find time to investigate this bug further. 

When performing the same operation from the C++ API, were you also calling `moveit_cpp_ptr->getPlanningSceneMonitorNonConst()->providePlanningSceneService()` as it done [here](https://github.com/ros-planning/moveit2_tutorials/blob/ca02877e650659f9dc941697f88e75623cc32137/doc/examples/moveit_cpp/src/moveit_cpp_tutorial.cpp#L52C54-L52C81).

Within the `MoveItPy` initialization we have a parameter for optionally starting the planning scene service as shown [here](https://github.com/ros-planning/moveit2/blob/6fa8480649fd21fee780ba8b4b8366575ca0d9f6/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L129).

If/when I start debugging I'll like start from this point.