moveit/moveit2 #2663: Does ROS2 humble support moveitpy ?

wei1224hf, 2024-01-26:
I download the moveit2 source code target humble branch,  build it , install it, 
but found that it has no moveitpy package installed, 
found it by running this code:
`ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py`
with this error:
`ERROR] [ros2_control_node-3]: process has died [pid 3035, exit code -6, cmd '/home/gene/ws_moveit2/install/controller_manager/lib/controller_manager/ros2_control_node --ros-args --params-file /home/gene/ws_moveit2/install/moveit_resources_panda_moveit_config/share/moveit_resources_panda_moveit_config/config/ros2_controllers.yaml -r /controller_manager/robot_description:=/robot_description'].
[rviz2-4] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[motion_planning_python_api_tutorial.py-1] Traceback (most recent call last):
[motion_planning_python_api_tutorial.py-1]   File "/home/gene/ws_moveit2/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py", line 13, in <module>
[motion_planning_python_api_tutorial.py-1]     from moveit.core.robot_state import RobotState
[motion_planning_python_api_tutorial.py-1] ModuleNotFoundError: No module named 'moveit'
`

can I install this package stand alone?  and how?

sjahr, 2024-01-26:
No, but you can build the main branch on humble. You find more information about our branching and updating strategy in [this blog post](https://moveit.ros.org/moveit%202/ros/2023/05/31/balancing-stability-and-development.html)

closed 2024-01-26 by sjahr

wei1224hf, 2024-01-27:
> No, but you can build the main branch on humble from source. Instructions can be found [here](https://moveit.ros.org/install-moveit2/source/). You find more information about our branching and updating strategy in [this blog post](https://moveit.ros.org/moveit%202/ros/2023/05/31/balancing-stability-and-development.html)

Thank you , 
I plan to switch to iron branch , it contains moveit_py package, I guess this branch may be stable
I failed to build the main branch on humble.

kidpaul94, 2024-06-04:
Any update on this? I also had the same issue when I installed the humble version.