moveit/moveit2 #2664: main branch build error while compiling on ROS-HUMBLE

wei1224hf, 2024-01-27:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [|Humble|]
* OS Version: Ubuntu 22.04
* Source build, main branch
* Which RMW (Fast DDS or Cyclone DDS): rmw_cyclonedds_cpp

### Steps to reproduce
1 install the humble branch already , from source .
2 find it do not contain moveit_py , so I install the main branch source and compile it , with command:
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release --allow-overriding moveit_common moveit_core moveit_ros_planning`

### Expected behaviour
The compile process should be done

### Actual behaviour
error comes out then compile process stopped

### Backtrace or Console output

`--- stderr: moveit_core
In file included from /home/gene/ws_moveit2_main/src/moveit2/moveit_core/online_signal_smoothing/src/acceleration_filter.cpp:35:
/home/gene/ws_moveit2_main/src/moveit2/moveit_core/online_signal_smoothing/include/moveit/online_signal_smoothing/acceleration_filter.h:84:10: fatal error: types.h: No such file or directory
   84 | #include <types.h>
      |          ^~~~~~~~~
compilation terminated.
gmake[2]: *** [online_signal_smoothing/CMakeFiles/moveit_acceleration_filter.dir/build.make:76: online_signal_smoothing/CMakeFiles/moveit_acceleration_filter.dir/src/acceleration_filter.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1824: online_signal_smoothing/CMakeFiles/moveit_acceleration_filter.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_core [15.8s, exited with code 2]

Summary: 11 packages finished [26.7s]
  1 package failed: moveit_core
  2 packages had stderr output: moveit_configs_utils moveit_core
  35 packages not processed
gene@gene-virtual-machine:~/ws_moveit2_main$ sudo http_proxy='http://192.168.17.1:10809'  apt-get install ros-humble-rmw-cyclonedds-cpp
[sudo] password for gene: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-humble-rmw-cyclonedds-cpp is already the newest version (1.3.4-1jammy.20231205.152504).
0 upgraded, 0 newly installed, 0 to remove and 44 not upgraded.`

wei1224hf, 2024-01-28:
fixed this bug by modify this code:

/home/gene/ws_moveit2_main/src/moveit2/moveit_core/online_signal_smoothing/include/moveit/online_signal_smoothing/acceleration_filter.h

line 84:

#include <types.h>

to 

#include <sys/types.h>

wei1224hf, 2024-01-28:
what version of osqp is moveit2 using?
I download and compile and using the newest osqp

wei1224hf, 2024-02-03:
I switch to ROS-IRON because humble dose not support moveit-py right now. 
With ROS-IRON , I failed with souce install moveit-py , successed with binary install moveit-py.

closed 2024-02-03 by wei1224hf

evelyn-fu, 2024-04-24:
Hi, did you ever figure out which version of OSQP is needed to build moveit2 on ros-iron? Thanks!

adamanov, 2024-05-27:
same here, are you guys could build `main` branch on humble?

Rainerino, 2024-06-14:
It doesn't work with humble out of the box for me neither. I fixed it related errors by using OSQP release 0.6.3 https://github.com/osqp/osqp/releases/tag/v0.6.3

adamanov, 2024-06-14:
I could compile the Moveit `main branch` on **humble** with the following 

``` 
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 2 
```

i.e. by setting  `--parallel-workers 2`

followthwhiterabbit, 2024-06-22:
> I could compile the Moveit `main branch` on **humble** with the following
> 
> ```
> colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 2 
> ```
> 
> i.e. by setting `--parallel-workers 2`

it didn't work form me, still struggling to run moveit for a week now

lnss1-ua, 2024-10-18:
> It doesn't work with humble out of the box for me neither. I fixed it related errors by using OSQP release 0.6.3 https://github.com/osqp/osqp/releases/tag/v0.6.3

Could you tell me how to use it? i downloaded the linux tar but I don't if i have to install something or what.

BedoHatem, 2025-04-01:
### This is how i made it work.

Error: No such file or directory
    4 | #include <moveit/move_group_interface/move_group_interface.hpp>

I changed the hello_moveit.cpp file of the node to use "**move_group_interface.h**" instead of "move_group_interface.hpp"