moveit/moveit2 #2694: Properly handle move_group_capabilities

forrest-rm, 2024-02-19:
### Description

In `moveit_configs_utils/moveit_configs_utils/launches.py`, the capabilities parameter is handled separately instead as part of moveit_config.to_dict() like the other moveit_config parameters.

### Environment
* ROS Distro: Humble, but also present in other distros
* OS Version: Ubuntu 22.04
* humble and main source branches

### Steps to reproduce
A temporary fix was made in PR: #2587
@sjahr suggested creating an issue for a more integrated fix

gavanderhoorn, 2024-06-05:
@sjahr: could you summarise the status of this issue?

With configuration packages generated by the MSA from `main` (bee0a2985bdf5e5c9ba67c0e2e0d74c6a8840c63) we're seeing #2734 here, which I believe is related to this one, and is somewhat confusing.

I'd like to avoid approaches like [this one](sam-xl/moveit2#1), as it would mean maintaining yet another fork.

mikeferguson, 2024-10-02:
I've opened https://github.com/moveit/moveit2/pull/3018 to fix the current issues that make the exported MSA config not actually run

From what I can see, we can't actually use the moveit_config.to_dict() unless we drop support for the launch parameters capabilities/disabled_capabilities (which I have no idea how widely the are used).