moveit/moveit2 #2714: Remove topic name parameters in favor of const topic names

sjahr, 2024-02-28:
In multiple places of moveit2 (for example [moveit_cpp](https://github.com/ros-planning/moveit2/blob/5887eb0e2eb8f6803d85fac458b11122fe495449/moveit_ros/planning/moveit_cpp/include/moveit/moveit_cpp/moveit_cpp.h#L65)), topic names can be configured with a parameter. This is not necessary because remapping topics on launch file level is trivial and the parameter API can be simplified by removing these parameters. Instead of having the topic names configurable with a ROS 2 parameters we should just use a const name in the cpp code.

yash-011, 2024-03-01:
first issue was good

Gaurang-1402, 2024-04-01:
Hi @sjahr  I have made a PR to the moveit2 repo that resolves this issue: https://github.com/ros-planning/moveit2/pull/2777

I'm curious to hear your feedback and improvise!

PR #2728 (2024-07-26): Const topic instead of parameters
### Description

Replaced `rclcpp::Node::get_parameter_or()` by instead assigning value directly to topic variables.
I could not find any other instances where this particular case has been repeated, i.e., topic names assigned with `get_parameter_or()` function.

Fixes #2714 

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials/documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the us [... cut]
closed 2024-07-26 by sjahr