moveit/moveit2 #2716: Bug introduced by f596b9df "Enable chaining planners": collision with attached objects with Pilz planner

galou, 2024-02-28:
### Description

Commit f596b9df introduces a bug at least with our configuration. Pilz planners (PTP and LIN) returns trajectories where the robot or attached objects are in collision with the robot or the environment.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build,  branch `main`

### Steps to reproduce
- Attempt to plan a trajectory where the goal target is in collision. I tried
  - robot with itself
  - attached object with robot
  - attached object with the a plane in the environment
  - attached object with the a box in the environment
- Pilz/PTP and Pilz/LIN return plans where a collision occur
- OMPL/AnyTimePathShortening with planners `RRTConnect[range=0.0],EST[goal_bias=0.5],PRM[max_nearest_neighbors=5],TRRT,SBL` fails as expected

### Expected behaviour
Pilz/PTP and Pilz/LIN should not return any trajectory in collision.

### Actual behaviour
Pilz/PTP and Pilz/LIN return plans where a collision occur.

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

galou, 2024-02-28:
@sjahr can you please confirm that you can reproduce this? This may be due to a configuration error on our side. Does the planner chaining require some configuration change?

galou, 2024-02-28:
Adding `default_planning_response_adapters/ValidateSolution` to the `response_adapters` list doesn't help (again, assuming there is no configuration error).

galou, 2024-02-29:
My doubts about the possibility that this is due to a configuration change are confirmed. We didn't use any `request_adapters`  but, since before https://github.com/ros-planning/moveit2/commit/f596b9df89c07805f5b911936486ac9f165543b0, `default_planning_response_adapters/ValidateSolution` should  be added to `response_adapters` to activate the collision check.

closed 2024-02-29 by galou

sjahr, 2024-02-29:
Sorry, just saw this issue.
Yes! PILZ doesn't do any collision checks and the ValidateSolution adapter is necessary to make sure that the result is collision free. Previously this functionality was included by default and couldn't be easily turned on and off (for other planners it doesn't necessarily make sense to do a second collision check after planning).
Happy to hear you got it solved :+1: