moveit/moveit2 #2729: Rosdep keys not found for jammy

Shobuj-Paul, 2024-03-06:
### Description

Dependencies could not be resolved for certain packages (urdfdom and rsl).
Keys might be outdated.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source install
* Branch: main
* Cyclone DDS

### Steps to reproduce
Run the following command in the moveit source install workspace.
```bash
rosdep install --from-paths src --ignore-src -r -y
```

### Expected behaviour
Rosdep should install all dependencies with the message `#All required rosdeps installed successfully`.

### Actual behaviour
Rosdep shows an error message.

### Backtrace or Console output

```bash
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
pick_ik: No definition of [rsl] for OS version [jammy]
chomp_motion_planner: No definition of [rsl] for OS version [jammy]
moveit_kinematics: No definition of [rsl] for OS version [jammy]
moveit_core: No definition of [urdfdom_headers] for OS version [jammy]
rviz_marker_tools: No definition of [urdfdom_headers] for OS version [jammy]
moveit_planners_stomp: No definition of [rsl] for OS version [jammy]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

gavanderhoorn, 2024-03-07:
Isn't this caused by [the migration of Rolling to 24.04](https://discourse.ros.org/t/preparing-ros-2-rolling-for-the-transition-to-ubuntu-24-04/35673)?

Shobuj-Paul, 2024-03-07:
> Isn't this caused by [the migration of Rolling to 24.04](https://discourse.ros.org/t/preparing-ros-2-rolling-for-the-transition-to-ubuntu-24-04/35673)?

That makes a lot of sense. Thanks.
I guess we would need to wait till the migration is complete for Rolling to be stable again.

closed 2024-06-27 by github-actions[bot]