moveit/moveit2 #2731: `moveit_servo` provides duplicate `/get_planning_scene` service server to `move_group`

mrjogo, 2024-03-08:
### Description

I was running into issues where objects I'd added to the planning scene (using `PlanningSceneInterface`) were disappearing, and discovered that running `ros2 service call /get_planning_scene moveit_msgs/srv/GetPlanningScene` would sometimes return a response with the objects and sometimes without. I narrowed the problem down to only happening when I launched `moveit_servo`, and by running `ros2 node info /servo_node_private_94365815282960` discovered that it was providing a `/get_planning_scene` service server in addition to the `move_group`. I'm assuming the issue is that there's two servers/planning scenes running, and any calls will arbitrarily hit one or the other of them.

I could namespace the servo_node, but it seems like the default behavior shouldn't cause problems.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Binary
* If binary, which release version? 2.5.5-1jammy.20240126.074529
* If source, which branch? N/A
* Which RMW (Fast DDS or Cyclone DDS)? Default for Humble

### Steps to reproduce
The MoveIt launchfile I'm using is based on https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_moveit_config/launch/ur_moveit.launch.py, and the relevant node that causes problems is:

```python
servo_yaml = load_yaml("ur_moveit_config", "config/ur_servo.yaml")
servo_params = {"moveit_servo": servo_yaml}
servo_node = Node(
    package="moveit_servo",
    executable="servo_node_main",
    parameters=[
        servo_params,
    ],
    output="screen",
)
```

### Expected behaviour
The servo node shouldn't clobber the move_group's planning scene

### Actual behaviour
Two dueling planning scenes

### Backtrace or Console output

N/A

mrjogo, 2024-03-08:
This is possibly root cause of https://github.com/ros-planning/moveit2/issues/1775

sea-bass, 2024-03-10:
Thanks for reporting this!

Does the issue stand if you set the paremeter `is_primary_planning_scene_monitor: false` -- for example, here:
https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/config/panda_simulated_config.yaml#L37

mrjogo, 2024-05-11:
Can confirm adding `is_primary_planning_scene_monitor: false` fixes the problem. Thanks!

closed 2024-05-11 by mrjogo