moveit/moveit2 #2734: Launch Error

aagiser, 2024-03-10:
### Description
I have a problem with launcher. This problem is in source build installation, not binary. When I use binary, it works correctly. But when I install MoveIt 2 from source, it doesn't work. Note, tutorials launch files works without problems.

### My environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build branch humble
* Cyclone DDS

### Steps to reproduce
I created a moveit package using setup_assistant. Its name is moveit_6r. After building and sourcing the workspace, I launched the launch file.
`$ ros2 launch moveit_6r demo.launch.py`

### Expected behaviour
Opening RViz2 with the robot
It is correct when I install MoveIt 2 from binary!

### Actual behaviour
An error occured.

### Backtrace or Console output
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'

aagiser, 2024-03-10:
The output of
`ros2 launch moveit2_package demo.launch.py --debug`
is:

```
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/aagiser/.ros/log/2024-03-10-09-36-20-886303-DP5540-5655
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7bc621cb04c0>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7bc621cb04c0>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7bc6231653c0>'
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 69, in try_get_launch_description_without_context
    return self._get_launch_description(expanded_location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/aagiser/FUM6R-WS/install/moveit2_package/share/moveit2_package/launch/move_group.launch.py", line 7, in generate_launch_description
    return generate_move_group_launch(moveit_config)
  File "/home/aagiser/ws_moveit2/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py", line 203, in generate_move_group_launch
    default_value=moveit_config.move_group_capabilities["capabilities"],
KeyError: 'capabilities'

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: 'capabilities'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> exception=KeyError('capabilities') created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.566 seconds
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
    raise completed_tasks_exceptions[0]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/actions/include_launch_description.py", line 148, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/aagiser/FUM6R-WS/install/moveit2_package/share/moveit2_package/launch/move_group.launch.py", line 7, in generate_launch_description
    return generate_move_group_launch(moveit_config)
  File "/home/aagiser/ws_moveit2/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py", line 203, in generate_move_group_launch
    default_value=moveit_config.move_group_capabilities["capabilities"],
KeyError: 'capabilities'

[... cut ...]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7bc6231653f0>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7bc6231653f0>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7bc62165acb0>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7bc6231653f0>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7bc623165d80>'
```

BillyMazotti, 2024-03-12:
Getting the same error on iron too, my environment:
- ROS Distro: Iron
- OS Version: Ubuntu 22.04
- Source build branch iron

mink007, 2024-03-13:
any luck with this issue?

I am having the same issue when launching "ros2 launch <robot config> demo.launch.py rviz_tutorial:=true "

mink007, 2024-03-14:
when i install moveit2 from build from source, then i don't get this error "[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'".
But it is causing the robot model to not load in rviz.

i was looking for previous version number than "ros-iron-moveit/jammy,now 2.8.0-1jammy.20240213.190919 amd64 [installed]"
but was unable to figure out which the previous version number is. Was thinking of installing an older version to get around the issue. 

**********
apt list -a ros-iron-moveit
Listing... Done
ros-iron-moveit/jammy,now 2.8.0-1jammy.20240213.190919 amd64 [installed]
**********

cananella, 2024-04-15:
I also had a similar problem
`kairos@kairos-ubuntu22:~$ ros2 launch mycobot_moveit demo.launch.py --debug
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/kairos/.ros/log/2024-04-15-14-29-22-598832-kairos-ubuntu22-109096
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f45a82b0400>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f45a82b0400>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f45a973af80>'
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 69, in try_get_launch_description_without_context
    return self._get_launch_description(expanded_location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/kairos/ros2WS/install/mycobot_moveit/share/mycobot_moveit/launch/move_group.launch.py", line 7, in generate_launch_description
    return generate_move_group_launch(moveit_config)
  File "/home/kairos/ros2WS/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py", line 204, in generate_move_group_launch
    default_value=moveit_config.move_group_capabilities["capabilities"],
KeyError: 'capabilities'

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: 'capabilities'
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> exception=KeyError('capabilities') created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.363 seconds
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
    raise completed_tasks_exceptions[0]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/actions/include_launch_description.py", line 148, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
[... cut ...]
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f45a9738a90>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f45a97385b0>'
`
I don't know if it's the right solution, but I solved it by editing the moveit_configs_utils code.

at ~/ros_ws/src/moveit2/moveit_configs_utils/moveit_configs_utils/launches.py  line203
[image]

default_value=moveit_config.move_group_capabilities["capabilities"],
-> default_value=moveit_config.move_group_capabilities,

[image]

Vincenzo-Saffioti, 2024-05-14:
I also have the same problem and could not solve it with the solutions provided so far, any other ideas?

Schteve-Earl, 2024-05-30:
Also getting this issue. Here's my console output, with and without the debug flags:
```
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'
```

```
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x717c5e501810>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x717c5e501810>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x717c5e8ea2c0>'
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 69, in try_get_launch_description_without_context
    return self._get_launch_description(expanded_location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/ros/ros2_ws/install/ur10e_moveit_config/share/ur10e_moveit_config/launch/move_group.launch.py", line 7, in generate_launch_description
    return generate_move_group_launch(moveit_config)
  File "/home/ros/ros2_ws/build/moveit_configs_utils/moveit_configs_utils/launches.py", line 203, in generate_move_group_launch
    default_value=moveit_config.move_group_capabilities["capabilities"],
KeyError: 'capabilities'

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: 'capabilities'
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> exception=KeyError('capabilities') created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.615 seconds
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
    raise completed_tasks_exceptions[0]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
[... cut ...]
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x717c5e500820>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x717c5e500820>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x717c5e12a380>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x717c5e500820>' ✓ '
```
I'm on ros2 humble and using I'm using a ur10e moveit config that I just created, with no additions to the urdf. I did not set virtual joints or an end effector.

Schteve-Earl, 2024-05-30:
> > Also getting this issue. Here's my console output, with and without the debug flags:
> > ```
> > [INFO] [launch]: Default logging verbosity is set to INFO
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > [ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ```
> > [INFO] [launch]: Default logging verbosity is set to DEBUG
> > [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x717c5e501810>'
> > [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x717c5e501810>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x717c5e8ea2c0>'
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > [DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 69, in try_get_launch_description_without_context
> >     return self._get_launch_description(expanded_location)
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
> >     return get_launch_description_from_python_launch_file(location)
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
> >     return getattr(launch_file_module, 'generate_launch_description')()
> >   File "/home/ros/ros2_ws/install/ur10e_moveit_config/share/ur10e_moveit_config/launch/move_group.launch.py", line 7, in generate_launch_description
> >     return generate_move_group_launch(moveit_config)
> >   File "/home/ros/ros2_ws/build/moveit_configs_utils/moveit_configs_utils/launches.py", line 203, in generate_move_group_launch
> >     default_value=moveit_config.move_group_capabilities["capabilities"],
> > KeyError: 'capabilities'
> > 
> > [DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: 'capabilities'
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> exception=KeyError('capabilities') created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.615 seconds
> > [DEBUG] [launch]: An exception was raised in an async action/event
> > [DEBUG] [launch]: Traceback (most recent call last):
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
> >     raise completed_tasks_exceptions[0]
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
> >     await self.__process_event(next_event)
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
> >     visit_all_entities_and_collect_futures(entity, self.__context))
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
> >     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
> >     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
> >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
> >     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
> >   [Previous line repeated 1 more time]
[... 80 lines cut ...]
[... cut ...]
        executable="move_group",
        commands_file=str(moveit_config.package_path / "launch" / "gdb_settings.gdb"),
        output="screen",
        parameters=move_group_params,
        extra_debug_args=["--debug"],
        # Set the display variable, in case OpenGL code is used internally
        additional_env={"DISPLAY": os.environ["DISPLAY"]},
    )
    return ld
```
I removed the launch argument as well as the capabilities parameter, and it works fine now. Do you know what the parameter is supposed to control?

yuan0623, 2024-05-30:
> > > Also getting this issue. Here's my console output, with and without the debug flags:
> > > ```
> > > [INFO] [launch]: Default logging verbosity is set to INFO
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > [ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > ```
> > > [INFO] [launch]: Default logging verbosity is set to DEBUG
> > > [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x717c5e501810>'
> > > [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x717c5e501810>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x717c5e8ea2c0>'
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > [DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 69, in try_get_launch_description_without_context
> > >     return self._get_launch_description(expanded_location)
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
> > >     return get_launch_description_from_python_launch_file(location)
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
> > >     return getattr(launch_file_module, 'generate_launch_description')()
> > >   File "/home/ros/ros2_ws/install/ur10e_moveit_config/share/ur10e_moveit_config/launch/move_group.launch.py", line 7, in generate_launch_description
> > >     return generate_move_group_launch(moveit_config)
> > >   File "/home/ros/ros2_ws/build/moveit_configs_utils/moveit_configs_utils/launches.py", line 203, in generate_move_group_launch
> > >     default_value=moveit_config.move_group_capabilities["capabilities"],
> > > KeyError: 'capabilities'
> > > 
> > > [DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: 'capabilities'
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> > > Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> exception=KeyError('capabilities') created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.615 seconds
> > > [DEBUG] [launch]: An exception was raised in an async action/event
> > > [DEBUG] [launch]: Traceback (most recent call last):
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
> > >     raise completed_tasks_exceptions[0]
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
> > >     await self.__process_event(next_event)
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
> > >     visit_all_entities_and_collect_futures(entity, self.__context))
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
> > >     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
> > >     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
> > >   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
[... cut ...]
>         commands_file=str(moveit_config.package_path / "launch" / "gdb_settings.gdb"),
>         output="screen",
>         parameters=move_group_params,
>         extra_debug_args=["--debug"],
>         # Set the display variable, in case OpenGL code is used internally
>         additional_env={"DISPLAY": os.environ["DISPLAY"]},
>     )
>     return ld
> ```
> 
> I removed the launch argument as well as the capabilities parameter, and it works fine now. Do you know what the parameter is supposed to control?

no idea.... sorry. :)

ukicomputers, 2024-07-21:
+1 same issue here, just on Docker Rolling tutorial source image.

WilliamAStreamline, 2024-07-27:
Same issue and the above changes to launches.py only seem to add more issues. Using Ubuntu 24.04 and Jazzy

b-adkins, 2024-08-08:
I made a PR to revert the change: https://github.com/moveit/moveit2/pull/2912

In the meantime, `7ce2fa9` is the last working commit on the Humble branch.

claralunasa, 2024-08-14:
I have the exact same issues and the changes to the launch file haven't solved it for me

OneOneLiu, 2024-08-19:
> I also had a similar problem
> 
> ```
> 
> ~$ ros2 launch mycobot_moveit demo.launch.py --debug
> [DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
> [INFO] [launch]: All log files can be found below /home/kairos/.ros/log/2024-04-15-14-29-22-598832-kairos-ubuntu22-109096
> [INFO] [launch]: Default logging verbosity is set to DEBUG
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f45a82b0400>'
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f45a82b0400>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f45a973af80>'
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> [DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 69, in try_get_launch_description_without_context
>     return self._get_launch_description(expanded_location)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
>     return get_launch_description_from_python_launch_file(location)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
>     return getattr(launch_file_module, 'generate_launch_description')()
>   File "/home/kairos/ros2WS/install/mycobot_moveit/share/mycobot_moveit/launch/move_group.launch.py", line 7, in generate_launch_description
>     return generate_move_group_launch(moveit_config)
>   File "/home/kairos/ros2WS/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py", line 204, in generate_move_group_launch
>     default_value=moveit_config.move_group_capabilities["capabilities"],
> KeyError: 'capabilities'
> 
> [DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: 'capabilities'
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> exception=KeyError('capabilities') created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.363 seconds
> [DEBUG] [launch]: An exception was raised in an async action/event
> [DEBUG] [launch]: Traceback (most recent call last):
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
>     raise completed_tasks_exceptions[0]
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
>     await self.__process_event(next_event)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
>     visit_all_entities_and_collect_futures(entity, self.__context))
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   [Previous line repeated 1 more time]
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
>     sub_entities = entity.visit(context)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/action.py", line 108, in visit
>     return self.execute(context)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/actions/include_launch_description.py", line 148, in execute
>     launch_description = self.__launch_description_source.get_launch_description(context)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
>     self._get_launch_description(self.__expanded_location)
[... cut ...]
Thanks, this solution works for me. 

To provide more context, I am working on Ubuntu 22.04, ROS humble in a docker container. I installed moveit2 from binary and used setup assistant to create config files for a customized robot arm and gripper, the `ros2 launch my_robot_moveit_config demo.launch` worked well, but when I built moveit2 from source (to use Python binding), the same launch file produced the error
```bash{.line-numbers}
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'
```

mikeferguson, 2024-10-02:
I've opened https://github.com/moveit/moveit2/pull/3018 to fix this - looks like we need to backport to both humble and iron as well

closed 2024-10-16 by mikeferguson