moveit/moveit2 #2738: Robot Model not Loaded: MoveIt w/ RVIZ

xardarii, 2024-03-12:
Hey there! 
I have been working on a custom robotic arm and I am kinda new to this but I ran into this issue after I successfully generated the moveit configuration files using Moveit Setup Assistant. I am not able to load the robot model into RVIZ when i run the "ros2 launch robot_moveit_config demo.launch.py" command and the motion planning shows an error.

Environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary build?

### Steps to reproduce
1. Generate the moveit configuration files using MoveIt Setup Assistant
2. cd into the workspace
3. Colcon build all the packages in the workspace
4. Source install/setup.bash
5. run the command "ros2 launch robot_moveit_config demo.launch.py"

### Expected behaviour
My custom robot should load in RVIZ and I should be able to plan motion.

### Current behaviour
[image]

### Console output

```
[INFO] [launch]: All log files can be found below /home/xardarii-dev/.ros/log/2024-01-14-22-24-10-585017-xardarii-dev-28764
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [static_transform_publisher-1]: process started with pid [28765]
[INFO] [robot_state_publisher-2]: process started with pid [28767]
[INFO] [move_group-3]: process started with pid [28769]
[INFO] [rviz2-4]: process started with pid [28771]
[INFO] [ros2_control_node-5]: process started with pid [28773]
[INFO] [spawner-6]: process started with pid [28793]
[INFO] [spawner-7]: process started with pid [28795]
[static_transform_publisher-1] [INFO] [1705253051.694828534] [static_transform_publisher0]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'world' to 'base_link'
[robot_state_publisher-2] [INFO] [1705253051.706974300] [robot_state_publisher]: got segment Link_1
[robot_state_publisher-2] [INFO] [1705253051.707163909] [robot_state_publisher]: got segment Link_2
[robot_state_publisher-2] [INFO] [1705253051.707189631] [robot_state_publisher]: got segment Link_3
[robot_state_publisher-2] [INFO] [1705253051.707208474] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1705253051.707224933] [robot_state_publisher]: got segment world
[rviz2-4] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ros2_control_node-5] [WARN] [1705253051.775095848] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-5] [INFO] [1705253051.782069248] [resource_manager]: Loading hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1705253051.784188161] [resource_manager]: Initialize hardware 'FakeSystem' 
[ros2_control_node-5] [WARN] [1705253051.785090442] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1705253051.785136454] [resource_manager]: Successful initialization of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1705253051.785191671] [resource_manager]: 'configure' hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1705253051.785204327] [resource_manager]: Successful 'configure' of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1705253051.785214607] [resource_manager]: 'activate' hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1705253051.785223733] [resource_manager]: Successful 'activate' of hardware 'FakeSystem'
[move_group-3] [INFO] [1705253051.803078818] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.011324 seconds
[move_group-3] [INFO] [1705253051.803177999] [moveit_robot_model.robot_model]: Loading robot model 'robot'...
[ros2_control_node-5] [INFO] [1705253051.812724383] [controller_manager]: update rate is 100 Hz
[... 32 lines cut ...]
[ros2_control_node-5] [INFO] [1705253052.385752753] [controller_manager]: Configuring controller 'robotplanner_controller'
[ros2_control_node-5] [INFO] [1705253052.385974574] [robotplanner_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-5] [INFO] [1705253052.386009886] [robotplanner_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ros2_control_node-5] [INFO] [1705253052.386027180] [robotplanner_controller]: Using 'splines' interpolation method.
[ros2_control_node-5] [INFO] [1705253052.388414112] [robotplanner_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-5] [INFO] [1705253052.393240866] [robotplanner_controller]: Action status changes will be monitored at 20.00 Hz.
[rviz2-4] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-4]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[spawner-6] [INFO] [1705253052.431667556] [spawner_robotplanner_controller]: Configured and activated robotplanner_controller
[ros2_control_node-5] [INFO] [1705253052.548884046] [controller_manager]: Loading controller 'joint_state_broadcaster'
[INFO] [spawner-6]: process has finished cleanly [pid 28793]
[spawner-7] [INFO] [1705253052.592620385] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-5] [INFO] [1705253052.594728237] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-5] [INFO] [1705253052.594850254] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-7] [INFO] [1705253052.622476574] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[INFO] [spawner-7]: process has finished cleanly [pid 28795]
[ERROR] [move_group-3]: process has died [pid 28769, exit code -6, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_l35i_082 --params-file /tmp/launch_params_t8nz45fa'].
[rviz2-4] [ERROR] [1705253055.430979629] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-4] [INFO] [1705253055.490134361] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-4] [ERROR] [1705253065.609118373] [rviz]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
[rviz2-4] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
[rviz2-4]          at line 715 in ./src/model.cpp
[rviz2-4] [ERROR] [1705253065.621057817] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[rviz2-4] [ERROR] [1705253065.639710877] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded
```

nisathav, 2024-03-13:
hey there,
I am also running into same issue with the same environment setup. When I run `ros2 launch panda_moveit_config demo.launch.py` everything is working fine. I tried to create the same setup like panda_description folder to my custom robot named gimbal_description. 

but when I run `ros2 launch gimbal_moveit_config demo.launch.py`, I am unable to see the robot in the rviz. 

Here is my console output:
`nisat@pc:~/control_ws$ ros2 launch gimbal_moveit_config demo.launch.py 
[INFO] [launch]: All log files can be found below /home/nisat/.ros/log/2024-03-13-09-14-23-854622-pc-67690
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [static_transform_publisher-1]: process started with pid [67691]
[INFO] [robot_state_publisher-2]: process started with pid [67693]
[INFO] [move_group-3]: process started with pid [67695]
[INFO] [rviz2-4]: process started with pid [67697]
[INFO] [ros2_control_node-5]: process started with pid [67699]
[INFO] [spawner-6]: process started with pid [67701]
[INFO] [spawner-7]: process started with pid [67703]
[INFO] [spawner-8]: process started with pid [67705]
[static_transform_publisher-1] [INFO] [1710301464.286382374] [static_transform_publisher0]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'world' to 'gimbal_link0'
[robot_state_publisher-2] [WARN] [1710301464.292067896] [kdl_parser]: The root link gimbal_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-2] [INFO] [1710301464.292151022] [robot_state_publisher]: got segment gimbal_link0
[robot_state_publisher-2] [INFO] [1710301464.292198245] [robot_state_publisher]: got segment gimbal_link1
[robot_state_publisher-2] [INFO] [1710301464.292208410] [robot_state_publisher]: got segment gimbal_link2
[robot_state_publisher-2] [INFO] [1710301464.292216156] [robot_state_publisher]: got segment gimbal_link3
[robot_state_publisher-2] [INFO] [1710301464.292223662] [robot_state_publisher]: got segment gimbal_link4
[ros2_control_node-5] [WARN] [1710301464.301751798] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-5] [INFO] [1710301464.301958681] [resource_manager]: Loading hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1710301464.302559632] [resource_manager]: Initialize hardware 'FakeSystem' 
[ros2_control_node-5] [WARN] [1710301464.302614865] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1710301464.302627870] [resource_manager]: Successful initialization of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1710301464.302686872] [resource_manager]: 'configure' hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1710301464.302695282] [resource_manager]: Successful 'configure' of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1710301464.302701365] [resource_manager]: 'activate' hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1710301464.302706020] [resource_manager]: Successful 'activate' of hardware 'FakeSystem'
[move_group-3] [INFO] [1710301464.302846664] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00157355 seconds
[move_group-3] [INFO] [1710301464.302874931] [moveit_robot_model.robot_model]: Loading robot model 'gimbal'...
[move_group-3] [WARN] [1710301464.304392981] [moveit_robot_model.robot_model]: Could not identify parent group for end-effector 'hand'
[move_group-3] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[move_group-3]   what():  parameter 'robot_description_planning.joint_limits.gimbal_joint2.max_velocity' has invalid type: expected [double] got [integer]
[move_group-3] Stack trace (most recent call last):
[ros2_control_node-5] [INFO] [1710301464.304640355] [controller_manager]: update rate is 100 Hz
[ros2_control_node-5] [INFO] [1710301464.304766322] [controller_manager]: RT kernel is recommended for better performance
[... cut ...]
[rviz2-4] [ERROR] [1710301478.386862068] [rviz]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
[rviz2-4] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
[rviz2-4]          at line 715 in ./src/model.cpp
[rviz2-4] [ERROR] [1710301478.391110425] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[rviz2-4] [ERROR] [1710301478.398484035] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded
`

LuisOPortela, 2024-03-13:
Meanwhile, I have discovered some corrections that have fixed the problem for me. I don't know if this will help in your cases, but they were the following:

- In the config folder of the created package, a number of variables are as ints and not as doubles, which causes the "move_group" process to crash. To fix this, in the "joint_limits.yaml" add ".0" to the limits. For example, change the max velocity from 1 to 1.0. This error can be seen in the log of @nisathav posted above: 
```
[move_group-3]   what():  parameter 'robot_description_planning.joint_limits.gimbal_joint2.max_velocity' has invalid type: expected [double] got [integer]
``` 

- Change the "sensors_3d.yaml" file to simply be:
```
 sensors: []
```
 In my case, I don't think I will need it to be anything more than that, and it was making the "move_group" crash once again.

With the fixes above Rviz was able to load my robot model but I was still not able to plan movements. To fix this, I added some extra lines to the "joint_limits.yaml" file. Here's what it looks like for one joint:
```
........

joint_limits:
 Joint_1:
   has_velocity_limits: true
   max_velocity: 1.0
   has_accelaration_limits: true
   max_accelaration: 1.0 
   has_jerk_limits: false

 .......

Hope this helps! Let me know if it does :)

nisathav, 2024-03-14:
Hi there,
tried them and received the following error. Nothing is loading up now.

```
nisat@pc:~/control_ws$ ros2 launch gimbal_moveit demo.launch.py 
[INFO] [launch]: All log files can be found below /home/nisat/.ros/log/2024-03-14-13-03-21-509953-pc-39842
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'
```

LuisOPortela, 2024-03-14:
I never had that issue @nisathav, so I don't know how to help you :(
It's also strange because that error seems to be before the fixes I mentioned interfere with the launch.
In your place, I would try to check if I changed anything that broke the launch.
But apparently, more people are having that issue aswell: https://github.com/ros-planning/moveit2/issues/2734

mink007, 2024-03-14:
@nisathav if you install moveit2 from source (not binary) then do you see the issue resolved?

mink007, 2024-03-14:
@nisathav please see if this https://github.com/ros-planning/moveit2/pull/2587 fixes 
********
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'
********

mink007, 2024-03-15:
~/ws_moveit/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py

added highlighted yellow line in above file 
[image]

[image]

resolves this issue: "[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'"

gitzhou2006go, 2024-03-26:
> ~/ws_moveit/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py
> 
> added highlighted yellow line in above file <img alt="image" width="308" src="https://private-user-images.githubusercontent.com/29087516/313061001-923466e1-a2a9-4b44-95ba-e3fb481e3742.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE0MTU2NzEsIm5iZiI6MTcxMTQxNTM3MSwicGF0aCI6Ii8yOTA4NzUxNi8zMTMwNjEwMDEtOTIzNDY2ZTEtYTJhOS00YjQ0LTk1YmEtZTNmYjQ4MWUzNzQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI2VDAxMDkzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM1ZTJiZDk5ZDJmNDgyN2E5MmJjOGJlYWY1Zjg4NzQxNjJmZDk3Y2M1OGIwMjUyOTQ1MDc3MmIxMWE1MTZiNzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PHsCDQjfJM-6d_yGI_q4Zem1NXN3qTpHPuOJGupfzik">
> 
> <img alt="image" width="152" src="https://private-user-images.githubusercontent.com/29087516/313060856-316ae946-5af4-4167-83b0-339f5580b776.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE0MTU2NzEsIm5iZiI6MTcxMTQxNTM3MSwicGF0aCI6Ii8yOTA4NzUxNi8zMTMwNjA4NTYtMzE2YWU5NDYtNWFmNC00MTY3LTgzYjAtMzM5ZjU1ODBiNzc2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI2VDAxMDkzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2NzQ5NzI2NTljZDA3NGRlODI5YzNlOGY5NGE4YjEyNzY5MzUxNjRiNGE5NGM5YTZiNTMwYzFkMjhhZWJjMDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xSBKYXOOcgAvt9K9gzknTySc1sBfKNHnBZdALRkeSYI">
> resolves this issue: "[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'"

Thank you for the correct solution! It resolved my issue.

jixiexiaoli, 2024-05-15:
Hello, I met the same problem, how to solve it, thank you very much!

mink007, 2024-05-15:
> ~/ws_moveit/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py
>
> added highlighted yellow line in above file <img alt="image" width="308" [...]

mink007, 2024-05-15:
> ~/ws_moveit/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py
>
> added highlighted yellow line in above file
> <img alt="image" width="308" [...]

jixiexiaoli, 2024-05-15:
> > ~/ws_moveit/安装/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/launches.py
> > 在上面的文件中添加了突出显示的黄线
> > <img alt="图像" width="308" src="https://private-user-images.githubusercontent.com/29087516/313061001-923466e1-a2a9-4b44-95ba-e3fb481e3742.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3MzgxNzQsIm5iZiI6MTcxNTczNzg3NCwicGF0aCI6Ii8yOTA4NzUxNi8zMTMwNjEwMDEtOTIzNDY2ZTEtYTJhOS00YjQ0LTk1YmEtZTNmYjQ4MWUzNzQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE1VDAxNTExNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiMjg3Yjg5YWExM2FiNDQ1ZTk4YWU5NDdjNzJmMjYyMWU0OWNjYzIwNWZmZGU0MGJiZTc3ODI4MWQ1ZTA5M2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.d29_dHKKY9u3Web-vPyURYZdl-s_mYvdhyUtvIoMC4w">
> > <img alt="图像" width="152" src="https://private-user-images.githubusercontent.com/29087516/313060856-316ae946-5af4-4167-83b0-339f5580b776.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3MzgxNzQsIm5iZiI6MTcxNTczNzg3NCwicGF0aCI6Ii8yOTA4NzUxNi8zMTMwNjA4NTYtMzE2YWU5NDYtNWFmNC00MTY3LTgzYjAtMzM5ZjU1ODBiNzc2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE1VDAxNTExNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ0MTIyNDc4ZWNlNzEzZjIxYjc0NzEyZGY5NWVjODkwZDQwMzQ2M2Y4MmYwOWNiMTBlZWJiMWFmNGJlZjdlZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sGQZX4XXNT8ZjMGTBJHHH4Opj2P_yQr8kXE7JS60Xas">解决了此问题：“[ERROR] [launch]：在启动中捕获异常（请参阅调试以获取回溯）：”capabilities“”

Thank you for your help.I've fixed this capabiliti issue, but I still can't load my model after opening it.Do you have a solution for this erro.
[rviz2-3] [ERROR] [1715739073.071318699] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[rviz2-3] [ERROR] [1715739073.101394782] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded

jixiexiaoli, 2024-05-15:
[image]

jixiexiaoli, 2024-05-15:
> Thank you for your help.I've fixed this capabiliti issue, but I still can't load my model after opening it.Do you have a solution for this erro. [rviz2-3] [ERROR] [1715739073.071318699] [...]

[image]

mink007, 2024-05-15:
see if the location(unix path) of the model file (xacro/urdf) is as expected.

AnishNavalgund, 2024-06-04:
I am also getting the same error. Please advise! 

Below are the logs.

INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [41805]
[INFO] [move_group-2]: process started with pid [41807]
[INFO] [rviz2-3]: process started with pid [41809]
[move_group-2] [WARN] [1717542851.019722879] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[robot_state_publisher-1] [INFO] [1717542851.025612211] [robot_state_publisher]: got segment base
[robot_state_publisher-1] [INFO] [1717542851.025666171] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1717542851.025672141] [robot_state_publisher]: got segment flange
[robot_state_publisher-1] [INFO] [1717542851.025675891] [robot_state_publisher]: got segment link_1
[robot_state_publisher-1] [INFO] [1717542851.025679141] [robot_state_publisher]: got segment link_2
[robot_state_publisher-1] [INFO] [1717542851.025682171] [robot_state_publisher]: got segment link_3
[robot_state_publisher-1] [INFO] [1717542851.025685371] [robot_state_publisher]: got segment link_4
[robot_state_publisher-1] [INFO] [1717542851.025688451] [robot_state_publisher]: got segment link_5
[robot_state_publisher-1] [INFO] [1717542851.025691671] [robot_state_publisher]: got segment link_6
[robot_state_publisher-1] [INFO] [1717542851.025694721] [robot_state_publisher]: got segment tool0
[move_group-2] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_PARSING_TEXT ErrorID=8 (0x8) Line number=1
[move_group-2]          at line 715 in ./src/model.cpp
[move_group-2] [ERROR] [1717542851.039141526] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[move_group-2] [ERROR] [1717542851.044008478] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded
[move_group-2] [ERROR] [1717542851.044178098] [moveit.ros_planning_interface.moveit_cpp]: Planning scene not configured
[move_group-2] [FATAL] [1717542851.044188918] [moveit.ros_planning_interface.moveit_cpp]: Unable to configure planning scene monitor
[move_group-2] terminate called after throwing an instance of 'std::runtime_error'
[move_group-2]   what():  Unable to configure planning scene monitor
[move_group-2] Stack trace (most recent call last):
[move_group-2] #12   Object "", at 0xffffffffffffffff, in 
[move_group-2] #11   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x5db3ee3e6724, in 
[move_group-2] #10   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7863f2629e3f]
[move_group-2] #9    Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7863f2629d8f]
[move_group-2] #8    Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x5db3ee3e52c2, in 
[move_group-2] #7    Object "/opt/ros/humble/lib/libmoveit_cpp.so.2.5.5", at 0x7863f3220dd1, in 
[move_group-2] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7863f2aae4d7, in __cxa_throw
[move_group-2] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7863f2aae276, in std::terminate()
[move_group-2] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7863f2aae20b, in 
[move_group-2] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7863f2aa2b9d, in 
[move_group-2] #2    Source "./stdlib/abort.c", line 79, in abort [0x7863f26287f2]
[move_group-2] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7863f2642475]
[move_group-2] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[move_group-2]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[move_group-2]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7863f26969fc]
[move_group-2] Aborted (Signal sent by tkill() 41807 1000)
[ERROR] [move_group-2]: process has died [pid 41807, exit code -6, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args -r __node:=move_group --params-file /tmp/launch_params_u4lpxe4y'].
[rviz2-3] [INFO] [1717542851.639251333] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1717542851.639355624] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1717542851.672795515] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-3]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-3] [ERROR] [1717542854.769990326] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-3] [INFO] [1717542854.798718286] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-3] [WARN] [1717542854.843135251] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will  [... cut ...]
[rviz2-3] [ERROR] [1717542864.892760455] [rviz2]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10,000000 seconds.
[rviz2-3] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
[rviz2-3]          at line 715 in ./src/model.cpp
[rviz2-3] [ERROR] [1717542864.917802914] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[rviz2-3] [ERROR] [1717542864.932787459] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded

jixiexiaoli, 2024-06-05:
> I am also getting the same error. Please advise!
>
> Below are the logs.
>
> INFO] [launch]: Default logging verbosity is set to INFO [INFO] [robot_state_publisher-1]: process started with pid [41805] [...]

Change the speed value in URDF file to double type, for example, change V=1 to V=1.000001

AnishNavalgund, 2024-06-05:
@jixiexiaoli Do you mean in the yaml file? I have done that and I still get the same error messages.

jixiexiaoli, 2024-06-05:
> @jixiexiaoli Do you mean in the yaml file? I have done that and I still get the same error messages.

URDF file

vovaberdibek, 2024-07-12:
We having the same error : [rviz2-1] Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
[rviz2-1] [INFO] [1720775001.712050596] [moveit_rdf_loader.rdf_loader]: Unable to parse URDF
 please someone knows how to fix it ?

calupino, 2024-09-04:
So I ran into this issue when trying to _add_ joint limits to an SRDF for a custom arm I'm building. The default joint_limits.yaml file has all the limits set to "false" and the maximum values set to "0". The model loads just fine. However if I want to set the limits to "true" and assign them a value of "1.0" then I get the same error everyone else in this thread is getting where it's unable to parse the SRDF.

mikeferguson, 2024-10-23:
#2803 fixes the invalid cast - but was never back ported to humble - I just opened https://github.com/moveit/moveit2/pull/3038 to do that. There are a few other unrelated failures posted as comments above, but I think those would be better debugged in their own tickets

closed 2024-10-23 by mikeferguson