moveit/moveit2 #2741: Pilz Industrial Motion Planner blend generates points with same time_from_start

TheSpeedM, 2024-03-14:
### Description

Using `pilz_industrial_motion_planner` to generate a motionsequence with a blend radius generates (multiple) points with the same value for `RobotTrajectory.joint_trajectory.points[].time_from_start`. Looks like it generates two pairs of wrong points per point with a blend. 

It does generate and execute the trajectory when blend radius is set to 0.0. Had the problem using the  [Universal_Robots_ROS2_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver) and was able to reproduce it with  the [MoveIt example robot](https://moveit.picknik.ai/main/doc/how_to_guides/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#examples) (used for this bug report).

Also ran into [this issue](https://github.com/ros-planning/moveit/issues/3553) from MoveIt1 while trying to reproduce. Looks related.

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04.4 LTS (native)
* Build from source following [Getting Started](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html), must be v2.9.0 then.

### Steps to reproduce
```
ros2 launch moveit_resources_panda_moveit_config demo.launch.py
```
Send MoveGroupSequence goal to `/sequence_move_group` with a blend radius using the following Python code.
```Python
import rclpy
from rclpy.action import ActionClient
from rclpy.node import Node

from std_msgs.msg import Header
from shape_msgs.msg import SolidPrimitive
from geometry_msgs.msg import Pose, Quaternion
from moveit_msgs.action import MoveGroupSequence, ExecuteTrajectory
from moveit_msgs.msg import MotionPlanRequest, PlanningOptions, Constraints, PositionConstraint, \
                            RobotState, BoundingVolume, OrientationConstraint, MotionSequenceRequest, MotionSequenceItem

def generate_orient_constraint(x: float, y: float, z: float, w: float, margin: float, z_margin: float = None):
    con = [OrientationConstraint()]
    con[0].orientation = Quaternion()
    con[0].header = Header()

    con[0].header.frame_id = 'world'
    con[0].weight = 1.0
    con[0].link_name = 'panda_hand'
    con[0].orientation.x = x
    con[0].orientation.y = y
    con[0].orientation.z = z
    con[0].orientation.w = w
    con[0].absolute_x_axis_tolerance = margin
    con[0].absolute_y_axis_tolerance = margin
    con[0].absolute_z_axis_tolerance = z_margin if z_margin else margin

    return con

def generate_pos_constraint(xyz: list, xyz_margin: float, orientation_margin: float) -> Constraints:
    con = Constraints()
    con.position_constraints = [PositionConstraint()]
    con.position_constraints[0].header = Header()
    con.position_constraints[0].constraint_region = BoundingVolume()
    con.position_constraints[0].constraint_region.primitives = [SolidPrimitive()]
    con.position_constraints[0].constraint_region.primitive_poses = [Pose()]

    # con.name = 'Frizo'
    con.position_constraints[0].header.frame_id = 'world'
    # con.position_constraints[0].link_name = 'wrist_3_link'
    con.position_constraints[0].link_name = 'panda_hand'
    con.position_constraints[0].weight = 1.0

    con.position_constraints[0].constraint_region.primitives[0].type = SolidPrimitive.SPHERE
    con.position_constraints[0].constraint_region.primitives[0].dimensions = [xyz_margin]

    con.position_constraints[0].constraint_region.primitive_poses[0].position.x = xyz[0]
    con.position_constraints[0].constraint_region.primitive_poses[0].position.y = xyz[1]
    con.position_constraints[0].constraint_region.primitive_poses[0].position.z = xyz[2]

    con.position_constraints[0].constraint_region.primitive_poses[0].orientation.x = 0.0
[... 151 lines cut ...]
[ros2_control_node-5] [INFO] [1710412387.951204233] [panda_arm_controller]: Received new action goal
[ros2_control_node-5] [ERROR] [1710412387.951283595] [panda_arm_controller]: Time between points 122 and 123 is not strictly increasing, it is 12.200000 and 12.200000 respectively
[move_group-4] [INFO] [1710412387.953570271] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: panda_arm_controller started execution
[move_group-4] [WARN] [1710412387.953620473] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-4] [ERROR] [1710412387.954978583] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-4] [ERROR] [1710412387.955033649] [move_group.moveit.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller panda_arm_controller
[move_group-4] [INFO] [1710412387.955058316] [move_group.moveit.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
```

### Dirty fix
Set `plan_only` to true and pass the planned_trajectories to the following function before execution. This shifts the generated points with the same time away from eachother by one ns. Trajectory does execute when doing this.
```Python
def shift_robottrajectory(trajectory: RobotTrajectory) -> None:
    time_shift_ns = 0

    for index, point in enumerate(trajectory.joint_trajectory.points):
        if index == 0:
            continue

        point.time_from_start.nanosec += time_shift_ns

        if trajectory.joint_trajectory.points[index-1].time_from_start == point.time_from_start:
            point.time_from_start.nanosec += 1
            time_shift_ns += 1
```

AGuthmann, 2024-03-19:
Hey,
Had the same problem and had to fixed it first with a small function like yours.
Further looking into it showed me, that the problematic points always were the first and last point inside the blend radius.

!!BE CAREFUL before you test this script with a real robot, In my case not only the time was off, but the velocity and acceleration at these points too, so the robot moves in an abrupt and jittery way when transitioning!!

In the MoveIt Humble Docs I found ['Time Parameterization'](https://moveit.picknik.ai/humble/doc/examples/time_parameterization/time_parameterization_tutorial.html) and the Time-Optimal Parameter Generation method which fixed time, acceleration and velocity for me. 
Unfortunately I only found a C++ version of this. Alternatively you can look into Ruckig-Smoothing of the trajectory.
Hope I could point you in the right direction

TheSpeedM, 2024-03-19:
First of all thank your proposed solution and heads-up. I haven't tested this on a real robot yet, though this is something that I will make a priority because of your experience.

> In the MoveIt Humble Docs I found 'Time Parameterization' and the Time-Optimal Parameter Generation method which fixed time, acceleration and velocity for me.

The problem with this solution is that adding the `AddTimeOptimalParameterization` request adapter to `pilz_industrial_motion_planner.yaml` doesn't work in my case. This is probably the same problem [this commenter](https://github.com/ros-planning/moveit/issues/2905#issuecomment-1282874903) in ros-planning/moveit#2905 was having; I can't use a blend with this request adapter enabled.

I might have to look into other ways of smoothing the trajectory.

AGuthmann, 2024-03-19:
Oh yeah, forgot to mention that the adapter didn't work for me either.

I had to manually create a class object of `TimeOptimalTrajectoryGeneration` and recalculate the trajectory with the `computeTimeStamps` function. For this to work I had to change the moveit-RobotTrajectory into a robot_trajectory-RobotTrajectory, get the new waypoints and convert back.
But this was all in C++ and unfortunately I don't know if there's a python class for the TOTG :/

ilijaljubicic, 2024-03-20:
@AGuthmann could you be so kind to  share a snippet of your code showcasing the approach you took?

I am facing same problem, in my case , the code is in C++.

AGuthmann, 2024-03-21:
sure, here you go:

```cpp
#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit_core/moveit/trajectory_processing/time_optimal_trajectory_generation.h>

void optimize_trajectory(moveit_msgs::msg::RobotTrajectory &trajectory, double max_velocity_scaling_factor, double max_acceleration_scaling_factor){
  moveit::core::RobotState robot_state(move_group_interface_->getRobotModel());
  robot_trajectory::RobotTrajectory robot_trajectory(move_group_interface_->getRobotModel(), robot_state.getJointModelGroup(PLANNING_GROUP)->getName());
  
  robot_trajectory.setRobotTrajectoryMsg(robot_state, trajectory); // convert moveit_msgs::msg::RobotTrajectory into robot_trajectory::RobotTrajectory

  trajectory_processing::TimeOptimalTrajectoryGeneration totg; // Create Optimization Object
  totg.computeTimeStamps(robot_trajectory, 0.1, 0.1); // First optimization with default scalings
  totg.computeTimeStamps(robot_trajectory,max_velocity_scaling_factor, max_acceleration_scaling_factor); // Second optimization with user chosen scalings
  RCLCPP_INFO(logger_, "First Iteration: %d, Second Iteration: %d", first, second);

  robot_trajectory.getRobotTrajectoryMsg(trajectory); // convert robot_trajectory::RobotTrajectory back to moveit_msgs::msg::RobotTrajectory
}
```

I had to optimize twice because sometimes longer trajectories weren't optimized at all. Couldn't figure out where this comes from...

Additionally if you have 180° turns in your trajectory the optimization will probably fail, because these aren't implemented / supported yet. For my tests it was sufficient to offset the Points before or after the 180° turn by 1mm perpendicular to the movement of the trajectory.

TheSpeedM, 2024-03-21:
I have tested the 'dirty solution' on a real robot and it indeed jitters a lot at these points. I also found that Pilz normally creates points with an interval of 0.1 seconds, so i opted to offset the point with 0.1 seconds instead of 1 ns and it works very smoothly.

```python
def shift_robottrajectory(trajectory: RobotTrajectory) -> None:
    time_shift_ns = 0

    for index, point in enumerate(trajectory.joint_trajectory.points):
        if index == 0:
            continue

        point.time_from_start.nanosec += time_shift_ns

        if trajectory.joint_trajectory.points[index-1].time_from_start == point.time_from_start:
            # Not sure if this works for seconds rollover but I didn't have any problem
            point.time_from_start.nanosec += int(1e8)
            time_shift_ns += int(1e8)
```

ilijaljubicic, 2024-03-25:
Thank you @AGuthmann .

AGuthmann, 2024-03-25:
@TheSpeedM just so I understand correctly: offsetting the point with identical `time_from_start` and all following points by 0.1 seconds fixed the jittery motion? That's nice to hear!

@ilijaljubicic you're welcome!

TheSpeedM, 2024-03-25:
@AGuthmann yes you're correct. The Pilz planner generates points at a 0.1s interval, except at the start and end of a blend. So the 'wrong' points - with the same time as the previous points - of a blend have to be offset by 0.1s and all of the following points have to be shifted too (so they won't overlap). This shifting adds up the more points there have to be corrected, see the diagram below.

[image]

gaspatxo, 2024-06-13:
Hi @TheSpeedM. Im trying to reproduce your steps but the code gets stuck on `self._plan_action_client.wait_for_server()`

Indeed, a `ros2 action list` shows there is no `/sequence_move_group` being advertised after launching the panda demo file.

I am using `humble`, could this be the issue?

gaspatxo, 2024-06-13:
Yes, it is https://github.com/moveit/moveit2/issues/2707

Danilrivero, 2024-06-27:
I have encountered the same issue as well. There is an ongoing issue at Moveit1 here https://github.com/moveit/moveit/issues/3583. 

I could solve it by doing the bypass commented here https://github.com/moveit/moveit/issues/3583#issuecomment-2035540755 and making some changes for ROS2:

```cpp
  // Ensure time_from_start is strictly increasing
  double epsilon = 0.01; 
  for (size_t i = 1; i < plan.trajectory_.joint_trajectory.points.size(); ++i)
  {
    auto& current_point = plan.trajectory_.joint_trajectory.points[i];
    auto& prev_point = plan.trajectory_.joint_trajectory.points[i - 1];

    if (current_point.time_from_start == prev_point.time_from_start)
    {
      // Shift current point by 0.1 seconds (100 milliseconds)
      current_point.time_from_start.sec = prev_point.time_from_start.sec;
      current_point.time_from_start.nanosec += 100000000; 
      RCLCPP_WARN(this->get_logger(), "Adjusted time_from_start to be strictly increasing at point %zu", i);
    }
  }
```

I do still rarely encounter the issues described in this https://github.com/moveit/moveit/issues/3553 with accelerations when working on the real robot only. 

Definitely solving that issue and being able to backport to MoveIt2 would be appreciated.

sea-bass, 2024-08-01:
@TheSpeedM I just ran into this and think I have a band-aid fix in https://github.com/moveit/moveit2/pull/2943 -- want to give it a try?

closed 2024-08-02 by sjahr

sea-bass, 2024-09-14:
Thanks for the reminder @Danilrivero -- mergify wasn't working properly at the time of this PR.

See https://github.com/moveit/moveit2/pull/3000 and https://github.com/moveit/moveit2/pull/3001!