moveit/moveit2 #2742: STOMP planning algorithm in HUMBLE

Jakubivan, 2024-03-14:
Reopened
https://github.com/ros-planning/moveit2/issues/2550

I see that STOMP planner is added in Iron, but not in humble.

If I wanted to use in humble, would it be as easy as coping the code from moveit_planners into my local source (humble)? are there any other nuances that I should be aware of?

angeelalg, 2024-04-16:
I am in a similar situation, I would like to be able to use STOMP in Foxy. By modifying moveit_planners could it work directly or does it require modifications in other parts of the repository?

Jakubivan, 2024-04-16:
> I am in a similar situation, I would like to be able to use STOMP in Foxy. By modifying moveit_planners could it work directly or does it require modifications in other parts of the repository?

Modifying `moveit_planners` worked for me. However, I was not able to plan under orientation constraints since it reached one of the stop conditions (mainly `num_iterations`). In the previous generation of ROS (ROS1 Noetic), I worked with a custom noise generator which was able to fulfill my requirements for planning under constraints. Sadly, this is not possible (or I do not know how) in ROS2 using rebuilt STOMP.

 I have cherry-picked all the commits from the main regarding MoveIt2 - https://github.com/Jakubivan/moveit2/tree/ivan/humble-w-stomp to my `Humble` branch.

In moveit config package, add file `stomp_planning.yaml` with the keys:

```
planning_plugin: stomp_moveit/StompPlanner
request_adapters: >-
  default_planner_request_adapters/AddTimeOptimalParameterization
  default_planner_request_adapters/ResolveConstraintFrames
  default_planner_request_adapters/FixWorkspaceBounds
  default_planner_request_adapters/FixStartStateBounds
  default_planner_request_adapters/FixStartStateCollision
  default_planner_request_adapters/FixStartStatePathConstraints

num_timesteps: 40
num_iterations: 10000
num_iterations_after_valid: 0
num_rollouts: 15
max_rollouts: 25
exponentiated_cost_sensitivity: 0.4
control_cost_weight: 0.1
delta_t: 0.1
path_marker_topic: 'path_marker'
```

antonio1matos, 2024-06-04:
Hi @Jakubivan 
I'm also using MoveIt Humble and i also wanted to use the planner stomp in this moveit.
So you basically just had to copy the folder stomp from: https://github.com/moveit/moveit2/tree/main/moveit_planners/stomp
and paste it in the folder: moveit_planners from your moveit2 workspace (which in my case is the path: ws_moveit2/src/moveit2/moveit_planners)?
After that you just did colcon build of the workspace and the moveit is already configured with this planner? And if so didn´t you had to change anything else from that workspace? Such as other files?
Regards
António Matos

Jakubivan, 2024-06-20:
> Hi @Jakubivan I'm also using MoveIt Humble and i also wanted to use the planner stomp in this moveit. So you basically just had to copy the folder stomp from: [...]

Sorry for the late response, I had rough past weeks. In short, you are right. I created a custom branch in my forked repo from the original one, see https://github.com/Jakubivan/moveit2/tree/ivan/humble-w-stomp

antonio1matos, 2024-06-25:
Hi @Jakubivan. I noticed that the changes that you made in that: https://github.com/Jakubivan/moveit2/tree/ivan/humble-w-stomp was the addition of the file stomp in the package moveit_planners but you also added the file: stomp_planning.yaml to the package: moveit2/moveit_configs_utils/default_configs. Right now i have the humble branch provided in here:  https://moveit.ros.org/install-moveit2/binary/ using the line code: sudo apt install ros-humble-moveit and also the moveit2_tutorials package: git clone --branch humble https://github.com/ros-planning/moveit2_tutorials from: https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html                                                                              I wanted to know if the only difference between the branch that i have and the custom branch that you create in here: https://github.com/Jakubivan/moveit2/tree/ivan/humble-w-stomp  is just the addition of the stomp planner or have you changed any other package from the provided branch from the moveit humble tutorials?                                                                                               Regards
António Matos

closed 2024-11-25 by github-actions[bot]