moveit/moveit2 #2744: Controlling two Robots simultaneously via Moveit

TE-2, 2024-03-17:
Hello everyone,
I have two Yaskawa robots, MotoMini and GP4, in a robot module, each controlled by a YRC1000micro controller, and I would like to control them simultaneously via ROS2 Humble. The robots should not work directly together, but their workspaces overlap. They should also be able to execute trajectories simultaneously so that one robot does not have to wait for the other.

The control of a single robot via Moveit works without problems, I used the [ros2-starter-for-yaskawa-robots repository](https://github.com/YaskawaEurope/ros2-starter-for-yaskawa-robots/tree/main) and adapted the ros2 package [motoXY](https://github.com/YaskawaEurope/ros2-starter-for-yaskawa-robots/tree/main/src/motoXY) to my two robots.

For the control of both robots I want a move_group node for each robot in a respective namespace e.g. (`/moto_mini_moveit `& `/moto_gp4_moveit`), because I read in a [forum post](https://answers.ros.org/question/362796/how-to-simulate-and-control-2-industrial-robots-with-moveit/) that the move group node is not able to execute two trajectories at the same time.

For this I have modified [xy_start.launch.py](https://github.com/YaskawaEurope/ros2-starter-for-yaskawa-robots/blob/main/src/motoXY/moto_moveit_config/launch/xy_start.launch.py) by passing `run_move_group_nodeX` the new name_space as an argument, as well as `joint_state_publisherX` and `robot_state_publisherX` and would like to start only these three nodes for the time being.
With the help of rqt_graph I was able to identify the incorrectly assigned topics and then wrote the suitable remappings. The direct mapping of actions is unfortunately not yet possible according to [this article](https://github.com/ros2/ros2/issues/1312#issuecomment-1705521109), so I used the workaround and remapped the corresponding topics.

The node `/moveit_simple_controller_manager`, for example, even had `/moto_mini_moveit/moto_mini/follow_joint_trajectory/_action/status` after the action topic. 

Customised `xy_start.launch.py`:

<details><summary>Click to expand</summary>

```python
import os
from launch import LaunchDescription
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory
from moveit_configs_utils import MoveItConfigsBuilder

import xacro

def generate_launch_description():
# =======================================================================
# NameSpace configuration
    
    robot_name_space        = "/moto_mini"
    name_space_moveit       = "/moto_mini_moveit"

# =======================================================================
    motoman_directory = "moveit_resources_moto_motomini_moveit_config"  #not Directory -> ROS Package
# URDF
    motoman_robot_description_config = xacro.process_file(
        os.path.join(
            get_package_share_directory(motoman_directory),
            "config",
            "motoman_motomini.urdf.xacro",
        ),
    )
    motoman_robot_description = {
        "robot_description": motoman_robot_description_config.toxml()
    }
# =======================================================================
# MoveGroupNode
    moveit_config = (
        MoveItConfigsBuilder("moveit_resources_moto_motomini")
        .robot_description(file_path="config/motoman_motomini.urdf.xacro")
        .robot_description_semantic(file_path="config/motoman_motomini.srdf")
        .trajectory_execution(file_path="config/moveit_controllers.yaml")
        .to_moveit_configs()
    )
    planning_scene_monitor_parameters = {
        "publish_planning_scene": True,
        "publish_geometry_updates": True,
        "publish_state_updates": True,
        "publish_transforms_updates": True,
        "publish_robot_description": True,
        "publish_robot_description_semantic": True,
    }
    # Start the actual move_group node/action server
    # robot
    run_move_group_nodeX = Node(
        package="moveit_ros_move_group",
        executable="move_group",
        namespace= name_space_moveit,
        output="screen",
        parameters=[
[... 124 lines cut ...]
[move_group-1] [INFO] [1710349288.907909092] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-1] [INFO] [1710349288.907929054] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-1] [INFO] [1710349288.907937800] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-1] [INFO] [1710349288.907985322] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1710349288.920418642] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1710349288.920457918] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-1] [INFO] [1710349288.920464835] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-1] [ERROR] [1710349288.920511311] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: moto_mini/follow_joint_trajectory
[move_group-1] [ERROR] [1710349288.920513743] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller moto_mini
[move_group-1] [INFO] [1710349288.920516026] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1710349288.920660570] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
```

</details>

It seems to fail because of the following error message:

```
[moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: moto_mini/follow_joint_trajectory
```

But I don't understand this, because according to rqt_graph and "ros2 node info ..." the `moveit_simple_controller_manager` is a client of `/moto_mini/follow_joint_trajectory/_action` ? (see ros graph picture above)

I would appreciate any helpful input.
Thanks

gavanderhoorn, 2024-03-17:
For context: Yaskawa-Global/motoros2#225.

gavanderhoorn, 2024-06-24:
Resolution posted here: https://github.com/Yaskawa-Global/motoros2/discussions/225#discussioncomment-9859474.