moveit/moveit2 #2748: `CurrentStateMonitor` does not correctly report attached collision objects, but `PlanningSceneMonitor` does.

sea-bass, 2024-03-18:
### Description

We had an issue with MoveIt Servo that was fixed in https://github.com/ros-planning/moveit2/pull/2747, where collision checking with attached objects was failing since the presence of the object was not correct.

Simply switching the line:

```cpp
robot_state_ = planning_scene_monitor_->getStateMonitor()->getCurrentState();
```

to

```cpp
robot_state_ = planning_scene_monitor_->getPlanningScene()->getCurrentState();
```

fixes the issue, so I wanted to file a bug tracking this mismatch.

sea-bass, 2024-05-03:
I think #2747 closes this.

mikeferguson, 2024-11-03:
Is this actually a bug? My understanding here is that:

 * CurrentStateMonitor just monitors the robot itself - the API doc is "Monitors the joint_states topic and tf to maintain the current state of the robot" - neither of those topics would provide information about attached objects.
 * PlanningScene is then the full planning scene, with attached objects and octomap, etc.

Perhaps we just need to update the API docs to be more clear about the differences between these two classes?

sea-bass, 2024-11-03:
Good point. I agree that should be sufficient.

mikeferguson, 2024-11-03:
Probably also a good idea to do a quick search of the codebase and make sure there aren't other cases where the wrong function has been used...