moveit/moveit2 #2749: KDL parameters fail to build on latest version of `generate_parameter_library`

sea-bass, 2024-03-18:
### Description

When building the KDL plugin in the `moveit_kinematics` package, I see build errors because the generated header file from the parameters places two `.` characters in a row, which is invalid C++.

This stems from the parameter map defined in https://github.com/ros-planning/moveit2/blob/main/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_parameters.yaml#L8-L16

I believe these were caused by this recent PR to generate_parameter_library: https://github.com/PickNikRobotics/generate_parameter_library/pull/166

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? Source, from `main`
* Which RMW (Fast DDS or Cyclone DDS)? Any

### Backtrace or Console output

```
--- stderr: moveit_kinematics
  In file included from /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/kdl_kinematics_plugin.h:42,
                   from /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:37:
  /home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘void kdl_kinematics::ParamListener::refresh_dynamic_parameters()’:
  /home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:146:36: error: expected unqualified-id before ‘.’ token
    146 |       auto& entry = updated_params..joints_map[value_1];
        |                                    ^
  /home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘rcl_interfaces::msg::SetParametersResult kdl_kinematics::ParamListener::update(const std::vector<rclcpp::Parameter>&)’:
  /home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:217:32: error: expected unqualified-id before ‘.’ token
    217 |                 updated_params..joints_map[value_1].weight = param.as_double();
        |                                ^
  /home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘void kdl_kinematics::ParamListener::declare_params()’:
  /home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:[308](https://github.com/ros-planning/moveit2/actions/runs/8332188070/job/22800777158?pr=2747#step:12:322):36: error: expected unqualified-id before ‘.’ token
    308 |       auto& entry = updated_params..joints_map[value_1];
        |                                    ^
  gmake[2]: *** [kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/build.make:76: kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/src/kdl_kinematics_plugin.cpp.o] Error 1
  gmake[1]: *** [CMakeFiles/Makefile2:363: kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/all] Error 2
  gmake[1]: *** Waiting for unfinished jobs....
  gmake: *** [Makefile:146: all] Error 2
  ---
  Failed   <<< moveit_kinematics [51.7s, exited with code 2]
```

Yadunund, 2024-03-20:
@sea-bass I opened #2758 to track regression we're seeing on Iron.

Do you think the issue can be patched and bloomed in `generate_parameter_library` by end of this week? Else we may have to revert https://github.com/ros/rosdistro/pull/40149. Let me know what you think!

sea-bass, 2024-03-20:
> @sea-bass I opened #2758 to track regression we're seeing on Iron.
> 
> Do you think the issue can be patched and bloomed in `generate_parameter_library` by end of this week? Else we may have to revert [ros/rosdistro#40149](https://github.com/ros/rosdistro/pull/40149). Let me know what you think!

It's possible that https://github.com/PickNikRobotics/generate_parameter_library/pull/185 will fix it, and we'll see what we can do for end of the week.

If not, then that (and the corresponding humble PR to `rosdistro`) should indeed be reverted and we'll get this in for the next sync.

Yadunund, 2024-03-20:
Sounds good! Thanks for the quick response!

peter-pri, 2024-03-23:
Today I got also this error on my build of moveit2 on a Rasberry Pi 4.
I remove one of the two dots (three times) as a workaround in "kdl_kinematics_parameters.hpp".
Does this make sense ? Clearly, the generator program has to be fixed in the next days.
My build was successful with 8G RAM after I added "--executor sequential" to the build command:
[build_movit2_on_raspberry_pi_8GB_RAM.txt](https://github.com/ros-planning/moveit2/files/14733883/build_movit2_on_raspberry_pi_8GB_RAM.txt)

sea-bass, 2024-03-24:
https://github.com/PickNikRobotics/generate_parameter_library/pull/185 has been merged, so you can use that branch!

kimsooyoung, 2024-04-09:
I'm working on moveit2 with ubuntu 22.04 & ROS 2 Humble and just resolved this issue.

For those who are encountering this one, Here's my command line log.

```bash
# download new generate_parameter_library which one can resolve this err
cd ~/ws_moveit/src/
wget https://github.com/PickNikRobotics/generate_parameter_library/archive/refs/tags/0.3.8.tar.gz
tar -zxvf 0.3.8.tar.gz
rm -rf 0.3.8.tar.gz

# build downloaded generate_parameter_library and sourcing
cd ~/ws_moveit/
colcon build --symlink-install --packages-select generate_parameter_library --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/local_setup.bash

# Now build whole moveit2 pkgs again
colcon build --mixin release --parallel-workers 4
```

closed 2024-05-24 by sea-bass