moveit/moveit2 #2758: `moveit_kinematics` regression on ROS 2 Buildfarm for Iron

Yadunund, 2024-03-20:
The [next Iron sync](https://discourse.ros.org/t/preparing-for-iron-sync-2024-03-22/) is blocked by regressions in `moveit_kinematics` which affects several downstream packages.

See https://build.ros2.org/view/Ibin_uJ64/job/Ibin_uJ64__moveit_kinematics__ubuntu_jammy_amd64__binary/

```
In file included from /tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/kdl_kinematics_plugin.h:42,
                 from /tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:37:
/tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/.obj-x86_64-linux-gnu/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘void kdl_kinematics::ParamListener::refresh_dynamic_parameters()’:
/tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/.obj-x86_64-linux-gnu/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:146:36: error: expected unqualified-id before ‘.’ token
  146 |       auto& entry = updated_params..joints_map[value_1];
      |                                    ^
/tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/.obj-x86_64-linux-gnu/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘rcl_interfaces::msg::SetParametersResult kdl_kinematics::ParamListener::update(const std::vector<rclcpp::Parameter>&)’:
/tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/.obj-x86_64-linux-gnu/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:217:32: error: expected unqualified-id before ‘.’ token
  217 |                 updated_params..joints_map[value_1].weight = param.as_double();
      |                                ^
/tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/.obj-x86_64-linux-gnu/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘void kdl_kinematics::ParamListener::declare_params()’:
/tmp/binarydeb/ros-iron-moveit-kinematics-2.8.0/.obj-x86_64-linux-gnu/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:308:36: error: expected unqualified-id before ‘.’ token
  308 |       auto& entry = updated_params..joints_map[value_1];
      |                                    ^
```

sea-bass, 2024-03-21:
Hi guys,

We did not find the time to fix this bug in a timely fashion, so I think we should revert this on Humble/Iron/Rolling and we'll bloom a fixed up version of this release whenever we are ready.

Thanks for flagging this! We've had it breaking MoveIt's CI as well.

Yadunund, 2024-03-22:
@sea-bass thanks a lot for the proactive update, really appreciate it. I'll go ahead and revert bump to `generate_parameter_library` https://github.com/ros/rosdistro/pull/40305

MatthijsBurgh, 2024-03-22:
This explains the missing rolling release on Noble (No `.deb` for Noble)
[image]

`moveit_msgs` does have release:
[image]

closed 2024-06-20 by github-actions[bot]