moveit/moveit2 #2776: MoveIt2 Setup Assistant generates integer values for joint limits when float is expected

JorgePRamos, 2024-03-30:
### Description

When generating a new MoveIt 2 package the resulting 'config/joint:limits.yaml' presents integer values when float are expected.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Binary  2.5.5-1

### Steps to reproduce
Generate a moveit package using the set up assistant.
Once generated and build proceed to launch the 'demo.launch.py '
This will prompt the error regarding the expected float values.

To fix it is enough to locate the file 'config/joint:limits.yaml' in your moveit package and chage the values from '0 or 3' to '0.0 or 3.0'
After building the package launching the 'demo.launch.py ' will work as intended

### Backtrace or Console output
`[move_group-3] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[move_group-3]   what():  parameter 'robot_description_planning.joint_limits.art1_yaw.max_velocity' has invalid type: expected [double] got [integer]`
[image]

JorgePRamos, 2024-03-30:
I would really like to contribute to its fix. 
The code is daunting but I think that the error may be in the '.joint_limits.yaml' file composition, so maybe the file 'time_optimal_trajectory_generation.cpp'

jaredgonzales, 2024-04-16:
I'm also having this issue. While it would be nice to accept integers, a different fix could be generating the numbers as doubles from MoveIt Setup Assistant. I believe that is defined in `SRDFConfig::GeneratedJointLimits::writeYaml` inside of the file `moveit_setup_assistant/moveit_setup_framework/src/srdf_config.cpp`

closed 2024-07-24 by github-actions[bot]

sjahr, 2024-07-24:
Closed with https://github.com/moveit/moveit2/pull/2803